gitbls/sdm #215: Looking for way to set hostname to MAC address during first run

clayton-girouard, 2024-06-03:
Would like to have the system automatically set the hostname of the system to the MAC address of the ethernet interface. 

I've gotten this sort of working using this in the user-data file:

runcmd:
 - printf "$(cat /sys/class/net/e*/address | sed 's/://g')" > /etc/hostname
 - printf "Ubuntu 22.04 LTS \nIP - $(hostname -I)\n" > /etc/issue

gitbls, 2024-06-03:
The "sdm-fu" method to do this would be a small plugin that creates a script in /etc/sdm/0piboot with the contents:
```
#!/bin/bash
printf "$(cat /sys/class/net/e*/address | sed 's/://g')" > /etc/hostname
printf "Ubuntu 22.04 LTS \nIP - $(hostname -I)\n" > /etc/issue
```
You can see examples of how the included sdm plugins do this in the sdm plugins, for example, `postfix`.

The filename needs to be of the pattern `0*-*.sh`.

Scripts in /etc/sdm/0piboot are run during sdm-firstboot on the first boot of the system. You don't need to worry about the file protection, sdm-firstboot sets the protection of all files iin /etc/sdm/0piboot to 755 before running them.

LMK if you have any further questions on this.

Also, are you really using sdm on Ubuntu 22.04?? That's not been tested at all. Can you tell me a bit about that?

gitbls, 2024-07-03:
No activity so closing issue.

closed 2024-07-03 by gitbls