gitbls/sdm #217: Add ability to assign FS label in parted plugin

ykharuzhy, 2024-06-14:
It would be great if we can set a FS label when creating partitions with parted plugin.

I did some PoC here (checked for ext4 and should work for FAT): https://github.com/ykharuzhy/sdm/commit/611edfc0ea613c3ffbfea925bec9126e52ebb7a0 

Example: 
```
sdm" --batch --burnfile "$out_image" --burn-plugin parted:"addpartition=1024,ext4,primary,overlayroot+128,ext4,primary,media" "$out_image.tmp"
```

gitbls, 2024-06-14:
What would the value of this be to users?

ykharuzhy, 2024-06-14:
I use this to make rootfs overlay backed by partition, not tmpfs. Mounting
by label allow to have one image for SD and NVMe.

gitbls, 2024-06-15:
I'll have a look into this next week.

gitbls, 2024-06-23:
This will be in the next sdm release. Thanks for the POC!

gitbls, 2024-07-03:
This is included in V12.4. I'd appreciate it if you could test/verify. Thx!

gitbls, 2024-07-12:
Closed due to no activity.

closed 2024-07-12 by gitbls