gitbls/sdm #218: Add a user and their SSH key?

torntrousers, 2024-06-15:
Hello, sdm looks cool.

I wondered is there a way to add a user and their SSH key instead of their password?

TIA

gitbls, 2024-06-15:
Not only does sdm look cool, it *is* cool. 😎 Thanks for checking it out!

There is definitely a way to add a user with no password and their SSH key, but this capability is not magically built into sdm. You'll need to do a small bit of work yourself.

Here's how I'd do it:
* use the `user` plugin to add the user, but don't specify a password. You'll definitely want to use these keywords: `adduser` and `homeddir`. Others would depend on how you want to configure the user account
* (this is the slightly more complex part) Copy the file /usr/local/sdm/plugins/sdm-plugin-template` to your directory somewhere, and modify it. Modify it to add the following:
```
    # Add this code in Phase 0
    cp /path/to/authorized_keys $SDMPT/etc/sdm/assets/username-authorized_keys
```
In the above, best to replace *username* by the actual username you're going to use, for your own sanity, but that's up to you.

```
    # Add this code in Phase 1
    mkdir /home/username/.ssh
    cp /etc/sdm/assets/username-authorized_keys /home/username/.ssh/authorized_keys
    chown -R username:users /home/usern`ame/.ssh
    chmod 700 /home/username/.ssh
    chmod 600 /home/username/.ssh/authorized_keys
```
Same comment on *username* as the code in Phase 0. The string used in Phase 0 and Phase 1 must be identical.

When the plugin is called during Phase 0, the user's authorized_keys file will be copied from your local system into the IMG for later use. The string $SDMPT points to the IMG during Phase 0.

When the plugin is called during Phase 1 the local system disk is not directly accessible (which is why we copied it during Phase 0), so the code run in Phase 1 will create the user's .ssh directory, copy the stashed authorized_keys file into the proper location, and then set the correct file owner and protection. $SDMPT isn't needed during Phase 1, but you can add it if you prefer (it is defined as the empty string during Phase 1).

On the command line use `--plugin user:adduser=username --plugin /path/to/myplugin`. The `user` plugin must be specified before the plugin you just created.

`user` plugin documentation: https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#user
Building your own plugin documentation: https://github.com/gitbls/sdm/blob/master/Docs/Programming-Plugins-and-Custom-Phase-Scripts.md

If you run into problems you aren't able to resolve, please include the source to the plugin you created, as well as the exact command line you're using, and the contents of `/etc/sdm/history` from inside the customized IMG.

maxstrobel, 2024-06-20:
Why not simply use the mkdir & copyfile commands?

~~~bash
sudo sdm \
...
--plugin mkdir:"dir=/home/<username>/.ssh|chown=<username>:<username>|chmod=700" \
--plugin copyfile:"from=authorized_keys|to=/home/<username>/.ssh|chown=<username>:<username>|chmod=600|mkdirif" \
...
~~~

gitbls, 2024-06-20:
> Why not simply use the mkdir & copyfile commands?
> 
> ```shell
> sudo sdm \
> ...
> --plugin mkdir:"dir=/home/<username>/.ssh|chown=<username>:<username>|chmod=700" \
> --plugin copyfile:"from=authorized_keys|to=/home/<username>/.ssh|chown=<username>:<username>|chmod=600|mkdirif" \
> ...
> ```

Indeed, this should work as well and is a much better approach.

I'll plead "traveling and obviously distracted" 😑

torntrousers, 2024-06-20:
thanks both!

gitbls, 2024-07-03:
No activity so closing issue.

closed 2024-07-03 by gitbls