gitbls/sdm #220: Image is not sdm-enhanced

thk686, 2024-06-18:
I see
```
% IMG './hostdir/source_images/2024-03-15-raspios-bookworm-arm64-lite.img' is not sdm-enhanced
  Logs and configuration updates will not be written
```
what does that mean?

gitbls, 2024-06-18:
Presumably you're seeing this on an sdm `--burn` or `--burnfile` command. The message is telling you that the IMG that you're burning has not been `--customize`d with sdm, so the directory /etc/sdm and all the files sdm needs are not there.

Since that directory doesn't exist, there's no place for sdm to write the logs from the `--burn*` command. 

Pretty much all other switches for the `--burn*` command are also not processed in this case. If you're still puzzled, please post the `--burn*` command you're using, and please confirm that you have NOT run `--customize` on the IMG before burning it.

gitbls, 2024-06-18:
Where did you get that understanding?

While it's true that a `--burn*` can do most customization as well, at least a minimal customize must be done to get the sdm "stuff" into the IMG (all in /etc/sdm) before you `--burn*`

gitbls, 2024-06-18:
I should add...--burn and --burnfile are equivalent except for the output. One goes to a disk type device and the other is to an IMG file.

thk686, 2024-06-19:
The documentation is rather vague and disorganized. For the command line, typically, one would have something like `sdm customize --the --set --of --customization --compatible --switches` and `sdm burn --the --set --of --burn --compatible --switches` and so on. But I now have it working for the most part, which is terrific. This will really help with the setup. Thanks.

gitbls, 2024-06-19:
I'd be very interested in your feedback on where you think the documentation is "rather vague and disorganized".

Without constructive feedback from users it will likely remain as it is.

Thx

thk686, 2024-06-19:
Maybe use the wiki or "read the docs", etc. It could use a bit more conceptual background about the workflow. There is a lot of detail and its hard to read through in a linear fashion. A lot of utilities have `cmd input.file output.file` rather than working in-place, so without some overview, its easy to think that eg `--burnfile` is like `output.file` in this sort of configuration.

gitbls, 2024-06-19:
I'll look into adding some documentation around switches that work in customize, burn, or both. But by the time I get it done you'll fully understand and won't need it 🤣

gitbls, 2024-06-21:
The switches that can be used with `--burn` and `--burnfile` are actually [already documented here](https://github.com/gitbls/sdm/blob/master/Docs/Burn-Scripts.md#switches-that-work-with---burn)

The README on https://github.com/gitbls/sdm provides a complete overview of how to use sdm. There is also an [index to all the documentation pages,](https://github.com/gitbls/sdm/blob/master/Docs/Index.md) and the link names are certainly helpful in knowing the contents of each page.

That said, I agree that there is a fair amount of documentation to wade through, so there may be a gap around providing _workflow conceptual background_ information.  I'm not sure what that would look like relative to the existing documentation, so would very much appreciate your thoughts on this.

closed 2024-06-29 by thk686