gitbls/sdm #221: Failure in docker image

thk686, 2024-06-19:
I am seeing:
```
* Phase 0 Completed
* Enter image './hostdir/customized_images/testing-2024-03-15-raspios-bookworm-arm64-lite.img' for Phase 1
Failed to open system bus: No such file or directory
Attempted to remove disk file system under "/run/systemd/nspawn/propagate/sdm", and we can't allow that.
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```
when running in docker. I think this is an issue with my docker setup rather than sdm. Just checking to see if this is a known or previously encountered issue.

gitbls, 2024-06-19:
If you provide the sdm command line that's failing, I may be able to tell you why it's failing and/or how to work around it.

I don't use docker for a variety of reasons, so unless you provide me with sufficient information, to replicate in a simple scenario there's not much I can do to help you.

thk686, 2024-06-19:
Sorry for the sparse info. I figured out that the issue is that I need to specify `--chroot`.

closed 2024-06-19 by thk686