gitbls/sdm #223: system:"name=s1|systemd-config=system=... fails

thk686, 2024-06-22:
I am getting:

```
> Run Plugin 'system' (/usr/local/sdm/plugins/system) Phase post-install with arguments: 
  'name=s1|systemd-config=system=biosense-config-files/remove-ssh-success.service'
* Plugin system: Start Phase post-install
> Plugin system: Copy systemd 'system' config file 
  '/etc/sdm/assets/system-s1/systemd-config/system/*' to /etc/systemd/system.conf.d
cp: cannot stat '/etc/sdm/assets/system-s1/systemd-config/system/*': No such file or directory
chown: cannot access '/etc/systemd/system.conf.d/*': No such file or directory
chmod: cannot access '/etc/systemd/system.conf.d/*': No such file or directory
* Plugin system: Complete Phase post-install
> Run Plugin 'system' (/usr/local/sdm/plugins/system) Phase post-install with arguments: 
  'name=s2|systemd-config=system=biosense-config-files/ssh-retry.service'
* Plugin system: Start Phase post-install
> Plugin system: Copy systemd 'system' config file 
  '/etc/sdm/assets/system-s2/systemd-config/system/*' to /etc/systemd/system.conf.d
cp: cannot stat '/etc/sdm/assets/system-s2/systemd-config/system/*': No such file or directory
chown: cannot access '/etc/systemd/system.conf.d/*': No such file or directory
chmod: cannot access '/etc/systemd/system.conf.d/*': No such file or directory
* Plugin system: Complete Phase post-install
```

Did I get the syntax wrong?

gitbls, 2024-06-22:
It's not clear to me what `biosense-config-files` is.

Replace that string with `/path/to/mysystem.conf`. sdm will then copy /path/to/mysystem.conf to /etc/systemd/system.conf.d/mysystem.conf

As another exasmple, you can update timesyncd.conf with:
```
systemd-config=timesyncd=/somedir/systemd/timesyncd.conf
```
systemd nicely provides the xxxx.conf.d directories that enable you to override settings for login, system, timesync, and user.

thk686, 2024-06-22:
Ah. So full paths. Got it.

gitbls, 2024-06-22:
Yes, and the file type MUST be ".conf". sdm doesn't check that. 

Based on this I'm updating the `system` plugin doc to reflect that a full path is required. Thx for pointing that out! 👍

thk686, 2024-06-22:
Thanks!Other quick question: rsync options are with or without the leading “—“?Sent from my iPhone

gitbls, 2024-06-22:
> Thanks!Other quick question: rsync options are with or without the leading “—“?

If this is for the `copydir` plugin argument `rsyncopts` then yes, the leading slash (or slashes, as the case may be) must be specified. the copydir plugin simply takes the string and drops it into the command line instead of using `-a`, so you need to include your own `-a` if you need it.

If I misunderstood, please clarify which rsync options? 

Thx

closed 2024-06-29 by thk686