gitbls/sdm #226: Run burn from inside docker container

thk686, 2024-06-27:
I'm having trouble mapping block devices into docker. Here is my command:

`docker run --privileged -it --rm -v /Users/keittth/Desktop/biosense-docker:/home/agent/hostdir -v /dev:/dev biosense-sdm /bin/bash -c 'sudo sdm --burn /dev/disk2 --hostname testing     --expand-root     --nowait-timesync     /home/agent/hostdir/customized_images/latest-lite.img'`

Results:

```
* Start Burn
? '/dev/disk2' is not a block device
```

Any suggestions?

gitbls, 2024-06-27:
What is `/dev/disk2` ?

thk686, 2024-06-27:
Supposedly the block device where the sd card sits. The docker container is running on osx.Sent from my iPhone

gitbls, 2024-06-27:
What is the output from `sudo lsblk -l /dev/disk2` both outside the container and inside?
``

thk686, 2024-06-28:
```
keittth@cns-f-bioa31235 biosense-docker % diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *2.0 TB     disk0
   1:                        EFI EFI                     314.6 MB   disk0s1
   2:                 Apple_APFS Container disk1         2.0 TB     disk0s2

/dev/disk1 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +2.0 TB     disk1
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD - Data     519.9 GB   disk1s1
   2:                APFS Volume Preboot                 2.2 GB     disk1s2
   3:                APFS Volume Recovery                1.2 GB     disk1s3
   4:                APFS Volume VM                      2.1 GB     disk1s4
   5:                APFS Volume Macintosh HD            10.3 GB    disk1s5
   6:              APFS Snapshot com.apple.os.update-... 10.3 GB    disk1s5s1

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *31.9 GB    disk2
   1:             Windows_FAT_32 bootfs                  536.9 MB   disk2s1
   2:                      Linux                         2.5 GB     disk2s2
                    (free space)                         28.8 GB    -
```

```
docker run --rm -it -v /dev:/dev --privileged biosense-sdm
```

```
NAME MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
vdb  254:16   0  134M  1 disk 
NAME MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
vda  254:0    0 59.6G  0 disk 
vda1 254:1    0 59.6G  0 part /etc/hosts
                              /etc/hostname
                              /etc/resolv.conf
NAME MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
vda1 254:1    0 59.6G  0 part /etc/hosts
                              /etc/hostname
                              /etc/resolv.conf
NAME MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
vdb  254:16   0  134M  1 disk 
NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
vdc  254:32   0 397.4M  1 disk 
```

gitbls, 2024-06-28:
> It looks like this will not work using OSX. I'll have to do it on a Linux machine.
Sorry. I don't have a Mac, and I don't use Docker (for a number of reasons). 

As you're discovering, there is absolutely zero sdm testing done on a Mac, but it works really well on several different Linux distros and processor architectures.

closed 2024-06-29 by thk686