gitbls/sdm #229: Dump kernel image and DBT

thk686, 2024-07-02:
For verification, I am trying to boot the customized rpi os image using qemu. Is there a way to copy the kernel image and DBT out of the image? I can do that manually after running sdm. I was just wondering if it could be done inside sdm since it has already mounted the image.

gitbls, 2024-07-02:
Here are a couple of ways using sdm that might work for you:
* Use `sudo sdm --mount /path/to/IMG` to mount the customized IMG as a disk in the running system. You can then copy off whatever you want. **Mind the warning!**
* After a customize you'll end up in the IMG container unless you use `--batch`. At that point you can copy files out of the IMG to somewhere using `scp`
* Use the `extractfs` plugin. You can use it to copy the `bootfs` from the IMG to a directory tree.  (Oh, maybe not...this is a *burn plugin* and only works after a `--burn` or `--burnfile`
* Write a plugin that copies off the bits you want to somewhere else. Kind of overkill IMHO but depending on your needs...

thk686, 2024-07-02:
Thanks. I ended up writing a custom script. But I'll probably move to using `sdm --mount` so it will be inside docker (debian).

closed 2024-07-02 by thk686