gitbls/sdm #233: Hyphens and underscores in hostname

thk686, 2024-07-09:
Are they allowed? I used `--hostname host_0_1` and it showed as host01 on boot. Trying hyphens now.

thk686, 2024-07-09:
I confirmed the underscores were preserved, so I guess the prompt does not display them.

closed 2024-07-09 by thk686

thk686, 2024-07-09:
Double checked and I guess underscores are not allowed, but hyphens are.

gitbls, 2024-07-09:
[RFC 952](https://www.rfc-editor.org/rfc/rfc952) specifies legal characters for hostnames: 
```
A "name" (Net, Host, Gateway, or Domain name) is a text string up
   to 24 characters drawn from the alphabet (A-Z), digits (0-9), minus
   sign (-), and period (.).  Note that periods are only allowed when
   they serve to delimit components of "domain style names". 
```

thk686, 2024-07-10:
Yep. Oddly, the first hit on google said underscores are allowed. Don't
believe everything (or the first thing) you read on the internet!