gitbls/sdm #238: Multiple sdm customize runs

thk686, 2024-07-15:
Can I break the process into two parts writing out a first phase image and then updating it? Some parts take a long time, but do not need to be repeated for each host, so I thought I would make a base customized image and specialize it for each target. What happens if I run sdm again on an image once its been run previously on that image?

gitbls, 2024-07-15:
See here: https://github.com/gitbls/sdm/issues/237#issuecomment-2227504480

thk686, 2024-07-15:
Thanks. I can't quite tell if that solves my problem. I create a series of images with different hostnames, but I need to download a bunch of files each time. It seems that should only need to be done once. I guess I could fix that if I was able to burn the images from the docker container. Can I burn to a file and then flash that file to the device?

gitbls, 2024-07-15:
If the IMGs only differ by the hostname, then you're doing it wrong, and maybe didn't understand what I wrote over on the above-referenced topic. 

If you explain how you're doing things now, I'll be happy to make improvement suggestions.

As far as "can I burn to a file then flash"?  Yes, it's just an IMG, so you can of course burn it to a disk. In case you're interested, the `--convert-root` switch does not work on burning to a file.

thk686, 2024-07-16:
Often I need to also set wifi ssid and password in addition to the hostname. I'll see if I can split it into a customize and  burn phase with the burn phase written to a file. Because I'm on a mac, its easier to use the rpi burning app.

gitbls, 2024-07-16:
You can definitely do network configuration during the burn (or burnfile) by using `--plugin network` configured as you want. 

Understand about there being burn issues on the Mac. The one thing you lose by not using sdm for the burn is that the root partition won't be expanded until the system first boots. I haven't looked at that in a while, but it may require an extra reboot (which is what caused me to add expand root in the first place).

Here's what I'd do given what you've told me:
* Do a customize on a fresh IMG and customize literally everything that should be on all systems
* Do a burnfile and run the plugins that are specific to the host you're burning for
* rpi-imager copy that resulting burn img to a disk and boot it

Obviously the more you put in the common IMG the less that will be required at boot time.

If you still need to do more than just hostname and network (especially installing additional apps) you might want to have a look at using apt-cacher-ng running on a Debian-based system (pi or otherwise). It will still install a bunch of files, but they'll be copied from the apt-cacher-ng server instead of over the internet. The time reduction increases significantly the more you would have downloaded.  Plus, you'll benefit each time you rebuild the common base disk (after the first time of course) since those bits will all be copied across the LAN as well.

thk686, 2024-07-16:
Thanks for the suggestions. I had started splitting the scripts/plugins as
you say. I think I just need to change the second stage to --burn and
--burnfile (or does --burnfile imply --burn?)

gitbls, 2024-07-16:
`--burn` takes an argument that is the name of the device (e.g., `/dev/sdX` or whatever it is)
`--burnfile` takes an argument that is the name of the file where the "burned" IMG will be written
The two switches are conflicting and cannot be used together on the command line.
They do the same thing, as noted, except that `--burnfile` cannot be used wih `--convert-root` or `--expand-root`

gitbls, 2024-08-08:
Closing due to no activity.

closed 2024-08-08 by gitbls