gitbls/sdm #239: Are plugin arguments robust to whitespace?

thk686, 2024-07-16:
I happen to be somewhere that the ssid has whitespace. Will that work?

* network
       Args: wifissid=Admin TBS|....

gitbls, 2024-07-16:
Bummer for you. Who does that???

I don't know what the correct incantation is offhand, and tied up so can't look into it.

Suggest you try using `\ ` for the space

thk686, 2024-07-16:
I "looks" like the string is handled correctly, at least from what sdm
prints to the screen. But who knows once it get to the rest of the system.

thk686, 2024-07-17:
Hell is other people's networks.

```
> Plugin network: Create NetworkManager WiFi configuration
  with SSID: 'Admin TBS' Password: 'Admin_TBS_2024' WiFi Country: 'US'
> Plugin network: Configure NetworkManager WiFi connection 'admin tbs' with SSID 'Admin TBS'
Error: invalid <setting>.<property> 'tbs'.
mv: target 'tbs.nmconnection': No such file or directory
```

gitbls, 2024-07-17:
Crazy. `\ ` should fix it right up 😉

thk686, 2024-07-18:
Needed \\

thk686, 2024-07-18:
Actually, I have tried up to \\\\\\ and still get the error.

* Plugin network: Start Phase post-install
> Plugin network: Keys/values found:
   wifissid: Admin TBS
   wifipassword: Admin_TBS_2024
   wificountry: US
   noipv6: y
> Plugin network: Set WiFi Country 'US' for sdm FirstBoot
> Plugin network: Enable SSH service
> Plugin network: Create NetworkManager WiFi configuration
  with SSID: 'Admin TBS' Password: 'Admin_TBS_2024' WiFi Country: 'US'
> Plugin network: Configure NetworkManager WiFi connection 'admin tbs' with SSID 'Admin TBS'
Error: invalid <setting>.<property> 'tbs'.
mv: target 'tbs.nmconnection': No such file or directory

thk686, 2024-07-18:
A quick run through chatgpt suggests that this is the error.

--- original_script.sh  2024-07-18 12:00:00.000000000 +0000
+++ modified_script.sh  2024-07-18 12:00:00.000000000 +0000
@@ -134,8 +134,8 @@
 		then
 		    if [ "$wifissid" != "" -a "$wifipassword" != "" -a "$wificountry" != "" ]
 		    then
-			logtoboth "> Plugin $pfx: Create NetworkManager WiFi configuration"
-			logtoboth "  with SSID: '$wifissid' Password: '$wifipassword' WiFi Country: '$wificountry'"
+			logtoboth "> Plugin $pfx: Create NetworkManager WiFi configuration with SSID: '$wifissid' Password: '$wifipassword' WiFi Country: '$wificountry'"
+			wificname=$(echo "${wifissid,,}" | tr -d ' ')
 			if [ $newnm -eq 1 ]
 			then
 			    # --offline should work
-			    logtoboth "> Plugin $pfx: Configure NetworkManager WiFi connection '$wificname' with SSID '$wifissid'"
-			    nmcli --offline c add type wifi con-name $wificname \
-				  ifname wlan0 ssid $wifissid wifi-sec.key-mgmt \
+			    logtoboth "> Plugin $pfx: Configure NetworkManager WiFi connection '$wificname' with SSID '$wifissid'"
+			    nmcli --offline c add type wifi con-name "$wificname" \
+				  ifname wlan0 ssid "$wifissid" wifi-sec.key-mgmt \
 				  wpa-psk wifi-sec.psk "$wifipassword" | tee /tmp/$wificname.nmconnection >/dev/null
 			    mv /tmp/$wificname.nmconnection /etc/NetworkManager/system-connections/$wificname.nmconnection
 			else

gitbls, 2024-07-18:
I'll have a look at this next week when I'm back at my desk.

thk686, 2024-07-18:
This is the problem with bash. (Believe me, my bash skills are incredibly simple.) I don't know if you are into chatgpt, but it can be a great linter in some cases. 

```
-original: source $SDMPT/etc/sdm/sdm-readparams
-modified: source "$SDMPT/etc/sdm/sdm-readparams"

-original: IFS="=" read a wificountry <<< $(grep 'country=' $assetdir/wpa.conf | head -n 1)
-modified: IFS="=" read -r a wificountry <<< "$(grep 'country=' "$assetdir/wpa.conf" | head -n 1)"

-original: logtobothex "? Plugin $pfx: Cannot find $assetdir/wpa.conf"
-modified: logtobothex "? Plugin $pfx: Cannot find \"$assetdir/wpa.conf\""

-original: cat > $assetdir/wpa.conf <<EOF
-modified: cat > "$assetdir/wpa.conf" <<EOF

-original: [ -f $assetdir/wpa.conf ]
-modified: [ -f "$assetdir/wpa.conf" ]

-original: IFS="=" read a wifissid <<< $(grep 'ssid=' $assetdir/wpa.conf | head -n 1)
-modified: IFS="=" read -r a wifissid <<< "$(grep 'ssid=' "$assetdir/wpa.conf" | head -n 1)"

-original: IFS="=" read a wifipassword <<< $(grep 'psk=' $assetdir/wpa.conf | head -n 1)
-modified: IFS="=" read -r a wifipassword <<< "$(grep 'psk=' "$assetdir/wpa.conf" | head -n 1)"

-original: IFS="=" read a wificountry <<< $(grep 'country=' $assetdir/wpa.conf | head -n 1)
-modified: IFS="=" read -r a wificountry <<< "$(grep 'country=' "$assetdir/wpa.conf" | head -n 1)"

-original: nmcli --offline c modify $cattr </etc/NetworkManager/system-connections/$fname.nmconnection | tee /tmp/$fname.nmconnection >/dev/null
-modified: nmcli --offline c modify "$cattr" </etc/NetworkManager/system-connections/"$fname".nmconnection | tee /tmp/"$fname".nmconnection >/dev/null

-original: rm -f /etc/NetworkManager/system-connections/$fname.nmconnection
-modified: rm -f /etc/NetworkManager/system-connections/"$fname".nmconnection

-original: mv /tmp/$fname.nmconnection /etc/NetworkManager/system-connections/$fname.nmconnection
-modified: mv /tmp/"$fname".nmconnection /etc/NetworkManager/system-connections/"$fname".nmconnection

-original: echo "nmcli c modify $cname $cattr" >> $nmdcfg
-modified: echo "nmcli c modify $cname $cattr" >> "$nmdcfg"

-original: phase=$1
-modified: phase="$1"

-original: pfx="$(basename $0)"
-modified: pfx="$(basename "$0")"

-original: args=$2
-modified: args="$2"

-original: assetdir=$SDMPT/etc/sdm/assets/network
-modified: assetdir="$SDMPT/etc/sdm/assets/network"

-original: plugin_getargs $pfx "$args" "$vldargs" "$rqdargs" || exit
-modified: plugin_getargs "$pfx" "$args" "$vldargs" "$rqdargs" || exit

-original: mkdir -p $assetdir
-modified: mkdir -p "$assetdir"

-original: netman=$(getdefnm "$netman")
-modified: netman=$(getdefnm "$netman")

-original: if [ -f $wpa ]
-modified: if [ -f "$wpa" ]

-original: tr -d '\r' < $wpa > $assetdir/wpa.conf
-modified: tr -d '\r' < "$wpa" > "$assetdir/wpa.conf"

-original: logtobothex "? Plugin $pfx: wpa file '$wpa' not found"
-modified: logtobothex "? Plugin $pfx: wpa file \"$wpa\" not found"

-original: if [ -f $dhcpcdappend ]
[... 100 lines cut ...]

-original: nmcli --offline c add con-name "$lancname" type ethernet ifname $ifname | tee /etc/NetworkManager/system-connections/$lancname.nmconnection >/dev/null
-modified: nmcli --offline c add con-name "$lancname" type ethernet ifname "$ifname" | tee /etc/NetworkManager/system-connections/"$lancname".nmconnection >/dev/null

-original: echo "nmcli c add con-name "$lancname" type ethernet ifname $ifname" >> $nmdcfg
-modified: echo "nmcli c add con-name "$lancname" type ethernet ifname "$ifname"" >> "$nmdcfg"

-original: while read nmfile
-modified: while read -r nmfile

-original: IFS="=" read idx cname < <(grep ^id= $nmfile)
-modified: IFS="=" read -r idx cname < <(grep ^id= "$nmfile")

-original: modnmconn "ipv6.method disabled" $cname $nmfile
-modified: modnmconn "ipv6.method disabled" "$cname" "$nmfile"

-original: while read nmfile
-modified: while read -r nmfile

-original: IFS="=" read idx cname < <(grep ^id= $nmfile)
-modified: IFS="=" read -r idx cname < <(grep ^id= "$nmfile")

-original: modnmconn "ipv6.method disabled" $cname $nmfile
-modified: modnmconn "ipv6.method disabled" "$cname" "$nmfile"
```

thk686, 2024-07-18:
I fixed the problem by hand editing the plugin using `sdm explore`.

gitbls, 2024-08-08:
Wow. Somehow I completely missed this. 😵 Haven't looked at the code yet, but clearly some of these are just wrong and will fix. Others...need to have a look at them more closely. Thanks!

gitbls, 2024-08-08:
Derived connection name now has spaces removed which eliminates all the issues, except for one other instance which is also fixed. Release coming soon.

gitbls, 2024-08-09:
V12.6 released which fixes this.

gitbls, 2024-08-24:
Closing due to no activity. Please re-open if you're still having issues.

closed 2024-08-24 by gitbls

thk686, 2024-08-25:
Thanks!Sent from my iPhone

kpotier, 2024-11-06:
Hello,

I’m facing an issue with a Wi-Fi SSID that includes spaces and special characters (double lol). When customizing the image, I encounter the following error in post-install phase, though it doesn’t cause a failure:

```
* Plugin network: Start Phase post-install
> Plugin network: Keys/values found:
   ifname: wlan0
   wifissid: Voie lactée
   wifipassword: password
   wificountry: FR
> Plugin network: Set WiFi Country 'FR' for sdm FirstBoot
> Plugin network: Create connection 'voielactée' on interface 'wlan0'
Error: invalid <setting>.<property> 'lactée"'.
* Plugin network: Complete Phase post-install
```

```
IMG Information
   Name: raspi64.img
   Date: 2024-10-22
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
```

Everything works perfectly when using ethernet and this is what I get when I list the connections:

```
$ nmcli connection show "voielactée"
Erreur : voielactée - profil de connexion introuvable. # Error: voielactée - connection profile not found.
```

I tried using a backslash (\ and \\) as suggested but had no success.

Thanks for your patience!

kpotier, 2024-11-06:
@gitbls should reopen maybe?

gitbls, 2024-11-06:
For NM connections with spaces in the SSID or password the best solution is to build a .nmconnection on your own and feed it into the network plugin using the `nmconn` argument.