gitbls/sdm #240: Can't set wifi ssid and password in the burn phase

thk686, 2024-07-18:
I just figured out why I spend days debugging headless wifi. You can't do the wifi settings with burn, at least from what I can gather from the docs. So I can't do two phases because every device typically needs different wifi settings.

gitbls, 2024-07-18:
There's no reason that configuring wifi settings as part of burn doesn't work. If it doesn't work, it's a bug.

thk686, 2024-07-18:
Those are not listed in the docs as being compatible with --burnfile. I am
testing now however.

gitbls, 2024-07-18:
Anything that is compatible with --burn is compatible with --burnfile except for --convert-root and --expand-root, which I thought I previously mentioned.

gitbls, 2024-08-08:
Closing. Reopen if necessary. Thx

closed 2024-08-08 by gitbls