gitbls/sdm #241: Specify wlan1 instead of wlan0?

thk686, 2024-07-18:
I have a usb wifi. How do I configure that one instead of the built-in wifi?

gitbls, 2024-07-19:
You'll have to wait for an update to the network plugin next week.

In the meantime you can edit the .nmconnection file to adjust it.

gitbls, 2024-07-23:
I pushed an updated network plugin that has the argument `iwname`. If not specified, the default is `wlan0`. If specified, the interface provided will be configured in the .nmconnection file.

There is obviously no checking on the validity of the device name since the target system may have devices that the host system does not.

Please test and LMK if this works for you. Thx

gitbls, 2024-08-08:
Closing due to no activity, and I know that it works 👍

closed 2024-08-08 by gitbls