gitbls/sdm #245: Installing on Fedora / Arch?

tgrushka, 2024-08-05:
Fedora (or Arch) does not support apt, but rather dnf. Maybe there are other differences.

Nonetheless, the install script fails on Fedora 40.

I would like / should be able to use this tool on Fedora, right?

Maybe in a VM... but that seems overkill.

Thanks!

tgrushka, 2024-08-06:
Well, I guess Docker is currently an exercise in futility due to #155 and #216. Can't burn images from it because of requiring loopback devices? Don't get it, but that's what I need it for.

gitbls, 2024-08-06:
Yea, sorry. Fedora is out of scope. You might be able to install it (just install the fedora equivalent of the  following packages:
```
binfmt-support gdisk parted qemu-user-static rsync systemd-container uuid
```
(If any of the fedora names are different that should help you understand why only Debian hosts are supported.)

Then, you can try using sdm to customize an IMG and it may work. Or it might not. It's never been tested.

If you really want to use sdm there's always the option of using the Pi system that you're building these IMGs for...

gitbls, 2024-08-24:
Closing issue since @tgrushka did not respond re helping to make this work.

closed 2024-08-24 by gitbls