gitbls/sdm #246: Network plugin fails to add wifi if nmconnections manually added previously

Phasip, 2024-08-09:
Hello!

I noticed that the network plugin (netman=nm) will not add wifi nor the default wired connection if we use the copy plugin to add .nmconnections files before the network plugin runs.

I think the culprit may be this line https://github.com/gitbls/sdm/blob/d2692a98064a4c926dd0fb6c8a450383ed83c0c8/plugins/network#L327
```
if ! compgen -G "/etc/NetworkManager/system-connections/*.nmconnection" >/dev/null && ! grep -qs interface-name=$iwname /etc/NetworkManager/system-connections/*.nmconnection >/dev/null 2>&1
```
Maybe the && should be ||, i.e "If we can't find any nmconnections files OR no files refer to iwname"

gitbls, 2024-08-09:
Please provide the full `network` plugin argument list you're using, as well as for the `copyfile` plugin, and the order in which they are invoked.

Note that if there is no Wired Connection when nm starts, it will create one by default. Precisely what behavior are you expecting to see, and in what way does the current behavior not meet your expectation?

Thx!

Phasip, 2024-08-09:
Ah sorry for the lack of details. I also made the example more succint by removing the use of copyfile and instead use the nmconn argument. 
Disregard the part about the wired connection, i was confused.

How to reproduce:
```
$ touch test.nmconnection
$ sudo sdm --customize --batch --plugin network:'netman=nm|wifissid=mywifi|wifipassword=mypass|wificountry=SE|nmconn=./test.nmconnection' rpi.img  
```

Expected outcome:
    /etc/NetworkManager/system-connections/ contains two files
        test.nmconnection
        mywifi.nmconnection

Current outcome:
    /etc/NetworkManager/system-connections/ contains
        test.nmconnection

Workaround:
```
$ mv test.nmconnection test
$ sudo sdm --customize --batch --plugin network:'netman=nm|wifissid=mywifi|wifipassword=mypass|wificountry=SE|nmconn=./test' rpi.img 
```
This results in both the wifi and the manually added one to be created:
```
    root@sdm:/etc/NetworkManager/system-connections# ls -la
    total 12
    drwxr-xr-x 2 root root 4096 Aug  9 16:48 .
    drwxr-xr-x 7 root root 4096 Jul  4 00:08 ..
    -rw------- 1 root root  252 Aug  9 16:48 mywifi.nmconnection
    -rw------- 1 root root    0 Aug  9 15:53 test
```

gitbls, 2024-08-09:
Ah, got it. Your diagnosis and proposed solution was correct. While I was in there I rearranged some code to make more sense. 

This will be fixed in V12.6 which should be released out later today 🤞

gitbls, 2024-08-09:
V12.6 is released with this problem corrected. OFC I'd appreciate your verifying this in your scenario.

Phasip, 2024-08-10:
Verified fixed.

closed 2024-08-10 by Phasip