gitbls/sdm #249: System don't boot after encryption - /dev/mapper/cryptroot does not exist

cAltaITc, 2024-08-13:
```
bash-5.2# sdmcryptfs /dev/nvme0n1 /dev/sda

    2024-08-05 10:45:06 Shrink partition '/dev/nvme0n1p2' and get its size
    2024-08-05 10:45:49 Device '/dev/nvme0n1' rootfs size: 1997902 4K blocks (8.2GB, 7.6GiB)
    2024-08-05 10:45:49 Save rootfs '/dev/nvme0n1p2' to '/dev/sda'
    2024-08-05 10:45:49 rootfs save should take less than 8 minutes
    1997902+0 records in
    1997902+0 records out
    2024-08-05 10:48:30 rootfs Save elapsed time: 00:02:41
    2024-08-05 10:48:30 Enable LUKS2 encryption on '/dev/nvme0n1p2' with cipher 'aes-xts-plain64'
    2024-08-05 10:48:30 Enabling encryption could take a while
    2024-08-05 10:48:30 OK to ignore superblock signature warning
    WARNING: Device /dev/nvme0n1p2 already contains a 'ext4' superblock signature.

WARNING!

This will overwrite data on /dev/nvme0n1p2 irrevocably.

Are you sure? (Type 'yes' in capital letters): YES
Enter passphrase for /dev/nvme0n1p2:
Verify passphrase:

    2024-08-05 10:50:09 Unlock encrypted partition '/dev/nvme0n1p2'
    2024-08-05 10:50:09 Unlock will take several seconds
    Enter passphrase for /dev/nvme0n1p2:
    2024-08-05 10:50:26 Restore '/dev/nvme0n1p2' from '/dev/sda'
    2024-08-05 10:50:26 rootfs restore should take less than 8 minutes
    1997902+0 records in
    1997902+0 records out
    2024-08-05 10:51:22 rootfs Restore elapsed time: 00:00:55
    2024-08-05 10:51:22 Restore complete; Expand rootfs...
    resize2fs 1.47.0 (5-Feb-2023)
    Resizing the filesystem on /dev/mapper/cryptroot to 244054452 (4k) blocks.
    The filesystem on /dev/mapper/cryptroot is now 244054452 (4k) blocks long.

    2024-08-05 10:51:22 rootfs partition size: 999647043584 Bytes (999.6GB, 931.0GiB)
    2024-08-05 10:51:22 cryptsetup status for /dev/mapper/cryptroot:
    /dev/mapper/cryptroot is active.
    type: LUKS2
    cipher: aes-xts-plain64
    keysize: 256 bits
    key location: keyring
    device: /dev/nvme0n1p2
    sector size: 512
    offset: 32768 sectors
    size: 1952435632 sectors
    mode: read/write
    Wipe scratch disk '/dev/sda' [y/N]? y
    2024-08-05 10:51:49 > Write random data on '/dev/sda'
    1997902+0 records in
    1997902+0 records out
    2024-08-05 10:54:57 rootfs Wipe elapsed time: 00:03:08

Encryption complete. System boot will continue in 10 seconds

Connection to 192.168.1.6 closed by remote host.
Connection to 192.168.1.6 closed.
```

[image]

Does anyone have any idea where this could be coming from please ?

I tried to reboot the raspberry pi but there is apparently a problem with cryptroot

[image]

gitbls, 2024-08-13:
Yes, something is broken, but you've provided literally no helpful information. Please describe how you used sdm to get to this point (sdm command line, etc), and the environment in which you're running it.

If you would like to move this process along more quickly, please also post /etc/sdm/history from the IMG you customized. If you added the encryption during the burn process, then also provide the `sdm --burn` command line. Please remember to redact information you don't want shown (e.g., passwords and SSIDs)

cAltaITc, 2024-08-13:
I just followed this tutorial [Disk-Encryption.md](https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md) with these steps :

```
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-cryptconfig -o /usr/local/bin/sdm-cryptconfig
sudo chmod 755 /usr/local/bin/sdm-cryptconfig
sudo sdm-cryptconfig --ssh --authorized-keys .ssh/authorized_keys
```

*Reboot*

`sdmcryptfs /dev/nvme0n1 /dev/sda`

-> Success

gitbls, 2024-08-13:
So you now have it working?

cAltaITc, 2024-08-13:
No, as I indicated it doesn't work after the last restart

gitbls, 2024-08-13:
OK, thanks for clarifying.  So this is on a system that you've already installed, booted, and run on disk /nvme0n1 and /dev/sda is your scratch disk? And, presumably this is on a Pi5?

PS Terse answers aren't very helpful.

gitbls, 2024-08-13:
OK, thanks.

cAltaITc, 2024-08-13:
Geekworm X1001 PCIe to M.2 NVMe SSD Shield

gitbls, 2024-08-13:
Needing to ask twice to get all the needed information is getting old.

As I previously asked: What manufacturer/model is your SSD?

cAltaITc, 2024-08-13:
I read a little quickly, sorry...

Crucial P1 1TB 3D NAND NVMe PCIe Internal SSD, up to 2000MB/s - CT1000P1SSD8

gitbls, 2024-08-13:
I just did a complete repro on a Pi5 with a Geekworm X1001 and a Kingston NV2 NVMe drive
* Install onto SD card
* Boot that SD card
* Install rpi-imager on system booted from SD card
* Burn to NVME using rpi-imager
* Boot NVME
* Download, install, and run sdm-cryptconfig
* Went through the entire process
* It's now up and running

After the sdmcryptfs stage completes, the system continues through the boot process and then reboots itself again. Did you let it run to completion on this and reboot? This is a critical step.

So, no clue why yours is failing. It could be some weird problem with the Crucial NVMe SSD, you didn't wait (previous paragraph) or you modified the system in some way before running sdm-cryptconfig

If you want further help on this, let's start by you telling me every step you took, just as I did above. With as much detail as possible. If you made ANY changes to the system after it boots the first time, and before you ran sdm-cryptconfig, please describe them in detail.

gitbls, 2024-08-15:
Closing issue since the problem initiator (cAltaITc) no longer exists on GitHub.

closed 2024-08-15 by gitbls