gitbls/sdm #250: How to best update sdm?

rvaneerd, 2024-08-15:
Hi,
I noticed some updates to the git repository have been applied.
That makes me wonder how to best update the existing sdm installation as there seems to be only a way of installing (and removing).

When I do run that installer again, I get this:
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
_I may have re-installed several times before I started this new issue causing the "9 not upgraded" because they might have been upgraded during the earlier attempts._

I did find the documentation https://github.com/gitbls/sdm/blob/master/Docs/Installing-or-Removing-sdm.md

I could remove SDM and re-install it but is running the installer again the way to update it?

Regards,
Rob.

gitbls, 2024-08-15:
Indeed, the supported way to update sdm is to run the installer again. At the moment EZsdmInstaller is a self-contained script, so the list of files that it downloads are contained within, necessitating the re-download. So, an upgrade is simply:
```
curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
```

EZsdmInstaller will also try to install the sdm-required system packages via sdm, but it will quickly find that they're already installed, which you are seeing with the '0 upgraded, 0 newly installed'.

The `9 not upgraded `means that if you do
```
sudo apt list --upgradable
```
you'll see the list of 9 packages that can be updated with `sudo apt upgrade`. These packages have nothing to do with sdm, but are reported by the `apt` command.

rvaneerd, 2024-08-16:
Hi @gitbls,
As mentioned in #251 I can continue.
How to access /etc/sdm/apt.log is now clear to me.

Because of that this issue can be closed.

Regards and thanks for the support,
Rob

gitbls, 2024-08-16:
> Hi @gitbls, As mentioned in #251 I can continue. How to access /etc/sdm/apt.log is now clear to me.
> 
> Because of that this issue can be closed.
> 
> Regards and thanks for the support, Rob

Great, thanks for the followup! Please don't hesitate to ping me if you run into any unsolvable issues.

closed 2024-08-16 by gitbls