gitbls/sdm #251: apt returned an error; review /etc/sdm/apt.log --> /etc/sdm/apt.log does not exist

rvaneerd, 2024-08-15:
Hi,

I'm experimenting with the new syncthing plugin but I'm running into errors:
`* Plugin syncthing: Start Phase 1
> Plugin syncthing: Add syncthing release 'stable' to apt sources
> Plugin syncthing: Perform update and install syncthing
? apt returned an error; review /etc/sdm/apt.log
? Plugin 'syncthing' exited with failure status '1'
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
debian@debian:~$ sudo cat /etc/sdm/apt.log
cat: /etc/sdm/apt.log: No such file or directory
`
How to check what is going on if the referenced file does not / no longer exist ?

Regards,
Rob.

gitbls, 2024-08-15:
Please do the following:
```
sudo sdm --explore /path/to/img.img
# Then in the img container you can
cd /etc/sdm
less apt.log
less history
# When done
exit
```
If you can't sort out what's going on, please provide those files (redacted as necessary for any passwords, etc). You can copy them out of the IMG with `scp`, or you can go back into the IMG as a mounted disk and copy them locally:
```
sudo sdm --mount /path/to/img.img
cd $SDMPT/etc/sdm
cp apt.log /path/to/somewhere/
cp history /path/to/somewhere/
# When done
exit
```
Then either post them here or email me if that's easier (email on sdm homepage).

rvaneerd, 2024-08-15:
Hi @gitbls ,

Thanks a lot for your swift response again!
It was a bit confusing but now I understand the log file is stored **within the image** ;-)

From the apt.log, I retrieved the following:
`2024-08-15 19:24:41 apt-get -qq install --no-install-recommends --yes syncthing

2024-08-15 19:24:44 /bin/sh: 1: /usr/sbin/dpkg-preconfigure: Bad message
2024-08-15 19:24:45 dpkg: warning: 'ldconfig' not found in PATH or not executable
2024-08-15 19:24:45 dpkg: error: 1 expected program not found in PATH or not executable
2024-08-15 19:24:45 Note: root's PATH should usually contain /usr/local/sbin, /usr/sbin and /sbin
2024-08-15 19:24:45 E: Sub-process /usr/bin/dpkg returned an error code (2)
2024-08-15 19:24:45 [Done]
`
The history file you mentioned appears to be a binary file, so peeking into it using "less" is not helpfull.

Regards,
Rob.

gitbls, 2024-08-15:
So you need to install some packages to make apt work.  You can install `ldconfig` with `sudo apt install libc-bin`, but if it fails because of ldconfig not being there, well, that's a problem for you to sort out. 

On my Debian 12 system, I see that ldconfig is actually there, so not sure what you did or didn't do to end up with this.

The history is most definitely a text file viewable with less. If it's not, then something is terribly wrong with your configuration that you need to sort out.

rvaneerd, 2024-08-15:
Thanks @gitbls, I need to get some things sorted out.
As root, "ldconfig" seems to be available and "libc-bin" is already at the newest version.

I started off with a UTM-provided Debian but will have to revert to a clean install of an image downloaded from Debian.
That will also give me the opportunity to do a "lite" install as I do not need al the fancy Debian stuff ;-)

Therefore, it will take some time before I get back on this.

Regards,
Rob.

gitbls, 2024-08-15:
> Thanks @gitbls, I need to get some things sorted out. As root, "ldconfig" seems to be available and "libc-bin" is already at the newest version.
>
> I started off with a UTM-provided Debian but will [...]

sdm runs as root, so unclear to me why it would not be available in sdm but is when you're root at the console.

Sorry you're having issues, but sdm can only solve problems that it knows about, and this is absolutely the first time I've seen this particular issue.

gitbls, 2024-08-15:
What is the original, downloaded filename of the IMG you're trying to customize?

rvaneerd, 2024-08-16:
No worries @gitbls,
I'm using the following image: 2024-07-04-raspios-bookworm-armhf-lite.img

In the meantime, I found a laptop lying around / catching dust and I installed Debian 12 on it.
At the moment, I'm in the progress of replicating my previous stuff and the image mentioned to this new hardware (so no more virtualization and better control).

I'll keep you posted on the progress.

Regards,
Rob.

rvaneerd, 2024-08-16:
Hi @gitbls,

I'm not finished yet and have not tested my syncthing installation but I've gotten further (on my newly configured hardware):
`> Run Plugin 'syncthing' (/usr/local/sdm/plugins/syncthing) Phase 1 with arguments:`
`  'enablesvc|gui-address=0.0.0.0:8384|gui-password=xxxxxxx|gui-user=rob|synchost=mysyncserver'`
`* Plugin syncthing: Start Phase 1`
`> Plugin syncthing: Add syncthing release 'stable' to apt sources`
`> Plugin syncthing: Perform update and install syncthing`
`* Plugin syncthing: Complete Phase 1`

So as far as I'm concerned, this issue can be closed.

As you mentioned, it is not ment to debug all "environment related issues".

Regards and thanks for your continued support.

Regards,
Rob.

closed 2024-08-16 by gitbls

rvaneerd, 2024-08-16:
Final update: My sdm configuration completed successfully (although the old laptop made it take quite some time ;-)).
To confirm and probably as expected, the history file is ASCII text and hence readable.