gitbls/sdm #253: Additional configuration of (user-specific) ssh

rvaneerd, 2024-08-17:
Hi there,

After building an image and loading it into a RPi, I know, the host related ssh key configuration is (re-)generated.

However, for the user(s) added to the RPi image, once up and running, the ssh configuration still needs some manual tasks:
- Generate a user specific ssh key: `ssh-keygen -t rsa`
- Copy the/a workstation public key (`MacBookPro_id_rsa.pub`) to the RPi's user specific `.ssh` folder
- Add that public key to the  `authorized_keys` file: `cat MacBookPro_id_rsa.pub >>authorized_keys`
- Change ownership of the `authorized_keys` file: `chmod 600 authorized_keys`

Is that somehow **already** possible?

Regards,
Rob.

gitbls, 2024-08-17:
It's not clear to me why you would want to generate new user keys for each host. That seems rather inconvenient to me, since you then need to propagate that key around to the authorized_keys file on other systems where you might want to use it.

I note that you bolded the word **already**. Since everyone likes to do certain things their own way, that's not really feasible. 

What is **already** available in sdm is the ability to write your own plugin to do EXACTLY what you want, if there is no plugin available. 99.999% of sdm is well-documented.

If you haven't skimmed through the [Plugins documentation](https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md) and the various Examples, especially [Example Plugin](https://github.com/gitbls/sdm/blob/master/Docs/Example-Plugin.md) you really should, as it will help you understand what sdm **already** does in the provided plugins vs things that you will need your own plugin to accomplish.

Back on SSH, I have a single, well-protected key that I use for SSH. It is kept in a directory thusly:
```
bls@mondo/l/work/sdm/plugins> ls -l /rpi/home/bls/ssh-mydomain/
total 20
-rw------- 1 bls users  85 Feb 16  2022 authorized_keys
-rw------- 1 bls users 893 Jun 21 20:13 config
-rw------- 1 bls users 444 Feb 16  2022 idbls
-rw-r--r-- 1 bls users  85 Feb 16  2022 idbls.pub
-rw------- 1 bls users 879 Jun 19 16:53 known_hosts
```
Then I copy that directory to my IMG during customize. See the above-mentioned Example Plugin for the two lines of code following `logtoboth "> Plugin $pfx: Copy $csrc/home/bls/ssh-mydomain to $SDMPT/home/$myuser"`. FYI $csrc is a holdover from a long time ago. You can certainly use a full path such as /mydisk/mydir/ssh-whatever 

This plugin example is _exactly_ my personal plugin and I include it in literally EVERY IMG that I customize so that EVERYTHING I want configured for me personally is perfectly configured. Every `bls` user on every system has the exact same SSH configuration and keys. This means that I can SSH from any system to any other system quite easily. I use the app `keychain` which is a simple interface to `ssh-agent`. See https://github.com/gitbls/rxapp for further information on this.

If you really want to have different SSH keys for each user on each host, you can add the appropriate code to your personal plugin to do what you've described (generate key, copy public key, update authorized_keys). You will, however, need to propagate those keys to your other hosts appropriately, and that is completely outside the scope of any sdm built-in capabilities.

rvaneerd, 2024-08-18:
Thanks for your elaborate answer!
I might have to dig in further.
My way of using ssh may not be the best one.

gitbls, 2024-08-18:
> Thanks for your elaborate answer! I might have to dig in further. My way of using ssh may not be the best one.

🤷‍♂️Feel free to ask further questions, of course!

gitbls, 2024-09-03:
Closing this issue due to lack of activity. @rvaneerd please feel free to reopen if required. Thx!

closed 2024-09-03 by gitbls