gitbls/sdm #255: Configure gsm/lte modem interface to start on boot

thk686, 2024-09-19:
I am trying to get NetworkManager to bring up a modem when the pi boots. I have a script that will configure and bring up the interface if I run it after logging in. But I can't get it to come up automatically. I have tried it as a oneshot service and in 0piboot. Once I manually run the script, the interface does come up automatically on reboot.

```
#!/bin/bash

# Currently, this script is installed in /etc/sdm/0piboot

# Path to the configuration file
CONFIG_FILE="/etc/biosense/setup-modem.conf"

# Function to read APN from the configuration file
read_apn() {
    local apn=""
    if [ -f "$CONFIG_FILE" ]; then
        apn=$(grep -i 'apn' "$CONFIG_FILE" | awk -F'=' '{print $2}' | xargs) # Extract APN value
    fi
    echo "$apn"
}

# Detect the modem interface (e.g., wwan0)
modem_iface=$(nmcli dev status | grep gsm | awk '{print $1}')

# Read APN from the configuration file
APN=$(read_apn)

if [ -n "$modem_iface" ]; then
    if [ -n "$APN" ]; then
        nmcli c add type gsm ifname "$modem_iface" con-name lte connection.interface-name "$modem_iface" gsm.apn "$APN" ipv4.method auto &&
            echo "Added LTE modem connection with APN: $APN" ||
            echo "Failed to add LTE modem connection with APN: $APN"
    else
        nmcli c add type gsm ifname "$modem_iface" con-name lte connection.interface-name "$modem_iface" ipv4.method auto &&
            echo "Added LTE modem connection without APN" ||
            echo "Failed to add LTE modem connection without APN"
    fi

    nmcli c modify lte connection.autoconnect yes &&
        echo "Set LTE connection to autoconnect" ||
        echo "Failed to set LTE connection to autoconnect"
else
    echo "No modem network interface found."
fi

```

gitbls, 2024-09-19:
In order to be run during sdm FirstBoot, scripts in /etc/sdm/0piboot need to be named `0*-*.sh`

Does your script conform to this naming?

If so, what do you find in the system journal as output from running the script?

thk686, 2024-09-19:
Naming is correct. I will look at the journal output. Thanks.

gitbls, 2024-10-05:
Closing since you have presumably resolved this. Pls re-open if more assistance needed. Thx.

closed 2024-10-05 by gitbls

thk686, 2024-10-05:
Yes we fixed. Network manager handles this well although we need to reboot once for some reason. Thanks for all your help with sdm.Sent from my iPhone