gitbls/sdm #260: Freeze autologin

MiroYld, 2024-10-10:
Hello,
I'm trying to enable autologin on my custom image I create with Packer
After doing 
```
sudo sdm --customize --autologin  mypi.img`
``
sdm blocks the 
```
at this level
IMG 'mypi.img' has 674540 1K-blocks (0.7GB, 0.6GiB) free at start of 'apt upgrade'
```
and I don't have any errors

MiroYld, 2024-10-10:
Thank you for your responsiveness. 
I have been using packer until now, for reasons of technical choice I decided to change technology, here is some information that may be useful to you. 
My conf packer basically only contains file copying and some script execution and it is based on an image from downloads.raspberrypi.org. [raspi-os-lite](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2024-03-15/2024-03-15-raspios-bookworm-arm64-lite.img.xz)
 Anyway, here I am facing the problem. After downloading the image
 ```
2024-07-04-raspios-bookworm-arm64-lite.img.xz
```
 I try the command again
``` 
sudo sdm --customize --autologin 2024-07-04-raspios-bookworm-arm64-lite.img
```
Here is the feedback I got, sdm a priori blocks the post-application installation/configuration of phase 1
[image]

Best Regards

gitbls, 2024-10-10:
Thank you for the additional details.

You're running sdm on an X64 system, so everything runs emulated with qemu, and everything takes longer. A LOT longer.

I just did the exact same command you did, on two different systems: A Pi5 and a 10th-Gen X64 Debian Bookworm system. 

Time on the Pi5: Less than one minute
Time on the X64: 7 minutes and 43 seconds

And OBTW, it failed to complete successfully due to insufficient disk space in the IMG to accommodate the updates. In /etc/sdm/history and on the console I see:
```
2024-10-10 15:11:03 % IMG '2024-07-04-raspios-bookworm-arm64-lite.img' is 100% full
2024-10-10 15:11:03 % Review /etc/sdm/apt.log in the IMG for insufficient disk space errors
2024-10-10 15:11:03 % If needed use --extend --xmb nnnn with --customize to create more space in the IMG
2024-10-10 15:11:03 % Customization terminated due to low disk space condition
2024-10-10 15:11:03 !!!
```
Use `--extend --xmb 2048` and apply an extra-large dose of patience and you should see a different result.

MiroYld, 2024-10-11:
hello, thank you for this information, indeed qemu takes a LOT of time and I also had the problem of disk space. 
I confirm that by adding 
```
--extend --xmb 2048
```
 I was able to get it to work correctly.
Thank you :)

closed 2024-10-11 by MiroYld