gitbls/sdm #261: Enable autologin + boot to console

mvsoom, 2024-10-15:
Hi,

This tool looks amazing, but a bit daunting for first use. I just have a quick question before I delve into this.

I am starting from the standard Raspberry Pi OS (64-bit, full desktop) because I cannot get Chromium to work with three.js accelerated code.

So starting from this image, I would like to:

1. Set hostname "raspberrypi.local"
2. Enable SSH
3. Copy locale from my local computer
4. Make a user "pi" with password "XYZ" (placeholder)
5. Enable autologin + boot to console
6. Execute a single script after first full boot to setup environment

I couldn't find how to do option 5. Could you help?

Thank you!

gitbls, 2024-10-15:
A few comments, and then the answer to your specific question:
* 2 - SSH is enabled by default. You don't need to do a thing
* 6 - Not completely clear what you mean by this, but this could also be done during the customize or burn rather than at boot time, unless there are constraints that completely preclude that.  Happy to offer suggestions if you provide more details.

As far as question 5, this should do it: `--plugin raspiconfig:"boot_behavior=B2"` or alternatively, if you're using a pluglist `raspiconfig:boot_behavior=B2`

Lastly, I'd suggest starting with ezsdm, which is available [here](https://github.com/gitbls/sdm/blob/master/ezsdm) and discussed [here](https://github.com/gitbls/sdm/blob/master/Docs/Example-Commands.md#official-getting-started-with-sdm-script), and uses a pluglist. This is SO much easier because you don't need to futz around with all those double quotes.

mvsoom, 2024-10-17:
Alright, I've skimmed the whole repo and docs and I understand better now. I've not managed to get my script working yet, but might post some additional questions after I'm back from holiday.