gitbls/sdm #262: Docker steps?

cmorganBE, 2024-10-18:
I like the idea of sdm but would like to run on osx and not run as root (or install to system folders) on the host OS. It would be great if there were notes on how to use this in docker.

gitbls, 2024-10-18:
I agree that it would be great to have a sdm run in a docker container. Unfortunately for you, I don't know Docker, have no interest in sorting it out, and I don't have a Mac.

I have made this offer before, but so far nobody has taken me up on it: `Provide a "sdm docker howto" that works straightaway, and I'll adopt it and include it with sdm.`

Until someone provides that, to use sdm you'll either need to sort this out yourself or use actual Raspberry Pi hardware for sdm. sdm runs great on the Pi, with no restrictions or Docker requirements.

techtasie, 2025-05-05:
@cjeanneret any updates on that front?

I wrote such a paragraph that your biggest problem would probably be that you don't have `qemu-user-static`. But you have since all this is doing is just wrapping an executable and this can still be done in containers (i.e. runc) and should in theory work also on MAC and Windows.

If nobody has taken you up on your offer @gitbls I will send you my Docker file tomorrow....