gitbls/sdm #265: Unable to rerun customize cmd with apps specified

mattie47, 2024-11-11:
Hi there,

I've been migrating our usage of SDM from v7 to the latest, and found we can't re-customize the same image multiple times when apps are specified.

We actually use a wrapper script on SDM, with a separate conf file specifying a list of apps to install (amongst all our other configuration). Bear in mind this comes from the days of many switches and so we created the wrapper script to parse a single conf file with desired config.

Anyway - so when we now attempt to customize or customize and then burn an image, the apps plugin dies with non zero due to:

```
? Plugin apps: Apps list 'rpi-apps' already exists; Add argument 'name=somename' to this apps 
  plugin invocation
  NOTE: This plugin is not idempotent; Restart with a fresh uncustomized IMG if this IMG is 
  previously customized
? Plugin 'apps' exited with failure status '1'
```

Can I ask why now allowing to recustomize with a list of the same apps ? Wouldn't the worst case be passing the list to apt and trying to install the same ones that have already been installed?

I would have thought even if this is not desirable, the ` --redo-customize`  may over-ride things ?

Why is this a problem:
* It means you need to know in advance if your pre-customized img has everything installed you're expecting in it. 
* You can't modify things later and add new apps to your list in your main configuration
  * I could be wrong here, but possibly if you've created a newer app list in a `@file` maybe it'll detect the newer version with `% sdm: Plugin 'user' on the host is newer` however in my case, I'm actually parsing a list of apps (given the historic switches) using a variable in a `sdm.conf` file (which contains all my overall configuration. 
* Alternatively, other modifications such as removing apps e.g. `--plugin apps:"remove=wolfram-engine"` at a later time also don't work due to the same above.

Thanks,
Matt

gitbls, 2024-11-11:
The apps plugin doesn't care at all about what apps you want to install. Rather, it cares about the `name=` parameter value, which is `default` if there is no `name=` parameter.

It is certainly feasible to un-restrict the apps plugin. HOWEVER, there is no guarantee that you won't run into this issue with other plugins, as being able to rerun a customize was never a design focus, and frankly, I'm not very interested in hearing "gee, thanks for fixing this plugin, can you please fix this next one as well" over and over.

I will look into updating the apps plugin to enable reruns with the same `name=` arg. At the same time, I'd encourage you to
* Consider whether you could apply a different `name=` value for subsequent customization runs on a customized IMG, eliminating the need for potential perturbation to the apps plugin
* If you are convinced that you need this, then verify that this is your only blocking issue WRT re-customization for the reason I mentioned above

As far as removing apps, you do realize that you can simply put a `name=` argument on the plugin invocation to eliminate that issue, don't you?

Lastly, I'm curious. How long does your customize run take, and why is it critically important to be able to do an update recustomize rather than starting over from a fresh IMG?

gitbls, 2024-11-14:
OK, I've updated the apps plugin so that it can handle multiple invocations with the same (`name=`) or with no name.

This will be in the next release coming soon.

mattie47, 2024-11-14:
Hi @gitbls 

Appreciate the response and quick turnaround.

>Consider whether you could apply a different name= value for subsequent customization runs on a customized IMG, eliminating the need for potential perturbation to the apps plugin

I worked around this by generating a unique name each time:
`--plugin apps:"name=$(dbus-uuidgen | cut -c1-8)|apps=$apps" `

>Lastly, I'm curious. How long does your customize run take, and why is it critically important to be able to do an update recustomize rather than starting over from a fresh IMG?

Currently customization can take anywhere from 7 minutes (with `--poptions noupgrade`) to 45 minutes :-( 

Although I'm averaging around 25-30 min. So it's been quite painful each time I work out what isn't working in my image to recustomize, and reburn. 

This is on 2024-10-22-raspios-bookworm-armhf.img 

Why is it taking so long? 90-95% of time is on the apt upgrade and it gets mostly stuck on:

```
Processing triggers for initramfs-tools (0.142+rpt1+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v8
W: Couldn't identify type of root file system for fsck hook
I: The initramfs will attempt to resume from /dev/sdb
I: Set the RESUME variable to override this.
'/boot/initrd.img-6.6.51+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v7l
W: Couldn't identify type of root file system for fsck hook
I: The initramfs will attempt to resume from /dev/sdb
I: Set the RESUME variable to override this.
'/boot/initrd.img-6.6.51+rpt-rpi-v7l' -> '/boot/firmware/initramfs7l'########################################################################################..]
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v7

W: Couldn't identify type of root file system for fsck hook##################################################################################################..]
I: The initramfs will attempt to resume from /dev/sdb
I: Set the RESUME variable to override this.
'/boot/initrd.img-6.6.51+rpt-rpi-v7' -> '/boot/firmware/initramfs7'
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v6
W: Couldn't identify type of root file system for fsck hook
I: The initramfs will attempt to resume from /dev/sdb
I: Set the RESUME variable to override this.#################################################################################################################..]
'/boot/initrd.img-6.6.51+rpt-rpi-v6' -> '/boot/firmware/initramfs'
Processing triggers for libvlc-bin:armhf (1:3.0.21-0+rpt3+deb12u1) ...
Warning: using insecure memory!
```

This is on a clean WSL Ubuntu install using nspawn (although --chroot is also the same in speed) and SDM 13.1. 

To workaround this, I'm now looking at:
* Download Vanilla image
* --Explore and only update/upgrade
* Use finished image as starting point for `customize`. 

**I did have a couple follow up questions:**

- What's the difference between systemd-nspawn and chroot from an SDM POV?

From the docs, I see chroot is considered simpler, but beyond that I wasn't too sure...

I can use either on WSL, however I was separately working on a Docker configuration where I then need to use chroot instead.

- Has SDM changed how it passes files into the image in a custom-phase script during phase 0 over the last year? 

Previously I would do something like:

```
if [ $phase == "0" ]; then
    #
    # In Phase 0 all references to directories in the image must be preceded by $SDMPT
    #
    logtoboth "* $pfx Phase 0"
    logfreespace "at start of $pfx Phase 0"

    # Check if env variable $SDMPT exists
    [ "$SDMPT" == "" ] && echo "? SDMPT is undefined in custom-phase.sh ; aborting" && return

    # Copy assets folder over
    logtoboth "* $pfx Copying asset files..."
    if [ -e "$SDMPT/home/pi/assets" ]; then
        logtoboth "* $pfx Found old assets folder in image, deleting..."
        rm -r "$SDMPT/home/pi/assets"
    fi
    cp -r "$ASSETS_LOCATION" "$SDMPT/home/pi/assets"

```
When I either `--explore` or look at the burned image, any files copied in the phase 0 period have user:group of root:root. 

Again I could be wrong, but in earlier versions of SDM, my output image would end up having the correct user's (pi:pi in my case) permissions once copied into the user's dir. 

(I'm not complaining about the above - just trying to understand as I migrate my script[s]). 

I will look at switching to `copydir` plugin instead

mattie47, 2024-11-14:
hmm...

I tried the copy dir plugin, with rsync flags -rpit, but still the copied content ends up being owned by root, despite the host's permissions being `matt:matt`. Am I being an idiot here, or just misunderstanding something?  

```
matt@host:~/work/new$ sudo sdm --customize \
--redo-customize \
--poptions noupdate,noupgrade,noautoremove \
--plugin copydir:"from=./vhs-files \
|to=/home/pi/test/|rsyncopts=-rpit" \
./images/2024-10-22-raspios-bookworm-armhf.img
* Mount IMG './images/2024-10-22-raspios-bookworm-armhf.img'
mount: /dev/loop3 mounted on /mnt/sdm.94983.
mount: /dev/loop8 mounted on /mnt/sdm.94983/boot/firmware.
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --redo-customize --poptions
  noupdate,noupgrade,noautoremove --plugin copydir:from=./vhs-files
  |to=/home/pi/test/|rsyncopts=-rpit ./images/2024-10-22-raspios-bookworm-armhf.img
* Host Information
   Hostname:  host
   Memory:    16236616 kB
   uname:     Linux <>5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
   os-release Name: Ubuntu 24.04.1 LTS
   Version: 24.04.1 LTS (Noble Numbat) (noble)
   Like OS: debian
   sdm Version: V13.1
* IMG Information
   Name: ./images/2024-10-22-raspios-bookworm-armhf.img
   Date: 2024-10-22
   RasPiOS Version: 12
   RasPiOS Architecture: 32-bit armhf
   os-release Version: 12 (bookworm)
% sdm will use qemu 'arm'
% sdm will use systemd-nspawn on this 64-bit x86-64 host
  Retry the command with --chroot if this fails
* Plugins selected:
   * copydir
       Args: from=./vhs-files |to=/home/pi/test/|rsyncopts=-rpit
> IMG './images/2024-10-22-raspios-bookworm-armhf.img' has 2583032 1K-blocks (2.6GB, 2.5GiB) free
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'copydir' (/mnt/sdm.94983/usr/local/sdm/plugins/copydir) Phase 0 with arguments:
  'from=./vhs-files |to=/home/pi/test/|rsyncopts=-rpit'
* Plugin copydir: Start Phase 0
> Plugin copydir: Keys/values found:
   from: ./vhs-files
   to: /home/pi/test/
   rsyncopts: -rpit
> Plugin copydir: rsync  -rpit ./vhs-files /mnt/sdm.94983//home/pi/test/
* Plugin copydir: Complete Phase 0
* Phase 0 Completed
* Enter image './images/2024-10-22-raspios-bookworm-armhf.img' for Phase 1
* Start Phase 1 image customization
> IMG './images/2024-10-22-raspios-bookworm-armhf.img' has 2567312 1K-blocks (2.6GB, 2.4GiB) free at start of Phase 1 image customization
> Configure and enable sdm FirstBoot service (sdm-firstboot)
> Skip 'apt update' per --apt-options noupdate
> Run Plugins Phase '1'
> Run Plugin 'copydir' (/usr/local/sdm/plugins/copydir) Phase 1 with arguments:
  'from=./vhs-files |to=/home/pi/test/|rsyncopts=-rpit'
* Plugin copydir: Start Phase 1
* Plugin copydir: Complete Phase 1
* Phase 1 post-app installation/configuration
> Skip 'apt upgrade' per --apt-options noupgrade
> Skip 'apt autoremove' per --apt-options noautoremove
> IMG './images/2024-10-22-raspios-bookworm-armhf.img' has 2567312 1K-blocks (2.6GB, 2.4GiB) free at end of Phase 1 image customization
* Phase 1 Completed
> Run Plugins Phase 'post-install'
> Run Plugin 'copydir' (/usr/local/sdm/plugins/copydir) Phase post-install with arguments:
  'from=./vhs-files |to=/home/pi/test/|rsyncopts=-rpit'
* Plugin copydir: Start Phase post-install
* Plugin copydir: Complete Phase post-install
> Run graphics post-install configuration
> Plugin sdm-phase1: Found Display Manager lightdm
> Plugin sdm-phase1: Display Manager lightdm will be disabled when the system first boots
% Plugin sdm-phase1: Desktop Autologin setting 'B3' will be set after sdm FirstBoot
> Plugin phase1: Configure SSH
> Plugin phase1: Enable SSH service
> IMG './images/2024-10-22-raspios-bookworm-armhf.img' has 2567312 1K-blocks (2.6GB, 2.4GiB) free at end of image customization
> Customize elapsed time: 00:00:07
* Enter Shell Command Prompt
  'exit' to exit back to host system
root@sdm:/# cd /home/pi
root@sdm:/home/pi# ls -lha
total 36K
drwx------ 6 pi   pi   4.0K Nov 14 19:25 .
drwxr-xr-x 4 root root 4.0K Oct 23 01:59 ..
-rw-r--r-- 1 pi   pi    220 Oct 23 01:47 .bash_logout
-rw-r--r-- 1 pi   pi   3.5K Oct 23 01:47 .bashrc
drwxr-xr-x 3 pi   pi   4.0K Oct 23 01:51 .local
-rw-r--r-- 1 pi   pi    807 Oct 23 01:47 .profile
drwxr-xr-x 2 pi   pi   4.0K Oct 23 01:51 Bookshelf
drwxr-xr-x 3 root root 4.0K Nov 14 19:25 test <--------------
drwxr-xr-x 6 root root 4.0K Nov 13 18:19 vhs-files <----------------
root@sdm:/home/pi# cd test
root@sdm:/home/pi/test# ls -lah
total 12K
drwxr-xr-x 3 root root 4.0K Nov 14 19:25 .
drwx------ 6 pi   pi   4.0K Nov 14 19:25 ..
drwxr-xr-x 6 root root 4.0K Nov 13 18:19 vhs-files <----------
root@sdm:/home/pi/test#

exit
* Customization complete
umount: /mnt/sdm.94983/boot/firmware unmounted
umount: /mnt/sdm.94983 unmounted
matt@host:~/work/new$ ls -lah
total 92K
drwxr-xr-x 6 matt matt 4.0K Nov 13 20:58 .
drwxr-xr-x 4 matt matt 4.0K Nov 13 18:17 ..
drwxr-xr-x 8 matt matt 4.0K Nov 14 19:18 .git
-rw-r--r-- 1 matt matt   48 Nov 13 18:17 .gitattributes
-rw-r--r-- 1 matt matt   75 Nov 13 18:17 .gitignore
-rw-r--r-- 1 matt matt 2.3K Nov 13 20:51 Dockerfile
-rw-r--r-- 1 matt matt  11K Nov 13 18:17 README.md
-rwxr-xr-x 1 matt matt  20K Nov 14 19:19 build.sh
-rw-r--r-- 1 matt matt  989 Nov 14 19:19 compose.yaml
-rw-r--r-- 1 matt matt 2.0K Nov 13 18:17 config.conf
drwxr-xr-x 2 matt matt 4.0K Nov 14 19:20 images
drwxr-xr-x 4 matt matt 4.0K Nov 13 18:17 sdm
-rwxr-xr-x 1 matt matt  14K Nov 13 18:17 vhs-custom-phase.sh
drwxr-xr-x 6 matt matt 4.0K Nov 13 18:19 vhs-files
matt@host:~/work/new$ cd vhs-files/
matt@host:~/work/new/vhs-files$ ll
total 24
drwxr-xr-x 6 matt matt 4096 Nov 13 18:19 ./
drwxr-xr-x 6 matt matt 4096 Nov 13 20:58 ../
drwxr-xr-x 2 matt matt 4096 Nov 13 18:17 assets/
drwxr-xr-x 4 matt matt 4096 Nov 13 18:17 configs/
drwxr-xr-x 8 matt matt 4096 Nov 13 20:55 repos/
drwxr-xr-x 4 matt matt 4096 Nov 13 18:17 tools/
```

Appreciate any input.

Thanks again,
Matt

gitbls, 2024-11-14:
OK, lots of stuff in your two posts, so here are comments and answers.

**Running sdm on WSL** X64 is certainly possible, as you've discovered, but a) it's painfully slow, and b) you can't burn disks from within WSL, at least I haven't been able to make that happen.

Why is sdm slow on WSL? EVERYTHING is emulated with qemu. So, every ARM instruction takes god-knows-how-many actual x64 processor instructions to execute.

Personally, I would NEVER use that in a production environment, especially since Pi's are so cheap, relatively speaking, and I'm extremely impatient (one of the reasons I built sdm TBH 🤣).

And a Pi5 is ***MUCH*** faster than an emulated NUC-10. I just did a test: WSL customize: 25 minutes, Pi5 same customize: 2 minutes and 38 seconds. 😵 How's THAT for inspiration?

**Re initramfs:** I've never seen this issue, have never seen the RESUME message, etc.. Without having a close look at what you're actually installing, no clue what's going on. update-initramfs does take a while on a Pi (and hence a *super-long* while on WSL), but you may have some other issue going on there.

**Diff between chroot and systemd-nspawn:** From an sdm pov there's no difference. They are both forms of containers. I personally like systemd-nspawn bc it's more complete and robust, but they should work the same for sdm. Note that sdm uses systemd-nspawn by default, and only falls back to chroot under certain situations with different architectures on host and IMG.

**Has SDM changed how it passes files into the image** in a custom-phase script during phase 0 over the last year? Not that I can recall. If you want the files to have the same owner as the source files, you should use `cp -a`. However, I personally prefer to explicitly set file owner/prot to ensure they're consistent with the accounts defined in the IMG. The sdm function `setfileownmode` can be used to set the owner and protection of files in the IMG in phase 1 or post-install. Or, you can use the `modattr` plugin if you prefer.

**Re copydir plugin:** The copydir plugin doesn't do anything about file ownership. What is the result if you also add `-o` to the rsync switches to preserve file ownership? I don't use copydir. I copy files using my personal plugin with `cp -a` and set the file ownership explicitly there when needed.

gitbls, 2024-11-16:
V13.2 is posted with the updated `apps` plugin. Please try it out. Thx

mattie47, 2024-11-18:
Thanks. Sorry - I will test this at some point this week.

mattie47, 2024-11-27:
Confirmed working as desired. Thanks for doing this 👍

Will close issue

closed 2024-11-27 by mattie47