gitbls/sdm #266: FYI: RPI 5 discrepancies from rootfs disk encryption

petterroea, 2024-11-15:
Writing this as a FYI. I followed this guide: https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md with the latest "rasbian with all recommended application" image. Used ubuntu 24.04LTS to modify the image.

The following happened:
* First time, rasbian booted as if nothing happened. No reboot.
* Second time, sdm-auto-encrypt ran as expected
* Third boot: No /dev/mapper errors. Otherwise as expected
* Fourth reboot: blank screen. Waited 20 minutes and restarted the pi. 
* Fifth(extra!) reboot: stuck at raspbian boot splash. Entered luks password, after which the pi booted.

gitbls, 2024-11-15:
Sorry you had a problem. Unfortunately, you didn't provide any useful information to enable me to help you if you're still having problems, or to investigate what might have happened.

Please provide the following information if you want to engage on this
* Did it ultimately work correctly for you on subsequent reboots? Is the rootfs encrypted as you expect?
* There is no software named 'rasbian'. Did you mean Raspberry Pi OS Bookworm version?
* And if it's RasPiOS Bookworm, was it the Lite version or one with a graphical desktop?
* What exactly do you mean by "Used ubuntu 24.04LTS to modify the image"? If you mean using sdm to customize the IMG before you burned it, please provide the file `/etc/sdm/history` from the disk that was on
* How did you burn the IMG to whatever disk you're using? Please provide the complete command line
* If you did use sdm to customize the img, did you add the `--reboot 10` switch either on the customize or when you burned the disk?
* What exactly does "Otherwise as expected" mean? Exactly what happened? What commands did you enter?
* On the free 5th reboot, Please clarify what you mean by `raspbian boot splash` so I understand precisely what you mean
* How did you enter the luks password if it was stuck at a boot splash screen?

Thanks! I'm always interested in helping people succeed with sdm, but there are a LOT of moving parts so really need complete and specific details.

petterroea, 2024-11-16:
Hello! Thank you for your quick reply and sorry for my lack of details. Some of the info you are requesting is on a machine I am currently not close to, I will get hold of the info for you on Tuesday. Cheers

gitbls, 2024-12-05:
Closing due to inactivity. OP: if desired, please re-open to provide details as requested. Thx!

closed 2024-12-05 by gitbls