gitbls/sdm #267: Wayland Wayfire usage with SDM

mattie47, 2024-11-18:
Hi Benn,

I can see SDM supports X11 or Wayland, however I'm confused how Wayfire with Wayland backend may get set?

In raspi-config, we have:

```
W1 X11     Openbox window manager with X11 backend
W2 Wayfire Wayfire window manager with Wayland backend
W3 Labwc   Labwc window manager with Wayland backend  - Bookworm Default
```

This is set in raspi-config here: https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L2520 

I can't see where in SDM Wayfire may specifically get set?

As per the plugin docs, I can see  we can figure `wayfire.ini` and `wf-panel-pi` via the lxde plugin:

https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#lxde

The graphics plugin: https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#graphics states:
```
wayland is the Default graphics subsystem on Bookworm with Desktop images, so graphics=wayland is ignored on those images. 
```

However, how does SDM actually then set configuration to W2?  Or does it possibly not at the moment? I'm assuming not given W1 seems to be the only option in graphics plugin - https://github.com/gitbls/sdm/blob/d61c6fbc3faf00d2b77dca3fa7ffb1fc7e07d942/plugins/graphics#L211

Thanks,

Matt

gitbls, 2024-11-18:
As I understand it, Wayfire is going away in RasPiOS in favor of labwc.

As you have discovered, there's no way to set wayfire with sdm, and in fact, you just pointed out an oops...W1 doesn't mean what it used to when wayland support got added.

Please clarify why you're looking at wayfire given that Pi central has pointed the way to labwc.

gitbls, 2024-11-19:
See this re RasPiOS directions wayfire vs labwc: https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/

mattie47, 2024-11-19:
Hi @gitbls 

Thanks for that feedback, and also pointing out the comments about wayfire!

I've been in the process of migrating our internal testing tool device from Bullseye to buster, and encountered a number of issues as a result.

As you can see from the other issues, I'm also upgrading from historic SDM to the latest as well. 

**Why Wayfire?**

I spent a day trying to get our physical power off button working (with its own circuitry as part of a custom built RPI HAT). Wayfire was the only config I could get it to work without pressing the latching push button twice ([as per docs](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/power-button.adoc)). 

I got this working on Wayfire, and at that point hadn't seen the article you'd shared.

I then tried looking at labwc where I then encountered:
* Issues with our touchscreen
* Power button functionality not working
* A need to rewrite our custom config again...

So had thought "okay wayfire it is then..". But now as you've rightly pointed out, it's going away, and I want long term support, so will have a look at labwc instead.

**Why not X11, especially if you have your config working??** 

I'd read a couple posts on the RPI forums which didn't fill me with confidence around using/supporting X long term. 
* https://forums.raspberrypi.com/viewtopic.php?p=2244619&sid=f1d733a492361e4f03be52ddc4bd55e9#p2244619 
* https://forums.raspberrypi.com/viewtopic.php?p=2269756#p2269756

Perhaps I should be sticking with X for now, but will do a brief investigation further into whether labwc may be much effort...

gitbls, 2024-12-05:
OK, after all that I'm updating the `graphics` plugin to support 3 values for the `graphics` arg: `labwc`, `wayfire`, and `x11`

All they will do is make the appropriate raspi-config W1/W2/W3 call. For wayfire or labwc, the appropriate bits must already be installed, so only useful on "with desktop" versions.

gitbls, 2025-01-04:
Closing due to no activity. OP please re-open if necessary.