gitbls/sdm #268: Pi5 running SDM: "Too many levels of symbolic links"

mattie47, 2024-11-19:
Hi @gitbls,

I decided to follow your suggestion of using a pi5 to build an image, but I can't seem to get SDM working on a fresh install of a Pi5. 

Instead, I get the following whenever it attempts to mount the image:
```
pi@rpi5:~ $ sudo sdm --customize 2024-10-22-raspios-bookworm-armhf.img
* Mount IMG '2024-10-22-raspios-bookworm-armhf.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
/usr/local/sdm/sdm-cparse: line 560: /usr/bin/date: Too many levels of symbolic links
/usr/local/bin/sdm: line 817: /usr/bin/mkdir: Too many levels of symbolic links
/usr/local/bin/sdm: line 818: /usr/bin/touch: Too many levels of symbolic links
/usr/local/bin/sdm: line 819: /usr/bin/chmod: Too many levels of symbolic links
/usr/local/bin/sdm: line 820: /mnt/sdm/etc/sdm/auto-1piboot.conf: No such file or directory
/usr/local/bin/sdm: line 821: /usr/bin/mkdir: Too many levels of symbolic links
/usr/local/bin/sdm: line 824: /usr/bin/cp: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 205: /usr/bin/chown: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 206: /usr/bin/chmod: Too many levels of symbolic links
/usr/local/bin/sdm: line 824: /usr/bin/cp: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 205: /usr/bin/chown: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 206: /usr/bin/chmod: Too many levels of symbolic links
/usr/local/bin/sdm: line 824: /usr/bin/cp: Too many levels of symbolic links
```

And trying explore (yes, different vanilla image, I tried downloading the two latest ones):

```
pi@rpi5:~ $ sudo  sdm --explore 2024-11-13-raspios-bookworm-armhf.img
* Mount IMG '2024-11-13-raspios-bookworm-armhf.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
% sdm will use qemu 'arm'
% Add and enable qemu binfmt for 'arm' processor architecture
* Enter IMG '2024-11-13-raspios-bookworm-armhf.img'
/usr/local/sdm/sdm-cparse: line 2067: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2067: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2067: /usr/bin/stty: Too many levels of symbolic links
```

This leaves the Pi in a bad state, and I physically have to hotswap power to the Pi to get out of this state.

**Reproduction steps**
* Flash SD card using RPI Imager with Raspi OS 32bit (Release 2024-11-13)
  * I also set things like local, hostname, WiFi etc. 
* Install SDM
  *  `curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash`
* Download latest Pi OS
  * `curl -c -O https://downloads.raspberrypi.org//raspios_armhf/images/raspios_armhf-2024-11-19/2024-11-19-raspios-bookworm-armhf.img.xz`
* Extract image
  * ` xz -v -d 2024-11-19-raspios-bookworm-armhf.img.xz`
* Attempt to customize or explore image above
  * Issue occurs 

**Log of results from above reproduction**
(Note, Apt failed to install a couple things from the SDM install, however I ran this manually as per logs, prior to running SDM)

```
ssh pi@rpi5.local
Linux rpi5 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Nov 13 14:20:19 2024
[... 266 lines cut ...]
% Add and enable qemu binfmt for 'arm' processor architecture
* Enter IMG '2024-11-19-raspios-bookworm-armhf.img'
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 2069: /usr/bin/stty: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1216: /usr/bin/systemctl: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1227: /usr/bin/systemd-nspawn: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1900: /usr/bin/findmnt: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1902: /usr/bin/findmnt: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1696: /usr/bin/findmnt: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1696: /usr/bin/findmnt: Too many levels of symbolic links
/usr/local/sdm/sdm-cparse: line 1925: /usr/bin/sync: Too many levels of symbolic links

```

Is this something you've seen before?

I also tried going back to V13.0 (thinking perhaps a recent regression had occurred, but found no difference). 

Thanks,

Matt

gitbls, 2024-11-19:
Why are you running an armhf 32-bit OS on a Pi5? It's 64-bit hardware so run the full 64-bit OS.

That said, I was definitely able to repro your scenario running armhf on a Pi5 and just doing a `--explore` into a 32-bit OS. Impressive failure mode, to say the least!

The problem does NOT repro when exploring a 64-bit OS on that same 32-bit system.

My suggestion: Install the 64-bit OS. on your Pi5 If you need to operate on 32-bit IMGs, add `kernel=kernel8.img` to your /boot/firmware/config.txt

You'll be able to operate on both 32 and 64 bit IMGs on it. 

Someday I'll have a look into what's going on with 32 on 32 (at least) on a Pi5 (need to test others as well) but I consider this a low priority issue unless you can 100% convince me otherwise.

mattie47, 2024-11-19:
>Why are you running an armhf 32-bit OS on a Pi5? It's 64-bit hardware so run the full 64-bit OS.

Because the docs say it's supported:
>Running on 32-bit RasPiOS sdm can also perform all functions.

https://github.com/gitbls/sdm/blob/master/Docs/Compatibility.md?plain=1#L19

I opted for 32bit to stick with the same environment I'm running on actual Pis (In an attempt to minimize variables). 

Why 32bit for the clients? Hardware issues on x64 and I don't have time to address. Because I already have Pis running 32bit in production, I don't want a mix-match without any major benefit for our scenario. 

>My suggestion: Install the 64-bit OS. on your Pi5 If you need to operate on 32-bit IMGs, add kernel=kernel8.img to your /boot/firmware/config.txt

I see. Thanks for suggesting that.  I will try that now. 

>Someday I'll have a look into what's going on with 32 on 32 (at least) on a Pi5 (need to test others as well) but I consider this a low priority issue unless you can 100% convince me otherwise.

Fair enough. In absence, a doc update would be helpful. 

Thanks,

Matt

gitbls, 2024-11-19:
Now YOU know, and in the year since the Pi5 was released, you're the only person to run into this, so my motivation to doc it is pretty low. Sorry.

Also FYI, here is a strong recommendation from one of the RPL devs on the subject: https://forums.raspberrypi.com/viewtopic.php?p=2167385#p2167385

gitbls, 2025-01-08:
Noted in the Compatibility documentation.

closed 2025-01-08 by gitbls