gitbls/sdm #269: Network Plugin - Spaces in SSID

mattie47, 2024-11-20:
Hi,

Does the network plugin support spaces in the WiFI SSID? 

I've tried a number of things to get SDM to customize an image using things such as:

```
--plugin network:"ifname=wlan0|cname=test|ctype=wifi|wifi-ssid=My\ SSID\ Here|wifi-password=pass|wificountry=NZ"
--plugin network:"ifname=wlan0|cname=test|ctype=wifi|wifi-ssid='My SSID Here'|wifi-password=pass|wificountry=NZ"
--plugin network:"ifname=wlan0|cname=test|ctype=wifi|wifi-ssid=My SSID Here|wifi-password=pass|wificountry=NZ" 
```

Each time however results in:
```
> Plugin network: Keys/values found:
   ifname: wlan0
   cname: test
   ctype: wifi
   wifi-ssid: 'My SSID Here'
   wifi-password: pass
   wificountry: NZ
> Plugin network: Set WiFi Country 'NZ' for sdm FirstBoot
> Plugin network: Create connection 'test' on interface 'wlan0'
Error: invalid <setting>.<property> 'SSID'.
* Plugin network: Complete Phase post-install
```

Full command:
```
sudo sdm --customize \
--plugin network:"ifname=wlan0|cname=test|ctype=wifi|wifi-ssid='My SSID Here'|wifi-password=pass|wificountry=NZ" \
--redo-customize \
--poptions noupdate,noupgrade,noautoremove \
./images/2024-11-19-raspios-bookworm-armhf.img
```

I have also found a couple discrepancies with the network documentation.

1. SSH doesn't seem to be an allowed option for the Network plugin:

Docs example has SSH in the network plugin:
https://github.com/gitbls/sdm/blob/master/Docs/Example-Commands.md?plain=1#L139

Example attempting to run:
```
* Plugins selected:
   * network
       Args: ssh|ifname=wlan0|cname=test|ctype=wifi|wifi-ssid='My SSID Here'|wifi-password=pass|wificountry=NZ
> IMG './images/2024-11-19-raspios-bookworm-armhf.img' has 2263088 1K-blocks (2.3GB, 2.2GiB) free
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'network' (/mnt/sdm/usr/local/sdm/plugins/network) Phase 0 with arguments:
  'ssh|ifname=wlan0|cname=test|ctype=wifi|wifi-ssid='My SSID
  Here'|wifi-password=pass|wificountry=NZ'
? Plugin network: Unrecognized key 'ssh' in argument list
  'ssh|ifname=wlan0|cname=test|ctype=wifi|wifi-ssid='My SSID
  Here'|wifi-password=pass|wificountry=NZ'
? Plugin 'network' exited with failure status '1'
```

2. Network plugin won't run with just ssid/pass/country set, despite doc example suggesting otherwise. 

https://github.com/gitbls/sdm/blob/master/Docs/Example-Commands.md?plain=1#L139 

This sentence also gives this impression: https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md?plain=1#L686

```
* Plugins selected:
   * network
       Args: wifi-ssid='My SSID Here'|wifi-password=pass|wificountry=NZ
> IMG './images/2024-11-19-raspios-bookworm-armhf.img' has 2263084 1K-blocks (2.3GB, 2.2GiB) free
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'network' (/mnt/sdm/usr/local/sdm/plugins/network) Phase 0 with arguments:
  'wifi-ssid='My SSID Here'|wifi-password=pass|wificountry=NZ'
> Plugin network: Configuring 'network-manager' for network configuration management
? Plugin network: WiFi SSID incompatible with ethernet network device
? Plugin 'network' exited with failure status '1'
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```

Thanks

gitbls, 2024-11-21:
Thought that I had tested this and it was working, but that was a while ago...

We have been bomb cycloned and I'm without power and internet, and the prognosis doesn't look good at the moment. So, it may be a while...

mattie47, 2024-11-21:
>We have been bomb cycloned and I'm without power and internet, and the prognosis doesn't look good at the moment. So, it may be a while...

Sorry to hear. Look after yourself and your family.

These things I'm raising are absolutely not critical nor is there any expectation on timeframe for responding/addressing them. 

Given I've just been focussing on SDM recently, I wanted to raise things as I become aware of them and provide feedback from someone taking a close look at the tool and how it works. 

In terms of the networking plugin, I believe there's also an issue with SDM adding "quotes" to the SSID and Password in the generated nmconnection file which is not what NetworkManager appears to expect. 

This is quite easy to test by taking a running NetworkManager system, and modifying an existing connection's SSID or Password to be wrapped in quotes. I find once doing this (and systemctl restart NetworkManager), it's unable to connect again to the network.

----

I've worked around the issue I raised yesterday by instead using the nmconn option with configuration files instead. e.g.

```
--plugin network:"nmconn=/home/pi/test.nmconnection"
```

Worthy of note is this will nicely also overwrite the file if re-running the SDM customize command (and have the same file specified), unlike the `--plugin network:"ifname=wlan0|cname=test|ctype=wifi|wifi-ssid=My SSID Here|wifi-password=pass|wificountry=NZ" ` which will error out on detection of the same cname. 

And another useful tip - nmconnection files can also use encrypted passwords created using wpa_passphrase

```
wpa_passphrase "my ssid with spaces and no quotes" mysupersecretpassword
network={
        ssid="my ssid with spaces and no quotes"
        #psk="mysupersecretpassword"
        psk=c30d7e30f80a27ca38d64a77936d48dd08ea4692c2f4436b1d2fce691ad305d6
}

root@sdm-test:/home/pi# cat /etc/NetworkManager/system-connections/home.nmconnection
[connection]
id=home
uuid=a8a40336-021c-4b7c-9d8e-2ad77374318d
type=wifi
interface-name=wlan0

[wifi]
mode=infrastructure
ssid=my ssid with spaces and no quotes

[wifi-security]
key-mgmt=wpa-psk
psk=c30d7e30f80a27ca38d64a77936d48dd08ea4692c2f4436b1d2fce691ad305d6

[ipv4]
method=auto

[ipv6]
addr-gen-mode=default
method=auto

[proxy]
```

Thanks,

Matt

alliefitter, 2024-11-22:
@mattie47 Thanks for posting the workaround! I just ran into this, but creating a `.nmconnection` file got me fixed.

gitbls, 2024-12-02:
The next release will properly handle spaces, and a plethora of other special characters (current list:`  @#$%^&()-_=+{[}]|\:;'\"!" `) in the SSID and password. There will also be an option to encrypt the psk. I used the `hotspot` plugin to build one with strange characters, and was able to connect to it from another client device.

Thanks for pointing out the Docs issues. Will get them (and all the other issues you mentioned) fixed.  When in doubt, check the information in the Plugins document, as it is pretty much always correct (although I may miss catching an example every once in a while).

gitbls, 2024-12-07:
Fixed in V13.4

gitbls, 2025-01-04:
Closing due to no activity.

closed 2025-01-04 by gitbls