gitbls/sdm #272: error on network plugin

tim-oe, 2024-12-01:
sdm V13.3
using:  pi img
2024-11-19-raspios-bookworm-arm64-lite.img.xz

it was working without the error before v13.3 update
not seeing anything in plugin docs that changed
https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-11-19/2024-11-19-raspios-bookworm-arm64-lite.img.xz

getting the following error:

> Run Plugin 'network' (/mnt/sdm/usr/local/sdm/plugins/network) Phase 0 with arguments: 
  'netman=nm|wifissid=xxx-xxx|wifipassword=x xxx xxxx xxxxx|wificountry=US'
> Plugin network: Configuring 'network-manager' for network configuration management
? Plugin network: WiFi SSID incompatible with ethernet network device
? Plugin 'network' exited with failure status '1'

tim-oe, 2024-12-01:
i back reved the image to 2024-07-04-raspios-bookworm-arm64-lite.img
it now is prompting me for keyboard config.

> Task :sdmImage
* Mount IMG 'build/2024-07-04-raspios-bookworm-arm64-lite.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* Unmount IMG for --extend
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Extend IMG by 4096MB (4.3GB, 4.0GiB)...
* Resize partition 2 of 'build/2024-07-04-raspios-bookworm-arm64-lite.img' to 6589MB
* Remount IMG to resize the file system
* Mount IMG 'build/2024-07-04-raspios-bookworm-arm64-lite.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* Resize the file system
% (Ignore 'on-line resizing required' message)
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/loop0 is mounted on /mnt/sdm; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop0 is now 1608624 (4k) blocks long.

* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin user:adduser=tcronin --plugin 
  user:setpassword=tcronin|password=hsa&743 --plugin 
  mkdir:dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700 --plugin 
  copyfile:from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=
  postinstall|chown=tcronin:tcronin|chmod=600 --plugin 
  copyfile:from=src/home/.bash_aliases|to=/home/tcronin|runphase=postinstall|chown=tcronin:tcronin
  |chmod=644 --plugin user:adduser=ansible --plugin user:setpassword=ansible|password=ans&123 
  --plugin mkdir:dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700 --plugin 
  copyfile:from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=
  postinstall|chown=ansible:ansible|chmod=600 --plugin user:deluser=pi --plugin 
  copyfile:from=src/etc/profile.d/motd.sh|to=/etc/profile.d|runphase=postinstall|chown=root:root|c
  hmod=644 --plugin L10n:keymap=us|locale=en_US.UTF-8|timezone=America/Chicago --plugin 
  disables:triggerhappy|piwiz --plugin network:netman=nm|wifissid=tec-wan|wifipassword=i love 
  sarah olson|wificountry=US --extend --xmb 4096 --regen-ssh-host-keys --reboot 5 
  --nowait-timesync --redact build/2024-07-04-raspios-bookworm-arm64-lite.img
* Host Information
   Hostname:  tec-pi-mgr
   Memory:    8245584 kB
   uname:     Linux tec-pi-mgr 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
   Raspberry Pi Model: Raspberry Pi 5 Model B Rev 1.0
   os-release Name: Debian GNU/Linux 12 (bookworm)
   Version: 12 (bookworm)
   sdm Version: V13.3
* IMG Information
   Name: build/2024-07-04-raspios-bookworm-arm64-lite.img
   Date: 2024-07-04
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
* Plugins selected:
   * user
       Args: adduser=tcronin
   * user
       Args: setpassword=tcronin|password=hsa&743
   * mkdir
       Args: dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700
   * copyfile
       Args: from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinstall|chown=tcronin:tcronin|chmod=600
   * copyfile
       Args: from=src/home/.bash_aliases|to=/home/tcronin|runphase=postinstall|chown=tcronin:tcronin|chmod=644
   * user
       Args: adduser=ansible
   * user
       Args: setpassword=ansible|password=ans&123
   * mkdir
       Args: dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700
[... 149 lines cut ...]
rootfs: Inode 615 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 1407 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 59046 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: Inode 59113 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 59142 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: Inode 59617 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: Inode 61866 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: Inode 61867 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 61868 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: 61587/389600 files (0.2% non-contiguous), 458374/1608624 blocks
resize2fs 1.47.0 (5-Feb-2023)
> Shrink the root file system from 1608624 (6.6GB, 6.1GiB) to 611773 4096-byte blocks (2.5GB, 2.3GiB)
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/loop0 to 611773 (4k) blocks.
Begin pass 3 (max = 50)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 611773 (4k) blocks long.

% Ignore Warning about shrinking a partition may cause data loss

Warning: Shrinking a partition can cause data loss, are you sure you want to continue?
Yes/No? Yes                                                               
> Shrink the image by truncating to 3046887424 (3.0GB, 2.8GiB)
* Shrink complete
  IMG 'build/2024-07-04-raspios-bookworm-arm64-lite.img' was (7.1GB, 6.6GiB) now (3.0GB, 2.8GiB)

gitbls, 2024-12-01:
Sorry you're having issues. As you can see, I've edited your input to add code tags around it so it's legible for me (did not change any content). In the future, please use code tags around sdm logs so that they are legible. Thx!

Re the `network` plugin, it changed so that you MUST explicitly specify the device with the `ifname` argument. It is documented that the default `ifname` if not specified is `eth0`. This was done to simplify the description and make the plugin invocation clearly specify what network interface is being configured. Sorry about that. Add `ifname=wlan0` to the network plugin and that will fix the first error.

As far as the 2nd problem, I have some questions and comments:
* What do you mean by `i back reved the image to 2024-07-04-raspios-bookworm-arm64-lite.img`?
* Probably not related, but you should do the `user:deluser=pi` BEFORE you add a new user. Why? Once the user `pi` is deleted, the UID (1000) can be reused for the user you add.
* When you say `it is now prompting for keyboard config`, please clarify when that is occurring. I could guess, but I'd rather use explicit information from you.
* Since the sdm run exited with an error the IMG is not in a consistent and customization completed state, so this may be why there is a keyboard config problem (but see my prior question). An IMG customization that fails is NOT complete and is likely to fail when booted in strange and wondrous ways.
```
? Plugin network: WiFi SSID incompatible with ethernet network device
? Plugin 'network' exited with failure status '1'
```
* I note that your WiFi password has spaces in it. It has been reported that this doesn't work. I will be looking into this today.

tim-oe, 2024-12-01:
thanks for the quick response.

> What do you mean by i back reved the image to 2024-07-04-raspios-bookworm-arm64-lite.img?
i changed from the latest image from rp here: https://downloads.raspberrypi.com/raspios_lite_arm64/images/
i initially tried to use the 2024-11-19 image then revered backt to the 2024-07-04 image above

>When you say it is now prompting for keyboard config, please clarify when that is occurring. I could guess, but I'd rather use >explicit information from you.
after burning the sd card and plugging it into the pi and booting up. it brings up the keyboard config installation screen.

I will try it /w out the wifi settings and see if it gets farther.

gitbls, 2024-12-01:
> thanks for the quick response.
>
>
>
> I will try it /w out the wifi settings and see if it gets farther.

To be clear, if you look at the log you posted, the customize run terminated in Phase 0 due to the `network` plugin problem. This means that none of the plugins ran Phase 1 or post-install phases, which is where the work gets done for the `L10n` and `disables` plugins. So, yes, I would fully expect that it would boot up into the keyboard config screen, since sdm didn't have an opportunity to reconfigure the keyboard, in addition to literally ALL the plugins you've specified. None of them ran.

Fix the `network` plugin as I suggested (`ifname=wlan0`, assuming that it's the wlan0 device you want to configure the WiFi for), and things will work MUCH better 😉

tim-oe, 2024-12-01:
i changed the network plugin and removed wifi settings:
`--plugin network:"netman=nm"`

the image built and initialized successfully.
i'll wait to close this when i try the wifi pwd fix.

gitbls, 2024-12-01:
> i changed the network plugin and removed wifi settings: `--plugin network:"netman=nm"`
> 
> the image built and initialized successfully. i'll wait to close this when i try the wifi pwd fix.

On bookworm you don't need to do `--plugin network:netman=nm`. It's the default. If you don't specify any network configuration, there's nothing for the plugin to do!

gitbls, 2024-12-07:
Spaces in WiFi password issue corrected in V13.4

tim-oe, 2024-12-10:
verified working

closed 2024-12-10 by tim-oe