gitbls/sdm #273: Setting of WiFi operating country

mattie47, 2024-12-02:
Hi,

I could be wrong, but I think there might be an issue with setting the `cfg80211.ieee80211_regdom` country. 

If I try set the country during either `--customize` or `--burn`  e.g.

```
sudo umount /dev/sda1 && sudo umount /dev/sda2 && sudo sdm --burn /dev/sda --host sdm-test --expand-root --plugin network:"wificountry=NZ" images/2024-11-19-raspios-bookworm-armhf.img
```

I see the following:

```
 Plugin network: Keys/values found:
   wificountry: NZ
 Plugin network: Set WiFi Country 'NZ' for sdm FirstBoot
```

However once the RPI completes first boot, `/boot/firmware/cmdline.txt` is still showing US for country.

For reference, I'm only passing in wifi country, as nmconnection files containing WiFi config are being passed in separately during `--customize`. 

Here is a demo device's output cmdline config:

```
 cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=25eb0bda-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
```

On reflection, perhaps SDM hasn't been updated to support the Bookworm changes around this? I'll note that I don't have `cfg80211.ieee80211_regdo` in `/boot/cmdline.txt` on a Bullseye device. Meanwhile SDM correctly configured said Bullseye device (although back on SDM 7 or 8) country in `/etc/wpa_supplicant/wpa_supplicant.conf`

Thanks

Matt

mattie47, 2024-12-02:
Actually just to follow up, I did notice the following in the `history` file on the device:

```
2024-12-03 10:16:07 > Run Plugin 'network' (/usr/local/sdm/plugins/network) Phase post-install with arguments:
2024-12-03 10:16:07   'wificountry=NZ'
2024-12-03 10:16:07 * Plugin network: Start Phase post-install
2024-12-03 10:16:07 > Plugin network: Keys/values found:
2024-12-03 10:16:07    wificountry: NZ
2024-12-03 10:16:07 > Plugin network: Set WiFi Country 'NZ' for sdm FirstBoot
2024-12-03 10:16:08 * Plugin network: Complete Phase post-install
2024-12-03 10:16:08 * Burn Completed
2024-12-03 10:16:08 > Burn elapsed time: 00:08:01
2024-11-20 02:45:04 > sdm FirstBoot: Start sdm FirstBoot Service
2024-11-20 02:45:04 > sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
2024-11-20 02:45:04 > sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
2024-11-20 02:45:04 > sdm FirstBoot: Set WiFi Country to 'US'
2024-11-20 02:45:15 > sdm FirstBoot: Waiting 10/120 seconds for system time synchronization
2024-11-20 02:45:25 > sdm FirstBoot: Waiting 20/120 seconds for system time synchronization
2024-12-03 10:21:24 > sdm FirstBoot: System time synchronization achieved
```

gitbls, 2024-12-02:
Found and will be fixed in next release.

gitbls, 2024-12-07:
Fixed in V13.4

gitbls, 2024-12-14:
Actually, it's fixed completely in V13.5. Please test/verify. Thx!

gitbls, 2025-01-04:
Closing as completed.

closed 2025-01-04 by gitbls