gitbls/sdm #276: sdm-collect-labwc-config does not collect wf-panel-pi.ini

mattie47, 2024-12-04:
yes, despite the name `wf-panel-pi.ini` (implying wayfire), it seems under labwc this file is still used to set the task bar location.

Example of my file:

```
/.config $ cat wf-panel-pi.ini
[panel]
launcher_000001=lxde-x-www-browser.desktop
launcher_000002=pcmanfm.desktop
launcher_000003=lxterminal.desktop
position=bottom
icon_size=32
window-list_max_width=200
monitor=DSI-1
```

As soon as I save position to top, the taskbar moves to the top whilst using labwc.

You may want to look for and include this file during the collection script.

Thanks

Matt

gitbls, 2024-12-04:
I knew about wf-panel-pi.ini but somehow it disappeared on the path between my head and fingers.

gitbls, 2024-12-07:
Fixed in V13.4

gitbls, 2025-01-04:
Closing as completed.

closed 2025-01-04 by gitbls