gitbls/sdm #277: not working disk encryption : (initramfs) never showing

hugo-akaora, 2024-12-07:
Hello,

I followed the process at https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md to set up disk encryption.

I used the plugin cryptroot with these args: `cryptroot:"crypto=xchacha|nopwd|keyfile=/root/xxx-xxx-xxx-xxxx-xxx.lek"`

I burnt my image "2024-07-04-raspios-bookworm-arm64-lite" customized and then, the pi rebooted twice or three times.

At a moment I have a prompt for connecting, and a message displaying that some process for initramfs is taking place (sorry I didn't have any phone to take a picture of the screen at that moment !). Then it reboot another time. I have quickly some info on screen and then, nothing, the screen shut down, so I cannot process with configuring initramfs.

Don't know if I explain correctly!

In the meantime in noticed I'm not using the last version of raspios bookworm (2024-07-04), so I'll try with the latest one (2024-11-19).

gitbls, 2024-12-07:
Yes, indeed, this was broken by two optimizations that RPL made to the initramfs configuration. Both the changes are correct, but sdm has suffered a bit of collateral damage 🫤. Between some travel and US Thanksgiving holiday, I just got to this a couple of days ago.

Presumably you're using sdm V13.3 or you would have received an error when customizing the 2024-11-19 IMG, or an older IMG that has an `apt update`, `apt upgrade` done to it.

I've pushed a new `sdm-cryptconfig` to the sdm GitHub. You can either download that single file or simply reinstall sdm on the system where you perform your customizations. Once that's in place rerun your customize and burn and it *should* work. 

Alternatively, if you'd prefer to take the no-muss no-fuss path, sdm V13.4 will be released in the next couple of days, and you can simply install that version.

gitbls, 2024-12-07:
I just checked in V13.4. Please give it a try. Encryption now works fine in my testing.

hugo-akaora, 2024-12-09:
Hey, it's working, thank you!

closed 2024-12-09 by hugo-akaora

gitbls, 2024-12-09:
Excellent...thx for confirming!