gitbls/sdm #279: Burn command not updating/changing locale previously set during customize

mattie47, 2024-12-10:
Hi,
I'm running the latest 13.4, and found when burning images to devices in different timezones, that this config doesn't seem to update.

This is despite the logging giving the impression that it is (wifi config removed from output):

```
2024-12-10 16:39:42 > Burn '/home/pi/repos/builder/images/2024-11-19-raspios-bookworm-armhf.img' (5.5GB, 5.1GiB) to '/dev/mmcblk0'...
2024-12-10 16:39:42 > Burn command line: /usr/local/bin/sdm --burn /dev/mmcblk0 --host VHS-4 --expand-root --plugin L10n:locale=en_US.UTF-8|timezone=America/New_York|keymap=us --plugin network:ifname=wlan0|cname=removed|ctype=wifi|wifi-ssid=<>|wifi-password=<>|wificountry=US /home/pi/repos/builder/images/2024-11-19-raspios-bookworm-armhf.img
2024-12-10 16:39:42 > dd command: dd if=/home/pi/repos/builder/images/2024-11-19-raspios-bookworm-armhf.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
2024-12-10 16:44:36 > dd Copy completed
2024-12-10 16:44:37 > Set new disk ID 'f331265c' on '/dev/mmcblk0'
2024-12-10 16:44:38 > Disk ID 'f331265c' set successfully with sfdisk
2024-12-10 16:44:40 * Resize the ext4 root file system
2024-12-10 16:44:40 % (Ignore 'on-line resizing required' message)
2024-12-10 16:44:41 > Set MBR rootfs partition ID 'f331265c' in cmdline.txt (was 8793bfd3)
2024-12-10 16:44:41 > Set MBR partition Disk IDs in fstab
2024-12-10 16:44:41 > RasPiOS firstboot service not disabled; at least one of --expand-root and
2024-12-10 16:44:41   --regen-ssh-host-keys not set
2024-12-10 16:44:41 > Set hostname 'VHS-4'
2024-12-10 16:44:41 > First System Boot Custom Boot Scripts enabled
2024-12-10 16:44:41 > First System Boot automatic restart enabled with a 1 second wait
2024-12-10 16:44:41 * Plugins selected:
2024-12-10 16:44:41    * L10n
2024-12-10 16:44:41        Args: locale=en_US.UTF-8|timezone=America/New_York|keymap=us
2024-12-10 16:44:41    * network
2024-12-10 16:44:41        Args: ifname=wlan0|cname=<>|ctype=wifi|wifi-ssid=<>|wifi-password=<>|wificountry=US
2024-12-10 16:44:41 > Run Plugins Phase '0'
2024-12-10 16:44:41 > Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments:
2024-12-10 16:44:41   'locale=en_US.UTF-8|timezone=America/New_York|keymap=us'
2024-12-10 16:44:41 * Plugin L10n: Start Phase 0
2024-12-10 16:44:41 > Plugin L10n: Keys/values found:
2024-12-10 16:44:41    locale: en_US.UTF-8
2024-12-10 16:44:41    timezone: America/New_York
2024-12-10 16:44:41    keymap: us
2024-12-10 16:44:41 > Plugin L10n: Save L10n configuration
2024-12-10 16:44:41 * Plugin L10n: Complete Phase 0
2024-12-10 16:44:41 > Run Plugin 'network' (/mnt/sdm/usr/local/sdm/plugins/network) Phase 0 with arguments:
2024-12-10 16:44:41   'ifname=wlan0|cname=<>|ctype=wifi|wifi-ssid=<>|wifi-password=<>|wificountry=US'
2024-12-10 16:44:41 > Plugin network: Configuring 'network-manager' for network configuration management
2024-12-10 16:44:42 > Run Plugins Phase '1'
2024-12-10 16:44:42 > Run Plugin 'L10n' (/usr/local/sdm/plugins/L10n) Phase 1 with arguments:
2024-12-10 16:44:42   'locale=en_US.UTF-8|timezone=America/New_York|keymap=us'
2024-12-10 16:44:42 * Plugin L10n: Start Phase 1
2024-12-10 16:44:42 % Plugin L10n: Keymap 'us' will be set during sdm FirstBoot
2024-12-10 16:44:42 > Plugin L10n: Disable keyboard-setup service; will be re-enabled during sdm FirstBoot
2024-12-10 16:44:42 > Plugin L10n: Set locale to 'en_US.UTF-8'
2024-12-10 16:44:50 > Plugin L10n: Set timezone to 'America/New_York'
2024-12-09 22:44:51 * Plugin L10n: Complete Phase 1
2024-12-09 22:44:51 > Run Plugin 'network' (/usr/local/sdm/plugins/network) Phase 1 with arguments:
2024-12-09 22:44:51   'ifname=wlan0|cname=<>|ctype=wifi|wifi-ssid=<>|wifi-password=<>|wificountry=US'
2024-12-09 22:44:51 * Plugin network: Start Phase 1
2024-12-09 22:44:51 * Plugin network: Complete Phase 1
2024-12-10 16:44:51 > Run Plugins Phase 'post-install'
2024-12-10 16:44:51 > Run Plugin 'L10n' (/usr/local/sdm/plugins/L10n) Phase post-install with arguments:
2024-12-10 16:44:51   'locale=en_US.UTF-8|timezone=America/New_York|keymap=us'
2024-12-10 16:44:51 > Run Plugin 'network' (/usr/local/sdm/plugins/network) Phase post-install with arguments:
2024-12-10 16:44:51   'ifname=wlan0|cname=<>|ctype=wifi|wifi-ssid=<>|wifi-password=<>|wificountry=US'
2024-12-10 16:44:51 * Plugin network: Start Phase post-install
2024-12-10 16:44:51 > Plugin network: Keys/values found:
2024-12-10 16:44:51    ifname: wlan0
2024-12-10 16:44:51    cname: <>
2024-12-10 16:44:51    ctype: wifi
2024-12-10 16:44:51    wifi-ssid: <>
2024-12-10 16:44:51    wifi-password: <>
2024-12-10 16:44:51    wificountry: US
2024-12-10 16:44:51 > Plugin network: Set WiFi Country 'US' for sdm FirstBoot
2024-12-10 16:44:53 > Plugin network: Create connection '<>' on interface 'wlan0'
2024-12-10 16:44:53 * Plugin network: Complete Phase post-install
2024-12-10 16:44:53 * Burn Completed
2024-12-10 16:44:53 > Burn elapsed time: 00:05:11
2024-11-20 02:45:06 > sdm FirstBoot: Start sdm FirstBoot Service
2024-11-20 02:45:06 > sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
2024-11-20 02:45:06 > sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
2024-11-20 02:45:06 > sdm FirstBoot: Set WiFi Country to 'US'
2024-11-20 02:45:17 > sdm FirstBoot: Waiting 10/120 seconds for system time synchronization
2024-11-20 02:45:28 > sdm FirstBoot: Waiting 20/120 seconds for system time synchronization
2024-12-10 16:47:16 > sdm FirstBoot: System time synchronization achieved
2024-12-10 16:47:16 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/010-configure-keyboard.sh'
2024-12-10 16:47:24 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/040-realvnc-server.sh'
2024-12-10 16:47:27 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/080-eeprom-update.sh'
2024-12-10 16:47:33 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/091-apt-update.sh'
2024-12-10 16:53:11 > sdm FirstBoot: Disable sdm-firstboot service
2024-12-10 16:53:13 > sdm FirstBoot: Set boot behaviour to B4
2024-12-10 16:53:17 > sdm FirstBoot: Set systemd default to graphical.target
2024-12-10 16:53:19 > sdm FirstBoot: Enable lightdm
2024-12-10 16:53:24 > sdm FirstBoot: Re-enable getty@tty1
2024-12-10 16:53:26 > sdm FirstBoot: System will restart in 1 seconds
2024-12-10 16:53:29 > sdm FirstBoot: System restarting now
```

Meanwhile when I check the timezone post-burn:

```
timedatectl
               Local time: Tue 2024-12-10 17:02:08 NZDT
           Universal time: Tue 2024-12-10 04:02:08 UTC
                 RTC time: n/a
                Time zone: Pacific/Auckland (NZDT, +1300)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

Thanks,

Matt

gitbls, 2024-12-10:
Please provide the output from the following commands when run on the burned disk after it's booted:
```
cat /etc/default/locale
cat /etc/timezone
sudo cat /etc/sdm/0piboot/010-configure-keyboard.sh
```

mattie47, 2024-12-11:
Honestly, this is quite odd.

Burn cmd on host:

```
sudo umount /media/pi/*
sudo sdm --burn /dev/mmcblk0 \
--host VHS-11 \
--expand-root \
--plugin L10n:"locale=en_US.UTF-8|timezone=America/New_York|keymap=us" \
--plugin network:"wificountry=US" \
/home/pi/repos/builder/images/2024-11-19-raspios-bookworm-armhf.img
```

Meanwhile after, we see the correct timezone in /etc/timezone below, however the actual time the device reports and `timedatectl` is whatever the host device that completed the burn command is set to (in my case `Europe/Dublin`) 🤔 

```
pi@VHS-11:~ $ cat /etc/default/locale
#  File generated by update-locale
LANG=en_US.UTF-8
pi@VHS-11:~ $ cat /etc/timezone
America/New_York
pi@VHS-11:~ $ sudo cat /etc/sdm/0piboot/010-configure-keyboard.sh
#!/bin/bash
source /etc/sdm/sdm-readparams
logger "sdm FirstBoot: Configure keyboard to 'us'"
SUDO_USER=$myuser raspi-config do_configure_keyboard "us" nonint
systemctl enable keyboard-setup
pi@VHS-11:~ $ cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=e0da6c75-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
pi@VHS-11:~ $ time
time         timedatectl  timeout      times
pi@VHS-11:~ $ timedatectl
               Local time: Wed 2024-12-11 10:50:52 GMT
           Universal time: Wed 2024-12-11 10:50:52 UTC
                 RTC time: n/a
                Time zone: Europe/Dublin (GMT, +0000)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
pi@VHS-11:~ $
```

However once I set the timezone in raspi-config, it updates correctly:

```
pi@VHS-11:~ $ sudo raspi-config

Current default time zone: 'America/New_York'
Local time is now:      Wed Dec 11 05:56:21 EST 2024.
Universal Time is now:  Wed Dec 11 10:56:21 UTC 2024.

pi@VHS-11:~ $
```

Why am I doing things with different timezones? I'm shipping a bunch of Pis to colleagues in US/Europe hence coming across this oddity (Sadly actually just been made redundant with my employer leaving the New Zealand market at the end of this year, so I'm frantically getting devices to colleagues before I finish up my job). 

Thanks,

Matt

gitbls, 2024-12-11:
Thx for confirming that things are set correctly. Best guess is that something in the ` dpkg-reconfigure -f noninteractive tzdata` that raspi-config does when changing the timezone needs to actually be done on the running system rather than in systemd-nspawn. 

If it's reproducible (guessing it is) then it's a simple fix to run the raspi-config timezone setting during first boot.

gitbls, 2024-12-11:
OK, this will be fixed in the next release. Should be complete tomorrow, Friday at the latest (depending on what additional issues you find 🤣)

gitbls, 2024-12-14:
Fixed in V13.5. Please test/verify. Thx!

gitbls, 2025-01-04:
Closing as completed.

closed 2025-01-04 by gitbls