gitbls/sdm #280: git-clone plugin can't run preclone script as non-root user

jpuderer, 2024-12-10:
I'm trying to run a preclone script, but if I use a non root user, the copied script cannot be accessed.
```
> Plugin git-clone: Run preclone script '/etc/sdm/assets/git-clone/addGithubKey.sh'
> Plugin git-clone: Run command '/etc/sdm/assets/git-clone/addGithubKey.sh' as user 'ptk'
sudo: /etc/sdm/assets/git-clone/addGithubKey.sh: command not found
```

```
root@sdm:/# ls -al /etc/sdm/assets/git-clone/addGithubKey.sh
-rwxr-xr-x 1 root root 886 Dec 10 07:24 /etc/sdm/assets/git-clone/addGithubKey.sh
```

```
root@sdm:~# ls -ald /etc/sdm
drwx------ 6 root root 4096 Dec  9 17:35 /etc/sdm
```

Looks like the /etc/sdm directory has odd permissions?  The ptk user can't access the script.

gitbls, 2024-12-10:
Whoops. Yes, the /etc/sdm directory is protected root-only. Will fix.

jpuderer, 2024-12-10:
Hey great, Thanks!  

It was easy enough to work around, but I figured I'd log an issue since I had the cause nailed down.

Great project!

gitbls, 2024-12-14:
Fixed in V13.5. Please test/verify. Thx!

jpuderer, 2024-12-28:
Almost...

You made a change to the invocation of sudo, which breaks the plugin.

```
> Plugin git-clone: Run command 'git clone git@github.com:ips-ca/ptk-sw.git /home/ptk/ptk-sw' 
  as user 'ptk'
Unknown host QEMU_IFLA type: 54
Unknown host QEMU_IFLA type: 56
Unknown host QEMU_IFLA type: 57
Unknown host QEMU_IFLA type: 54
Unknown host QEMU_IFLA type: 56
Unknown host QEMU_IFLA type: 57
sudo: unknown user -i
sudo: error initializing audit plugin sudoers_audit
```

The change you made is:
```
- sudo -u $user $cmd
+ sudo -u -i $user $cmd
```

Reverse the order of '-u' and  '-i' and it's good.

jpuderer, 2024-12-28:
Added pull request.

gitbls, 2024-12-28:
> Almost...
>
> You made a change to the invocation of sudo, which breaks the plugin.
>
> ```
>   as user 'ptk'
> Unknown host QEMU_IFLA type: 54
> Unknown host QEMU_IFLA type: 56
> Unknown host QEMU_IFLA type: 57 [...]

Crap. Thanks for testing and providing the fix. Will get to it after the US new year.

gitbls, 2025-01-08:
Fixed in V13.6

closed 2025-01-31 by gitbls