gitbls/sdm #286: Stuck in bash : decryption password not asked after encryption complete

s0llvan, 2024-12-22:
Steps :

1. Install on SD card
2. Boot SD card
3. Install [rpi-clone](https://github.com/geerlingguy/rpi-clone)
4. Burn to NVME using rpi-clone
5. Boot NVME
6. Download, install, and run sdm-cryptconfig"
7. Reboot
8. Stuck in bash

Error :
```
Encryption complete. System boot will continue in 10 seconds

Connection to 192.168.1.95 closed by remote host.
Connection to 192.168.1.95 closed.
toor@mycomputer:~$ ssh root@192.168.1.95
bash-5.2#
```

Logs :
[logs.txt](https://github.com/user-attachments/files/18222520/logs.txt)

gitbls, 2024-12-22:
Curious. What happens if you run `cryptroot-unlock` after you SSH in?

s0llvan, 2024-12-22:
OS : 2024-11-19-raspios-bookworm-arm64-lite

`cryptroot-unlock` do nothing :

```
bash-5.2# cryptroot-unlock
bash-5.2#
```

s0llvan, 2024-12-22:
> Curious. What happens if you run `cryptroot-unlock` after you SSH in?
> 
> Also, what OS (version and 32 vs 64) are you using?

OS : 2024-11-19-raspios-bookworm-arm64-lite

`cryptroot-unlock` do nothing :

```
bash-5.2# cryptroot-unlock
bash-5.2#
```

gitbls, 2024-12-22:
It appears that something is horribly wrong with the system.

Does the system boot correctly if you enter the passphrase on the system console?

When you're in the bash prompt in the initramfs do other commands work, such as `ls`? If so, please provide the contents of /etc/dropbear/initramfs/dropbear.conf

s0llvan, 2024-12-22:
Does the system boot correctly if you enter the passphrase on the system console?

How can i do that...? because password not asked and `cryptroot-unlock mypassword` do nothing.

> When you're in the bash prompt in the initramfs do other commands work, such as `ls`? If so, please provide the contents of /etc/dropbear/initramfs/dropbear.conf

I don't have the **dropbear.conf** file in this directory, but in the parent directory:

```
bash-5.2# ls /etc/dropbear/
dropbear.conf              dropbear_ecdsa_host_key    dropbear_ed25519_host_key  dropbear_rsa_host_key
```

```
bash-5.2# cat /etc/dropbear/dropbear.conf 
#
# Configuration options for the dropbear-initramfs boot scripts.
# Variable assignment follow shell semantics and escaping/quoting rules.
# You must run update-initramfs(8) to effect changes to this file (like
# for other files in the '/etc/dropbear/initramfs' directory).

#
# Command line options to pass to dropbear(8)
#
DROPBEAR_OPTIONS="-I 300 -j -k -s -p 22 -c bash -r /etc/dropbear/dropbear_ed25519_host_key"

#
# On local (non-NFS) mounts, interfaces matching this pattern are
# brought down before exiting the ramdisk to avoid dirty network
# configuration in the normal kernel.
# The special value 'none' keeps all interfaces up and preserves routing
# tables and addresses.
#
#IFDOWN="*"

#
# On local (non-NFS) mounts, the network stack and dropbear are started
# asynchronously at init-premount stage.  This value specifies the
# maximum number of seconds to wait (while the network/dropbear are
# being configured) at init-bottom stage before terminating dropbear and
# bringing the network down.
# If the timeout is too short, and if the boot process is not blocking
# on user input supplied via SSHd (ie no remote unlocking), then the
# initrd might pivot to init(1) too early, thereby causing a race
# condition between network configuration from initramfs vs from the
# normal system.
#
#DROPBEAR_SHUTDOWN_TIMEOUT=60
```

gitbls, 2024-12-22:
Something is not making sense here. Even if you have installed SSH to unlock, you should still be able to enter your unlock passphrase on the console. There probably isn't a prompt for it, but if you type in the passphrase, it will be recognized.

Secondly, please explain how you are able to view files in the encrypted rootfs.

s0llvan, 2024-12-22:
> Something is not making sense here. Even if you have installed SSH to unlock, you should still be able to enter your unlock passphrase on the console. There probably isn't a prompt for it, but if you type in the passphrase, it will be recognized.

I entered the password alone, and the password with the command `cryptroot-unlock mypassword`, but nothing happens.

> Secondly, please explain how you are able to view files in the encrypted rootfs.

I don't have access to the encrypted partition, only to the unencrypted one, no...?

```
bash-5.2# ls /
bin              dev              lib              root-S83sp3zbvx  scripts          usr
conf             etc              proc             run              sys              var
cryptroot        init             root             sbin             tmp
bash-5.2# fdisk -l
Disk /dev/nvme0n1: 932 GB, 1000204886016 bytes, 1953525168 sectors
953869 cylinders, 64 heads, 32 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/nvme0n1p1    4,0,1       515,63,32         8192    1056767    1048576  512M  c Win95 FAT32 (LBA)
/dev/nvme0n1p2    516,0,1     1023,63,32     1056768 1953525167 1952468400  931G 83 Linux
bash-5.2# parted -l
Model: CT1000P3SSD8 (nvme)
Disk /dev/nvme0n1: 1000GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  541MB   537MB   primary  fat32        lba
 2      541MB   1000GB  1000GB  primary
```

s0llvan, 2024-12-22:
> Something is not making sense here. Even if you have installed SSH to unlock, you should still be able to enter your unlock passphrase on the console. There probably isn't a prompt for it, but if you type in the passphrase, it will be recognized.

I entered the password alone, and the password with the command `cryptroot-unlock mypassword`, but nothing happens.

> Secondly, please explain how you are able to view files in the encrypted rootfs.

I don't have access to the encrypted partition, only to the unencrypted one, no...?

```
bash-5.2# ls /
bin              dev              lib              root-S83sp3zbvx  scripts          usr
conf             etc              proc             run              sys              var
cryptroot        init             root             sbin             tmp
bash-5.2# fdisk -l
Disk /dev/nvme0n1: 932 GB, 1000204886016 bytes, 1953525168 sectors
953869 cylinders, 64 heads, 32 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/nvme0n1p1    4,0,1       515,63,32         8192    1056767    1048576  512M  c Win95 FAT32 (LBA)
/dev/nvme0n1p2    516,0,1     1023,63,32     1056768 1953525167 1952468400  931G 83 Linux
bash-5.2# parted -l
Model: CT1000P3SSD8 (nvme)
Disk /dev/nvme0n1: 1000GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  541MB   537MB   primary  fat32        lba
 2      541MB   1000GB  1000GB  primary
```

> Thirdly, the file that I asked about is most certainly found in /etc/dropbear/initramfs/dropbear.conf in the encrypted rootfs. If it's not there, then this is something else that does not make sense to me.

I understand, but the installation went smoothly according to the **logs.txt** file, didn't it? I'm having trouble identifying where the error is...

gitbls, 2024-12-22:
I agree that there is nothing obviously wrong in the logs you provided, but there is clearly something wrong.

What unencrypted partition do you have access to that has dropbear installed on it? That should only be installed in the encrypted rootfs, which is on the NVME. What disk are you referring to?

s0llvan, 2024-12-22:
> I agree that there is nothing obviously wrong in the logs you provided, but there is clearly something wrong.
>
> What unencrypted partition do you have access to that has dropbear installed on it? [...]

My mistake, there is indeed no encrypted partition, it's strange... and the **/etc/fstab** file is empty...

gitbls, 2024-12-22:
WHAT /etc/fstab? You said you can't see the encrypted partition. 

I'm sorry, I don't know how to help you with this because you are dribbling out information and forcing me to continue asking clarifying questions.

s0llvan, 2024-12-22:
> WHAT /etc/fstab? You said you can't see the encrypted partition.
> 
> I'm sorry, I don't know how to help you with this because you are dribbling out information and forcing me to continue asking clarifying questions.

I'm lost lol.

Why after installing sdm-cryptconfig, does the reboot not land on `(initramfs)` but in `bash-5.2` ?

```
Once at the (initramfs) prompt, connect the SSD or SD Card that will be used as a scratch drive

When you have the drive name enter the command:
    (initramfs) sdmcryptfs /dev/nvme0n1 /dev/sdX

Where:

    /dev/nvme0n1 is the name of your system disk
    /dev/sdX is the name of your scratch disk

sdmcryptfs will:

* Print the size of rootfs /dev/nvme0n1p2
* Save the contents of /dev/nvme0n1p2 to /dev/sdX
* NOTE: There will be no prompts for passphrases if --nopwd specified
* Enable encryption on /dev/nvme0n1p2
  * You will be prompted to enter YES (all in upper case) to continue
  * You will then be prompted to provide the passphrase for /dev/nvme0n1p2 (unless --nopwd)
    ** Be sure that your CapsLock is set correctly (in case you changed it to type YES)!!! **
* After a short pause you'll be prompted for the passphrase again to unlock /dev/nvme0n1p2 (unless --nopwd)
* The saved rootfs content will be restored from /dev/sdX to the encrypted rootfs
* When the restore finishes sdmcryptfs will exit and drop you to the (initramfs) prompt
* Type 'exit' to continue the boot sequence

* Once the system boots the sdm-cryptfs-cleanup service will run which:
  * Removes some one-time content and rebuilds initramfs
  * Reboots the system one last time

* As the system reboots you'll once again be prompted for the rootfs passphrase (Without the 30 tries)
  ** The system will now ask for the rootfs passphrase like this every time the system boots **

NOTE: You have configured SSH
      Please review https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md

pi@myserver:~ $ sudo reboot

Broadcast message from root@myserver on pts/1 (Sun 2024-12-22 16:42:02 CET):

The system will reboot now!

pi@myserver:~ $ Connection to 192.168.1.95 closed by remote host.
Connection to 192.168.1.95 closed.
toor@mycomputer:~$ ssh root@192.168.1.95
bash-5.2# sdmcryptfs /dev/nvme0n1 /dev/sda
```

gitbls, 2024-12-22:
It DOES land in initramfs. initramfs runs dropbear that you are then able to SSH into. Here are some questions that I need clear and complete answers from you on so that.

If you provide clear and complete information to these questions, I will try to repro your scenario since this path is leading nowhere. Some of these questions may seem obvious, but I still need responses from you on all of them to help sort out what went wrong for you.
* What Pi model?
* Was this a fresh install, or had you been running the system for a while before using rpi-clone?
* What software (rpi-imager, dd, etc) did you use to burn the OS to your disk?
* Did you fully configure keyboard, locale, after the system booted?
* What else did you do after the system fully booted the first time? 
* What was the exact rpi-clone command you used?
* Did you remove the original disk from your system and verify that you booted from the NVME before starting sdm-cryptconfig?
* I see that you're getting `locale` errors when you ran sdm-cryptconfig. Did you not configure the locale correctly or at all?
* Presumably you rebooted onto the NVME after you ran rpi-clone? Please confirm
* After the 'The system will reboot now!' did the system actually reboot? And, after that, did it reboot again? This last reboot is absolutely REQUIRED for clean up and completing the process. Since the dropbear.conf you can see shows `bash` instead of `cryptroot-unlock` it appears that the last reboot did not occur.
* I'm confused by your last message. Why are you re-running sdmcryptfs again (as you've shown at the end of your message)?
* Please provide a picture of The Pi's console after you SSH into your Pi and get the bash prompt. This should have some of the boot sequence messages on it, and then it will show dropbear being started.
* Where is the empty /etc/fstab file you previously mentioned? In the initramfs or somewhere else?

I KNOW that this works if done properly, so we need to figure out what is wrong with what you did. I can't do that with incomplete information.

s0llvan, 2024-12-23:
**Raspberry Pi 5 8G**

I just performed a fresh installation from scratch, here are the exactly steps I followed :

**Steps:**

1. Burn 2024-11-19-raspios-bookworm-arm64-lite.img to SD card

[image]

2. Insert SD card into Raspberry Pi

3. Start Raspberry Pi

4. Log in SSH
```
toor@mycomputer:~$ ssh pi@192.168.1.95
The authenticity of host '192.168.1.95 (192.168.1.95)' can't be established.
ED25519 key fingerprint is SHA256:Q6kqC8E+h8cCx59gkb6jG3uhySurnm8HRKmeQvXcv8c.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.1.95' (ED25519) to the list of known hosts.
Linux myserver 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
pi@myserver:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 16384 bytes
I/O size (minimum/optimal): 16384 bytes / 16384 bytes
[... 458 lines cut ...]

* Once the system boots the sdm-cryptfs-cleanup service will run which:
  * Removes some one-time content and rebuilds initramfs
  * Reboots the system one last time

* As the system reboots you'll once again be prompted for the rootfs passphrase (Without the 30 tries)
  ** The system will now ask for the rootfs passphrase like this every time the system boots **

NOTE: You have configured SSH
      Please review https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md

Broadcast message from root@myserver on pts/1 (Mon 2024-12-23 00:43:43 CET):

The system will reboot now!

Connection to 192.168.1.95 closed by remote host.
Connection to 192.168.1.95 closed.
```

16. Log in
```
toor@mycomputer:~$ ssh root@192.168.1.95
bash-5.2# 
```
instead of  `(initramfs)`

gitbls, 2024-12-23:
Thank you for all that. However, given the steps that you've stated, you should not be seeing what you are seeing.

So...What about these questions, please?

* After the 'The system will reboot now!' did the system actually reboot? And, after that, did it reboot again? This last reboot is absolutely REQUIRED for clean up and completing the process. Since the dropbear.conf you can see shows `bash` instead of `cryptroot-unlock` it appears that the last reboot did not occur.
* I'm confused by your last message. Why are you re-running sdmcryptfs again (as you've shown at the end of your message)?
* Please provide a picture of The Pi's console after you SSH into your Pi and get the bash prompt. This should have some of the boot sequence messages on it, and then it will show dropbear being started.
* Where is the empty /etc/fstab file you previously mentioned? In the initramfs or somewhere else?

Thanks

s0llvan, 2024-12-23:
> After the 'The system will reboot now!' did the system actually reboot? And, after that, did it reboot again? This last reboot is absolutely REQUIRED for clean up and completing the process. Since [...]

No, I only heard it reboot once.

>     I'm confused by your last message. Why are you re-running sdmcryptfs again (as you've shown at the end of your message)?

I only run it once here, no ?

`pi@myserver:~ $ sudo sdm-cryptconfig --ssh --authorized-keys .ssh/authorized_keys --hostname myserver --reboot`

>     Please provide a picture of The Pi's console after you SSH into your Pi and get the bash prompt. This should have some of the boot sequence messages on it, and then it will show dropbear being started.

There is nothing, the only thing that appears on the screen is `bash-5.2#` as I mentioned.

[image]

>     Where is the empty /etc/fstab file you previously mentioned? In the initramfs or somewhere else?

The /etc/fstab file I mentioned, I accessed it after the last connection, when `bash-5.2#` is displayed., with cat /etc/fstab.

s0llvan, 2024-12-23:
> After the 'The system will reboot now!' did the system actually reboot? And, after that, did it reboot again? This last reboot is absolutely REQUIRED for clean up and completing the process. Since [...]

No, I only heard it reboot once.

>     I'm confused by your last message. Why are you re-running sdmcryptfs again (as you've shown at the end of your message)?

I only run it once here, no ?

`pi@myserver:~ $ sudo sdm-cryptconfig --ssh --authorized-keys .ssh/authorized_keys --hostname myserver --reboot`

>     Please provide a picture of The Pi's console after you SSH into your Pi and get the bash prompt. This should have some of the boot sequence messages on it, and then it will show dropbear being started.

There is nothing, the only thing that appears on the screen is `bash-5.2#` as I mentioned.

[image]

There was no additional information.

>     Where is the empty /etc/fstab file you previously mentioned? In the initramfs or somewhere else?

The **/etc/fstab** file I mentioned, I accessed it after the last connection, when `bash-5.2#` is displayed., with `cat /etc/fstab`.

gitbls, 2024-12-23:
Perhaps I'm not being clear on "the console". "The console" is the monitor attached via HDMI to your Pi. If you don't have a monitor there, please connect one. I cannot help you without seeing the console, given the information you've provided.

s0llvan, 2024-12-23:
> Perhaps I'm not being clear on "the console". "The console" is the monitor attached via HDMI to your Pi. If you don't have a monitor there, please connect one. I cannot help you without seeing the console, given the information you've provided.

I didn't use a screen the last times (juin 2024) and it worked, it's strange... But now I just tried with the screen and dropbear starts correctly, then initramfs appears. But why doesn't it work via SSH ?

[image]

gitbls, 2024-12-23:
Thank you for the picture. It confirms my suspicion.

Here's what I'm seeing from my perspective: The encryption process did not complete successfully. It looks like the service that runs after the encryption is done (sdm-cryptfs-cleanup) either didn't run at all, or didn't run to completion. As a result, the initramfs that's in place on your system disk is still configured to encrypt the rootfs.

Why did this happen? I can't tell from the information you've provided. After this step:
```
Wipe scratch disk '/dev/sda' [y/N]? y
> 2024-12-22 15:45:29 > Write random data on '/dev/sda'
1180773+0 records in
1180773+0 records out
> 2024-12-22 15:46:59 rootfs Wipe elapsed time: 00:01:30

Encryption complete. System boot will continue in 10 seconds

Connection to 192.168.1.95 closed by remote host.
Connection to 192.168.1.95 closed.
```
The system should reboot, run the sdm-cryptfs-cleanup service, and then reboot again. This is what appears to not have happened as I mentioned above. 

If you're interested, you could try to sort this out by the following method. This is not super-precise, but should work:
* Create another MicroSD card with RasPiOS Lite on it
* Boot it on the Pi in question
* Install sdm into this system
* `sudo sdm --explore /dev/nvme0n1` -- encrypted
* Enter the passphrase
* Once you're in the container, we can look at the system logs. Things of interest:
  *  `journalctl --list-boots`
  * `journalctl | grep -E "sdm|sdm-cryptfs-cleanup"`
  * ...And probably other things once I have a look at the above

I should mention: I did exactly what you did on a Pi5 with an NVME drive this afternoon, and it worked correctly. I did NOT use rpi-clone (sorry, life is way too short), but I also doubt that it is part of the problem, since the system boots and runs on the clone copy. 

So, I'm quite convinced that it's not an sdm issue, and the above is my best explanation of what happened.

s0llvan, 2024-12-23:
The problem is that I cannot run this command `sudo sdm --explore /dev/nvme0n1 --encrypted` because the NVMe disk is not yet encrypted, as I cannot access `(initramfs)` via SSH :

```
pi@myserver:~ $ sudo sdm --explore /dev/nvme0n1 -- encrypted
* Mount device '/dev/nvme0n1'
mount: /mnt/sdm: unknown filesystem type 'crypto_LUKS'.
       dmesg(1) may have more information after failed mount system call.
? Error mounting device '/dev/nvme0n1
```

s0llvan, 2024-12-23:
I don't understand, I tried doing it differently using this command:

`sudo sdm --customize --plugin user:"setpassword=pi|password=test" --plugin disables:piwiz --plugin cryptroot:"ssh|authkeys=/home/pi/.ssh/authorized_keys|ihostname=myserver" --plugin L10n:"keymap=fr|locale=en_US.UTF-8|timezone=Europe/Paris" --hostname myserver --regen-ssh-host-keys --restart --expand-root 2024-11-19-raspios-bookworm-arm64-lite.img`

then

`sudo sdm --burn /dev/nvme0n1 --hostname myserver --expand-root 2024-11-19-raspios-bookworm-arm64-lite.img,`

but in SSH it's still impossible to reach initramfs...(even though it works with a screen)

s0llvan, 2024-12-23:
It's fine, I solved the problem differently by using `sdm --customize` and `sdm --burn`.

Thanks for the help !

closed 2024-12-23 by s0llvan

gitbls, 2024-12-23:
> The problem is that I cannot run this command `sudo sdm --explore /dev/nvme0n1 --encrypted` because the NVMe disk is not yet encrypted, as I cannot access `(initramfs)` via SSH :
>
> ```
> pi@myserver:~ [...]

Ah, I neglected to mention the step "apt install cryptsetup-bin". Sorry! But, glad that you've resolved the problem.