gitbls/sdm #289: rpi 4 lag and splah screen appear without no reasons

BobSquarePants, 2025-01-05:
Hi,

I've just installed `Raspberry Pi OS` on a raspberry pi 4 B

Version: 6.6.51+rpt-rpi-v8 (2024-10-08)

I've run [sdm-cryptconfig](https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#standalone-rootfs-encryption-configuration) and it worked flawlessly.

but now, when I reboot my system , unlock the luks partion, login, my system lag, when I tape something it miss some key !
and weirdly when I press twice the `up arrow key` this appear:
[image]

Sometime the systemd boot check reappear too then the shell `$:`

Any ideas what's going on ?

gitbls, 2025-01-05:
I've never seen anything like that.

What exact IMG name did you start with? How did you install the IMG onto the PI? rpi-imager or something else?

Did you wait until the system rebooted the last time? After the encryption step, the sdm-cryptfs-cleanup service needs to run to complete the cleanup and leave the system in a fully operational state.

Can you look at the system journal via journalctl and see that it ran and then rebooted the system again?

BobSquarePants, 2025-01-05:
Hi @gitbls 

I installed `Raspberry Pi OS` trough the `Raspberry Pi Imager` .appimage

> Did you wait until the system rebooted the last time?

of course.

`journalctl | grep cleanup ` returned (this is [OCRed ](https://en.wikipedia.org/wiki/Optical_character_recognition))
```shell
systemd[1]: Started sdm-cryptfs-cleanup.service - sdm cryptroot cleanup service.
sdm-cryptfs-cleanup[15571: /usr/local/bin/sdm-cryptfs-cleanup: line 19: /etc/sdm/sdm-readparams: No such file or directory
root[1559]: sdm-cryptfs-cleanup: Clean up initramfs content
root[15641: sdm-cryptfs-cleanup: Rebuild initramfs
sdm-cryptfs-cleanup[1565]: update-initramfs: Generating zbootzinitrd.img-6.6.51+rpt-rpi-v6
sdm-cryptfs-cleanup[9435]: '/boot/initrd.img-6.6.51+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
systemctl[94381: Removed "/etc/systemd/system/multi-user.target.wants/sdm-cryptfs-cleanup.service".
root[9493]: sdm-cryptfs-cleanup: The system will restart in 10 seconds
root[9495]: sdm-cryptfs-cleanup: System restarting now
systemd[1]: sdm-cryptfs-cleanup.service: Deactivated successfully.
systemd[1]: sdm-cryptfs-cleanup.service: Consumed 34.013s CPU time.
```
Line 2 show a problem in _line 19: /etc/sdm/sdm-readparams_

gitbls, 2025-01-05:
Huh. I've definitely tested sdm-cryptconfig on non-sdm-customized systems but obviously never saw the missing sdmr-readparams call. Thanks for finding that!

Unfortunately for solving the state of your system, it appears that there's nothing in sdm-cryptfs-cleanup that depends on that actually being there. 

I'm off to attempt a repro of this issue. Stay tuned.

gitbls, 2025-01-05:
Just did a full test. Here are the complete steps:
* Using rpi-imager to burn a virgin 2024-11-19 64-bit Lite IMG to an SD card
* Completed the boot process and logged in
* Downloaded sdm-cryptconfig and ran it
* After that completed, rebooted the system
* Went through the 30 tries to mount rootfs and dropped into initramfs
* Did the sdmcryptfs thing
* Exited and continued the boot process
* sdm-cryptfs-cleanup ran (with the sdm-readparams error your noted)
* System rebooted as expected
* When it came back up, everything works as expected
* Pressing the up arrow key twice does not cause a problem either at the `login: ` prompt or at the bash shell prompt.

So, I'm at a complete loss to explain the behavior you're seeing.  Did you vary from the above steps in ANY way?

I would really like to reproduce this so I can fix it, but if I can't repro it, I'm stumped 🤷‍♂️

gitbls, 2025-01-05:
One more thought. Although I don't think it matters, based on a quick test I just did, what disk drives are you using for
* The boot disk (that gets encrypted) (e.g., /dev/mmcblk0, /dev/sdX...)
* The scratch disk (ditto)

...and did you remove the scratch disk from the system after the encryption process completed?

BobSquarePants, 2025-01-08:
Hi @gitbls ,

Once again thanks for your time and guidance.

ok so I will retry with **upd software: yes** and I will post the outcome.

**fsarchiver** 
No worry I didn't use a restore from `fsarchiver` in previous try, I just did a copy, so normally `fsarchver` do not modify the content just read it.

Anyway I've realized that I can't restore **raspberry OS** backup made with `fsarchiver`.. with other Distro no problems... weird
Do you have any alternative ? maybe `sdm` can do backup/restore ?

gitbls, 2025-01-08:
OK, wasn't clear whether you used any output of fsarchiver or just did a backup.

For backups I use a combination of sdm and a nightly backup script.

sdm enables me to rebuild any Pi system disk in very short order. So, my strategy is to build a _common_ IMG used for all my "production" systems (actually I use it for all of them, even test systems).

Then, when I use sdm to burn that common IMG to a disk, I also include the appropriate server-specific plugins for that server.  I typically use Lite for all my systems (32 and 64-bit) but I could just as easily use Desktop versions.

For normal testing, I just use the common IMG. All my test systems boot up exactly the same, readily and easily operate on the network as soon as I login (ssh, NFS disks, etc), 

Here's a rough outline of it:
* Common IMG
  * All apps that I want (editors, tools, etc)
  * User configuration
  * System configuration (services, and a common "fstab extension"
* Specific server configuration (plugins run during burn). These are examples. I install specific stuff on specific servers
  * postfix for mail
  * logwatch
  * Host-specific network configuration (static IPs for servers)
  * Host-specific /etc/exports
  * VPN (pistrong/strongswan)
  * ndm (DNS/DHCP)
  * apt-cacher-ng
  * Samba
  * Travel hotspot (obviously not for a server 🤣)
  
Some of the above service installs use config files from a share where my "backups" go via arguments provided to each plugin. Basically, my backups are an updated-nightly per-host copy of all the config files I need to rebuild each of the above services. As part of incorporating a new service into my environment I sort out the config files up front, and make sure that they are appropriately preserved in this structure. 

The nightly backup script is a dumb script that copies newer versions of all these config files to the backup share, which, as a reminder, is used directly by sdm when building new disks.

Net net, I can rebuild any of my servers in less than 30 minutes. Even less if I have the correct Common IMG sitting around (I don't maintain that as a rule, since I can rebuild it in less than 5 minutes).

The last _**huge**_ benefit of this strategy is that moving to a new release of Debian is pretty simple as well. In fact, once sdm has been updated to accommodate the new Debian release (graphics, networking, etc etc ... all the fun things that changes with each release) moving a server to the new Debian release is literally two commands: `sdm --customize` and `sdm --burn`.

Sorry, I think sdm is so freaking cool, I get carried away talking about it 🤣

gitbls, 2025-01-10:
I missed the part in your previous description about changing boot to cli. When I made these changes I was able to reproduce the "weird typing problem", but not the very bizarre splash screen.

sdm-cryptfs-cleanup is not doing something completely correctly with the fact that you have an OS with Desktop but have changed it to boot to CLI. The net result is that plymouthd is still running which causes the "weird typing problem". 

If you switch to another TTY (CTRL-ALT-F2 for instance) the typing problem is not present. If you kill the running plymouthd that also eliminates the typing problem.

And, as you've discovered, the encryption works correctly until you make those last 3 changes in the Raspberry Pi configuration.

If you don't want the graphical login, don't use a Desktop version, use Lite.

I will investigate how to correct this, but since this can be worked around by using Lite, I do not consider this to be a high priority issue. I will document the restriction until it is corrected.

gitbls, 2025-01-11:
It was easier to fix than I had anticipated.  Switching to CLI when using a Desktop RasPiOS IMG will no longer have the laggy keyboard. I can't comment on the strange "splash screen" (or whatever it is), since I have been unable to repro that.

BobSquarePants, 2025-01-12:
@gitbls thanks for your fast reaction.

I've tested with **Raspberry OS Lite** and indeed on this version no error with `sdm-crytconfig` 👍

gitbls, 2025-01-12:
> @gitbls thanks for your fast reaction.
> 
> I've tested with **Raspberry OS Lite** and indeed on this version no error with `sdm-crytconfig` 👍

Great to hear!

I've made some improvements to sdm-cryptconfig and your scenario (using Desktop version and then changing to CLI) now works correctly. However, I'd still recommend using the Lite version if you want to boot to the CLI with encryption. 

If you're interested in trying it on the Desktop version again, LMK and I'll drop it to GitHub ahead of the next release.

Thanks for your patience in working through this with me.

gitbls, 2025-01-16:
The updated sdm-cryptconfig is available in V13.7

BobSquarePants, 2025-01-17:
> If you're interested in trying it on the Desktop version again
Humm now that I tested the Lite version I will never go back to the desktop one... :)

and I have tested the RPI with other distro that are way faster than this, but I fail to enable the luks encryption :/

closed 2025-01-17 by gitbls