gitbls/sdm #290: nosudo in user plugin documentation is misleading

gndControl, 2025-01-07:
First of all thanks for making such an awesome tool! I've been trying it out for the last few days setting up a production image creation script and this a bit confusing.

In the user plugin documentation, there is "nosudo" option with the descriotion "Do not enable this account for sudo" but this is not what it does. Per the code in the script all this does is either create or not the no password file in /etc/sudoers.d dir so that you skip inputing the password when running sudo. Even worse, a user could believe this prevents a user from running sudo but that is determined by being in the sudo group which is the default.

I recommend making the description clearer on what this option actually does. Ideally you'd also rename it to something like "nosudopass" but that would brake too many scripts and existing setups

gitbls, 2025-01-08:
Doc updated in V13.6

gndControl, 2025-01-09:
Thanks for the quick fix! Closing this issue as resolved.

closed 2025-01-09 by gndControl