gitbls/sdm #292: SDM fails to enter image's root, independently of using systemd-nspawn or chroot

Alouettesu, 2025-01-17:
I ran the sdm command for the initial configuration of the OS image. Whether I specified the --chroot switch or not, the command terminates with an error:

```
a@localhost:~/buildimg$ sudo sdm --customize images/myimg.img
* Mount IMG 'images/myimg.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize images/myimg.img
* Host Information
   Hostname:  localhost
   Memory:    15036660 kB
   uname:     Linux localhost 5.15.167.4-microsoft-standard-WSL2+ #1 SMP Sat Dec 28 09:37:58 CST 2024 x86_64 x86_64 x86_64 GNU/Linux
   os-release Name: Ubuntu 24.04.1 LTS
   Version: 24.04.1 LTS (Noble Numbat) (noble)
   Like OS: debian
   sdm Version: V13.7
* IMG Information
   Name: images/myimg.img
   Date: 2024-11-19
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
% sdm will use qemu 'aarch64'
% sdm will use systemd-nspawn on this 64-bit x86-64 host
  Retry the command with --chroot if this fails
* Plugins selected:
   * No plugins are selected
> IMG 'images/myimg.img' has 513688 1K-blocks (0.5GB, 0.5GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
* Phase 0 Completed
* Enter image 'images/myimg.img' for Phase 1
execv(/usr/local/sdm/sdm-phase1) failed: No such file or directory
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```

The message is uninformative, the file /usr/local/sdm/sdm-phase1 is present in the image. 

SDM output wrote that it will use qemu ‘aarch64’, however I found that qemu is not installed in the image, it is installed in host. So then I did this:

```
a@localhost:~/buildimg$ sudo sdm --mount images/myimg.img
* Mount IMG 'images/myimg.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* IMG 'images/myimg.img' mounted on /mnt/sdm
** BE VERY CAREFUL!! **
** Precede all path references by '/mnt/sdm' or you WILL modify your running system **
** Use 'exit' to Exit the bash shell and unmount the IMG
root@localhost:/mnt/sdm# cp $(which qemu-aarch64-static) /mnt/sdm/$(which qemu-aarch64-static)
root@localhost:/mnt/sdm# exit
exit
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```

After that the configuration worked, however it would have been better if the SDM script copied qemu into the OS image on its own.

gitbls, 2025-01-17:
I am unable to replicate this on my Debian WSL system.  There is absolutely no need to copy qemu into the IMG manually.

I'm looking into this.

gitbls, 2025-01-17:
I think there's something wrong with the binfmt support in Ubuntu 24.04, which is why you're running into this problem. I just started looking into this and found this: https://github.com/microsoft/WSL/issues/12013

As I mentioned before, this should all work seamlessly without you needing to copy anything into the IMG. This seamless behavior is implemented via binfmt. So, if there's an issue with binfmt on Ubuntu 24.04, it's no surprise that it's not working as it should.

Would you please provide the output from:
```
sudo journalctl -b | grep binfmt
```
It would help me determine which investigation paths to pursue.

Thx

Alouettesu, 2025-01-17:
This is my output for `sudo journalctl -b | grep binfmt`:

```
Jan 17 08:15:48 localhost systemd[1]: systemd-binfmt.service - Set Up Additional Binary Formats was skipped because of an unmet condition check (ConditionVirtualization=!wsl).
Jan 17 08:15:48 localhost systemd[1]: Starting binfmt-support.service - Enable support for additional executable binary formats...
Jan 17 08:15:48 localhost systemd[1]: Finished binfmt-support.service - Enable support for additional executable binary formats.
Jan 17 08:15:49 localhost systemd[1]: Starting wsl-binfmt.service - WSL binfmt restore...
Jan 17 08:15:49 localhost sh[182]: /bin/sh: 1: cannot create /proc/sys/fs/binfmt_misc/WSLInterop-late: Permission denied
Jan 17 08:15:49 localhost systemd[1]: wsl-binfmt.service: Deactivated successfully.
Jan 17 08:15:49 localhost systemd[1]: Finished wsl-binfmt.service - WSL binfmt restore.
Jan 17 08:49:24 localhost sudo[11514]:   a : TTY=pts/5 ; PWD=/home/a/buildimg ; USER=root ; COMMAND=/usr/sbin/update-binfmts --enable qemu-aarch64
```

Maybe SDM scripts copies qemu to another location of image? Actually, location of qemu on host machine must match to location of qemu in image.

I also had to manually add a description of the ELF aarch64 format, as it wasn't there after installing binfmts:

```
emulate_platform_spec_file=/usr/share/binfmts/qemu-aarch64
if [ ! -e $emulate_platform_spec_file ]; then
    bash -c "cat <<EOF >$emulate_platform_spec_file
package qemu-user-static
interpreter /usr/bin/qemu-aarch64-static
flags: OC
offset 0
magic \x7fELF\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7
mask \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff
EOF"
fi
```

gitbls, 2025-01-17:
As I mentioned previously, sdm on Debian Bookworm running on WSL works correctly with no manual changes such as the ones you are suggesting.

The issue you are seeing appears to be a problem with Ubuntu 24.04.

From my perspective you have a couple of options:
* Bring up a Debian WSL instance and use that, since it is known to work
* Use real Pi hardware running RasPiOS or Ubuntu (other than 24.04, which appears to have other issues on the Pi)
* Continue down the path you're on

If you wish to continue down the path you are currently on, I'm afraid that you're pretty much on your own. I don't have time in the next several weeks to do much with this, and given that Debian works while Ubuntu 24.04 doesn't, I'm fairly convinced that this is an issue with Ubuntu 24.04.

Sorry that I can't offer you any better alternatives.

Lastly, You asked "Maybe SDM scripts copies qemu to another location of image?".  sdm does not copy any qemu images either on the host OS or in the IMG being customized. Use of qemu by sdm is completely implemented via binfmt and works properly on several different OS. Just not Ubuntu 24.04

Alouettesu, 2025-01-22:
Really, it's working on Debian without copying qemu into image.

gitbls, 2025-01-22:
Sorry to disappoint you like that 🤣.

Seriously, glad that it's working. Too bad that Ubuntu 24.04 has broken this.

closed 2025-01-31 by gitbls