gitbls/sdm #296: Plugins fail to handle tilde in arg value

GeReV, 2025-01-24:
When running the following command containing a `~` in the value of a plugin argument, the script fails with the following error message:
```
sudo sdm --runonly plugins --plugin network:"ifname=wlan0|wifissid=foo|wifipassword=foo~bar"
```
It fails with the following error message:
```
? Unrecognized plugin 'bar'
```

My WiFi password does actually have a tilde in it 😅

gitbls, 2025-01-24:
Yep, sorry. Sdm uses the tilde to separate plugins internally. 

You'll have to provide a premade .nmconnection file or (gasp) change your WiFi password to not include a tilde.

GeReV, 2025-01-24:
Ah, I see.

I guess the first solution will be less of a hassle :)
Wasn't familiar with NM and how you configure it. Thanks for the quick reply!

closed 2025-01-24 by GeReV