gitbls/sdm #303: One or more switches conflict with --aptmaint

maxpavlov, 2025-02-23:
Help me understand what's wrong.

On a fresh .img I am calling 
'sdm --customize --chroot "$WORK_DIR/$IMAGE_NAME" --apt-options noupdate,noupgrade,noautoremove'

Everything works great.

Then I am calling:
'echo "y" | sdm --customize --chroot "$WORK_DIR/$IMAGE_NAME" --aptmaint update,upgrade,autoremove'

just to get: 
'One or more switches conflict with --aptmaint'

Is it somehow bringing back initial call switches?

gitbls, 2025-02-23:
If you review the output from `sdm --help | less`, you'll notice that several "commands" are specified at the top of the output, followed by various command switches.

You'll see that `sdm --customize` and `sdm --aptmaint` are both commands, which means you can only do one or the other.

In other words, customize and aptmaint are conflicting if used together. Use customize when customizing an IMG and aptmaint when you want to simply update the installed packages inside the IMG.

Also, as I previously mentioned to you, if you use the command switch `--redo-customize` when doing a 2-Nth customize on an already-customized IMG, you will not be prompted.

"Is it somehow bringing back initial call switches?" I have no clue what you mean by this. Please explain if you'd like a response.

maxpavlov, 2025-02-23:
Hehe, enterprise level help as usual, thank you @gitbls.
I was simply unattentive when you suggested the install sdm -> do aptmaint approach, missing out that aptmaint is a command like customize.

"Is it somehow bringing back initial call switches?"
By this I meant that sdm might be seeing initial switches to customize, -apt-options and was conflicting with it. It's clear now that this route of thinking was incorrect. 

I am lucky it's not the -chroot which is conflicting with aptmaint, otherwise my path would be at its end and I'd need to abando my "CI->docker" journey :)

closed 2025-03-16 by gitbls