gitbls/sdm #304: Unable to disable rfkill and get wlan0 to become available for nmcli

maxpavlov, 2025-03-02:
Hello. 

I've done a series of customizations to my .img from it's default bookworm stage but non specific to networking. But I am using a gadget mode like so in plugins file:
`gadgetmode:gadget-mode=shared|static-mac=b0:32:47:b2:bb:40|autoconnect-retries=5|dhcp-timeout=60`

I am using an image on a Raspberry Pi Zero 2 W. 

I can ssh into the Pi if my host has dhcp server running and I even tether internet to the pi. However default state of wireless is bt unblocked but wifi blocked. 

If I soft-unblock it via rfkill unblock wlan rfkill shows it's unblocked after the exectuion of the unblock command. However nmcli still lists wlan0 as disabled. 

`nmcli
usb0: connected to usb0-dhcp
        "usb0"
        ethernet (g_ether), B0:32:47:B2:BB:40, hw, mtu 1500
        ip4 default
        inet4 192.168.7.92/24
        route4 192.168.7.0/24 metric 100
        route4 default via 192.168.7.1 metric 100
        inet6 fe80::4ac9:47a9:d270:2770/64
        route6 fe80::/64 metric 1024

lo: connected (externally) to lo
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536
        inet4 127.0.0.1/8
        inet6 ::1/128

wlan0: unavailable
        "Broadcom BCM43438 combo and Bluetooth Low Energy"
        wifi (brcmfmac), E4:5F:01:51:67:0B, sw disabled, hw, mtu 1500

DNS configuration:
        servers: 8.8.8.8
        interface: usb0
`

Intuitively I'd need to restart network manager. When I do, rfkill goes back to the soft block status blocking wifi, thus wlan0 is permanently unavailable for me to manage via NM. 

I understand that this might be a non-sdm related questions however I see several plugins resolve the blockage via only calling rkfill unblock $interface at Stage 1 or something, thus perhaps I can get an answer here. 

Thanks in advance for any hints.

maxpavlov, 2025-03-02:
Oh, I think `rfkill unblock` as seen in scripts isn't enough. If I am trying to nm-control wifi on a runnin system, I need to do `nmcli radio wifi on` to enable wifi within nm "domain".

gitbls, 2025-03-02:
sdm accepts the `wificountry` argument in the `network` plugin and the `L10n` plugin.

If it's set via either plugin, then at the end of Phase 1 the script /etc/sdm/xpiboot/010-l10n-setup.sh is created with an appropriate `raspi-config do_wifi_country` command that is run during FirstBoot.

raspi-config's `do_wifi_country` function sets the WiFi country and does an `rfkill unblock wifi`.