gitbls/sdm #305: tutorial on raspberry

gogo2464, 2025-03-16:
The soft is great but the doc very confusing. It first guess that I have to run the script under the link https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#standalone-rootfs-encryption-configuration and then I realize that I have to install https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#the-sdm-cryptconfig-script so as I did not

Ensures the required disk crypto software is installed: cryptsetup, cryptsetup-initramfs, and cryptsetup-bin

then I have to reinstall my full raspberry.

It would be absolutely amazing if you could just make a tutorial of how to use the script to encrypt the raspberry sdcard.  it would be really less confusing instead of speaking of the options.

Best regards.

gogo2464, 2025-03-16:
@gitbls maybe should I just reboot?

gitbls, 2025-03-16:
I'm sorry that you're having problems with the document. It's quite simple actually.  As the document explains:

- https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#standalone-rootfs-encryption-configuration: "Download and run sdm-cryptconfig"
- Then, in that same section it directs you to run sdm-cryptconfig: `sudo sdm-cryptconfig`, which sets everything up AND ensures that sdmcryptfs is copied down and installed into the initramfs

The document IS a tutorial. However, it's possible that it assumes a level of expertise beyond your level.

I suggested on IRC that you read the entire document FIRST, before starting. Reading implies a certain level of comprehension.

Did you actually **read and understand the document** before starting the process?

If you're still interested in pursuing this method, please start over, and note each and every step that you do, and where you are running into a problem. Provide that information here and I'll be happy to help you sort out where you went astray.

Many other people have had this work very successfully for them. If you slow down, read **and understand** the steps, you'll quickly and easily join that group of happy people with encrypted rootfs on their systems.

gogo2464, 2025-03-16:
IO did:

```
# sdm-make-luks-usb-key and sdm-add-luks-key are only needed if you want to use a USB Keyfile Disk
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-make-luks-usb-key -o /usr/local/bin/sdm-make-luks-usb-key
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-add-luks-key -o /usr/local/bin/sdm-add-luks-key
sudo chmod 755 /usr/local/bin/sdm-add-luks-key /usr/local/bin/sdm-make-luks-usb-key

# sdm-cryptconfig is required for rootfs encryption
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-cryptconfig -o /usr/local/bin/sdm-cryptconfig
sudo chmod 755 /usr/local/bin/sdm-cryptconfig
sudo sdm-cryptconfig 

sudo reboot
```

now I am stuck at initramfs.

gogo2464, 2025-03-16:
I  am just confused about this chapter: https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#terse-description should I reboot during the initramfs to continue?

gitbls, 2025-03-16:
> I  am just confused about this chapter: https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md#terse-description should I reboot during the initramfs to continue?

Please describe in detail the steps you've taken this far.

gogo2464, 2025-03-16:
@gitbls Yes sure! Thank you very much for the attention and help.

I installed a fresh raspberry pi OS 5.

I ran sdm-cryptconfig :

```
# sdm-make-luks-usb-key and sdm-add-luks-key are only needed if you want to use a USB Keyfile Disk
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-make-luks-usb-key -o /usr/local/bin/sdm-make-luks-usb-key
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-add-luks-key -o /usr/local/bin/sdm-add-luks-key
sudo chmod 755 /usr/local/bin/sdm-add-luks-key /usr/local/bin/sdm-make-luks-usb-key

# sdm-cryptconfig is required for rootfs encryption
sudo curl --fail --silent --show-error -L https://github.com/gitbls/sdm/raw/master/sdm-cryptconfig -o /usr/local/bin/sdm-cryptconfig
sudo chmod 755 /usr/local/bin/sdm-cryptconfig
sudo sdm-cryptconfig 
```

and I rebooted with `sudo reboot`. Then I immediately did `sdmcryptfs /dev/mmcblk0 /dev/sda` but my initramfs does not sees sdmcryptfs at all.

sdmcryptfs has not been installed then.

gogo2464, 2025-03-16:
also I go to iniramfs becasue /dev/mapper/cryptroot dies not exists.

gogo2464, 2025-03-16:
I think there might be a missing empty partition in my sdcard

gogo2464, 2025-03-16:
should I add an usb stick or should I reduce the partition size?

gitbls, 2025-03-16:
Too many questions at once! I can't answer all of these. You didn't provide impotant information such as what OS and version. 

After you run sdm-cryptconfig, the initramfs (/boot/firmware/initramfs_2712 for a Pi5) will contain sdmcryptfs. This can be verified by `sudo lsinitramfs /boot/firmware/initramfs_2712 | grep sdmcryptfs`.

If it doesn't show up in the output of this command, then something is broken on your system. So...
* Did you `apt update` and `apt upgrade` before starting the encryption process? This is a really good idea
* Did you get any errors from sdm-cryptconfig?

Since you've run sdm-cryptconfig but it appears that your initramfs is hosed, you cannot recover this. I suggest:
* Start over with RasPiOS (either with desktop or Lite)
* Bring it fully up to date with apt update / apt upgrade
* Run sdm-cryptconfig
* Before rebooting, check the initramfs contents with the above lsinitramfs command. If it's not there, we'll have to figure out why. Don't reboot the system at this point

You have likely done something wrong, but I can't tell what because the information you've provided is incomplete.

gogo2464, 2025-03-16:
I am on raspios buellesyes arm64 (./2024-10-22-raspios-bullseye-arm64.img.xz)

gitbls, 2025-03-16:
> I am on raspios buellesyes arm64 (./2024-10-22-raspios-bullseye-arm64.img.xz)

Oh. This only works on Bookworm and later. Sorry! You should have mentioned that MUCH earlier.

Why are you using Bullseye anyhow? It's almost 4 years old.

gogo2464, 2025-03-16:
ah no sorry! 

It is weird lsb_relrease -a shows:

bookworm.

When ran on raspios

gitbls, 2025-03-16:
I'm sorry...I can't help you unless you slow down and provide thoughtful complete responses. So far you haven't done that.

Please see my prior note with a suggestion on how to proceed (your actions start with "I suggest"). 

Be deliberate, take good notes, and if you get to a stuck point (like no sdmcryptfs in initramfs), stop there and provide complete and detailed information.

This definitely works, as you saw from the completely unsolicited input from tesseract on IRC. If you do it correctly, it will work.

But please understand, a) this is not as simple as installing an app with apt, and b) It's not reversible. Once encrypted, your OS install will remain encrypted.

gogo2464, 2025-03-16:
Yes of course. Sorry I paniced. I strongly believed I was on bulleyes because of the name of the file but actually I think I am on bookworm according to the terminal.

I suggest that I just try a 5th install and then I will:
-1 past the command of sudo sdm-cryptconfig
-2 after try again with another install of bookworm

and then I will tell you how it works.

gitbls, 2025-03-16:
> Yes of course. Sorry I paniced. I strongly believed I was on bulleyes because of the name of the file but actually I think I am on bookworm according to the terminal.
>
> I suggest that I just try a [...]

I'm sorry, this makes no sense. Why would you try again with another install of bookworm AFTER running sdm-cryptconfig?

gogo2464, 2025-03-16:
because I wanted to ensure I really installed bookworm!

gogo2464, 2025-03-16:
Hello @gitbls !

I read very attentively your book!

It is now clearer for me!

I am preparing a 1TB hard disk. I will ping you this week when I will be ready to finish.

gogo2464, 2025-03-16:
I also confirm: I was on the wrong OS version!!! I will use https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2024-11-19/2024-11-19-raspios-bookworm-arm64.img.xz tomorrow!

gitbls, 2025-03-16:
> Hello [@gitbls](https://github.com/gitbls) !
> 
> I read very attentively your book!
> 
> Also I have both files /boot/firmware/initramfs_2712 and /boot/firmware/initramfs8 .It may explain.
> 
> It is now clearer for me! I also need a scratch hard disk. I am preparing a 1TB hard disk. I will ping you this week when I will be ready to finish.
> 
> It may explain why I do not have sdmcryptfs. It really could be a version issue!

On 64-bit systems you will have BOTH initramfs_2712 and initramfs8. Both are maintained and kept up to date. the 2712 version is for the Pi5 ONLY (16KB pages), while initramfs8 works on both the Pi4 and the Pi5 with 4KB pages. Read the RasPiOS docs at https://raspberrypi.com if you want to learn more about this.

This is COMPLEX stuff! I've tried to make it as simple as possible, and everything is documented. However, it may not be written clearly for everyone (an impossible task!), but all the information is there.

Read the entire doc again before you start.

And, remember, **EVERY detail matters**. If you run into any issues, provide **COMPLETE** information (OS version, what you did, what specific messages you received, etc etc). Trying to pull all that out of you (or anyone else reporting an issue) is *really tedious* (at best) or annoying AF (at worst).

If you use 64-bit RasPiOS as you indicated you will AND you have a scratch disk larger than your ROOTFS, you will succeed. It has worked for many people. Good luck!

gogo2464, 2025-03-17:
Here is the output of the command sudo sdm-cryptconfig  on raspberry 64 bits arm probably legacy bulleyes instead of bookworm:

```
pi@raspberrypi:~ $ sudo sdm-cryptconfig 
> Copy sdmcryptfs from GitHub
Install cryptsetup cryptsetup-initramfs cryptsetup-bin 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libcamera0.3 libwlroots12
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  cryptsetup cryptsetup-bin cryptsetup-initramfs
0 upgraded, 3 newly installed, 0 to remove and 4 not upgraded.
Need to get 715 kB of archives.
After this operation, 3,085 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 cryptsetup-bin arm64 2:2.6.1-4~deb12u2 [467 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 cryptsetup arm64 2:2.6.1-4~deb12u2 [213 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 cryptsetup-initramfs all 2:2.6.1-4~deb12u2 [35.6 kB]
Fetched 715 kB in 0s (2,289 kB/s)           
Preconfiguring packages ...
Selecting previously unselected package cryptsetup-bin.
(Reading database ... 149120 files and directories currently installed.)
Preparing to unpack .../cryptsetup-bin_2%3a2.6.1-4~deb12u2_arm64.deb ...
Unpacking cryptsetup-bin (2:2.6.1-4~deb12u2) ...
Selecting previously unselected package cryptsetup.
Preparing to unpack .../cryptsetup_2%3a2.6.1-4~deb12u2_arm64.deb ...
Unpacking cryptsetup (2:2.6.1-4~deb12u2) ...
Selecting previously unselected package cryptsetup-initramfs.
Preparing to unpack .../cryptsetup-initramfs_2%3a2.6.1-4~deb12u2_all.deb ...
Unpacking cryptsetup-initramfs (2:2.6.1-4~deb12u2) ...
Setting up cryptsetup-bin (2:2.6.1-4~deb12u2) ...
Setting up cryptsetup (2:2.6.1-4~deb12u2) ...
Setting up cryptsetup-initramfs (2:2.6.1-4~deb12u2) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-v8
'/boot/initrd.img-6.6.74+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-2712
'/boot/initrd.img-6.6.74+rpt-rpi-2712' -> '/boot/firmware/initramfs_2712'
> Update initramfs configuration for 'MODULES=most' and 'update_initramfs=all'
> Create /usr/bin/sdmluksunlock; runs in initramfs to unlock rootfs
> Create /etc/initramfs-tools/hooks/luks-hooks
> Update /etc/initramfs-tools/modules to include crypto 'aes' modules
> Enable KEYMAP=y in /etc/initramfs-tools/initramfs.conf
> Update initramfs with the rootfs encryption settings in place
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-v8
'/boot/initrd.img-6.6.74+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-2712
'/boot/initrd.img-6.6.74+rpt-rpi-2712' -> '/boot/firmware/initramfs_2712'
> Update root statement in cmdline.txt
> Add cryptdevice 'cryptroot' to cmdline.txt
> Updated cmdline:
console=serial0,115200 console=tty1 root=/dev/mapper/cryptroot rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles rw cryptdevice=/dev/mmcblk0p2:cryptroot luks.crypttab=no
> Update /etc/fstab for the encrypted rootfs
> Update /etc/crypttab
> Set up run-once service to rebuild initramfs after encrytped rootfs boot
> Create /usr/local/bin/sdm-cryptfs-cleanup script
> Ignore RequiresMountsFor errors in the system journal
> Enable console boot for next system restart; will be reset to graphical.target subsequently
> Enable verbose system restart

> Rootfs '/dev/mmcblk0p2' has 5808570368 bytes (5.8GB, 5.4GiB) used

> You will need another disk for the encryption process to use as a scratch disk
  This disk must be larger than (5.8GB, 5.4GiB) and it will be over-written

Reboot the system when you are ready to continue

The system will start to reboot, but hang trying to read rootfs. It will try 30 times
before giving up and dropping to the initramfs prompt: (initramfs)

** Ignore the 'ALERT! missing /dev/mapper' message. That is expected. **

Once at the (initramfs) prompt, connect the SSD or SD Card that will be used as a scratch drive

When you have the drive name enter the command:
    (initramfs) sdmcryptfs /dev/mmcblk0 /dev/sdX

Where:

    /dev/mmcblk0 is the name of your system disk
    /dev/sdX is the name of your scratch disk

sdmcryptfs will:

* Print the size of rootfs /dev/mmcblk0p2
* Save the contents of /dev/mmcblk0p2 to /dev/sdX
* NOTE: There will be no prompts for passphrases if --nopwd specified
* Enable encryption on /dev/mmcblk0p2
  * You will be prompted to enter YES (all in upper case) to continue
  * You will then be prompted to provide the passphrase for /dev/mmcblk0p2 (unless --nopwd)
    ** Be sure that your CapsLock is set correctly (in case you changed it to type YES)!!! **
* After a short pause you'll be prompted for the passphrase again to unlock /dev/mmcblk0p2 (unless --nopwd)
* The saved rootfs content will be restored from /dev/sdX to the encrypted rootfs
* When the restore finishes sdmcryptfs will exit and drop you to the (initramfs) prompt
* Type 'exit' to continue the boot sequence

* Once the system boots the sdm-cryptfs-cleanup service will run which:
  * Removes some one-time content and rebuilds initramfs
  * Reboots the system one last time

* As the system reboots you'll once again be prompted for the rootfs passphrase (Without the 30 tries)
  ** The system will now ask for the rootfs passphrase like this every time the system boots **
```

I have not set anythoer hard disk/ usb stick > 6GB during the command. Can I put it during the reboot or must I put it during the commadn start please?

gitbls, 2025-03-17:
As it says in the text you've shared:
```
Once at the (initramfs) prompt, connect the SSD or SD Card that will be used as a scratch drive
```

gogo2464, 2025-03-19:
I wanted to reoganie the full doc  of https://github.com/gitbls/sdm/blob/master/Docs/Disk-Encryption.md into 4 types templates:
-tutorials
-how to # there removed
-api ref # there removed
-explanation guide

Then the goal is to START with tutorial and then at the end, maybe an explanation. I want just to say (grosso modo):

- run the command
- insert usb stick
- run in initramfs
- reboot

it sounds greater. I felt your template might have a bit too much explanation. It is very interesting but pretty confusing and not necessary for first time user that want to deploy your, I remember it ABSOLUTELY amazing tools easily for real life purposes.

continue.

gitbls, 2025-03-19:
Thanks for your outline with your thoughts. TBH this sounds like a huge amount of work to create, and over time, creates even more work bc there would be 4 docs to update instead of just 1, so I'm not really interested in replacing the existing doc with the 4 docs you've outlined.

You're the first person to suggest that the current Disk Encryption doc is difficult to use, which also dampens my interest in rehashing an already apparently good enough document. 

**What do other people think about this?**

gogo2464, 2025-03-19:
It is 4 chapters. Not 4 websites.

Yes definitely. I think people worry about asking because it is the only tool that works on raspberry pi and:
- cryptography is generally a very technical subject. People do not want to seem pedant or idiot.
- sdm is the easiest way to use it with no raw luks primitives! Once again nobody wants to seem pedant.
- sdm is the greatest project of it category for raspios. Nobody wants to seem malicious or harmfull.
- documentation is always boring.

I need your opinion as well. :)

gitbls, 2025-03-26:
I spent a lot of time thinking about this and have decided to keep the 1-page doc as it is now.

You said `I read very attentively your book!`, after asking a LOT of questions, so it's pretty clear that you didn't read the doc earlier.

Thx for your thoughts on this.

closed 2025-03-26 by gitbls

gogo2464, 2025-03-27:
Yes. Sorry.

A great thank you for the help anyway!!!