gitbls/sdm #307: Burning an image to usb disk on Raspberry Pi fails due to `devmon` service

yogendra, 2025-03-19:
I am using a raspberry pi (server, no desktop) to burn the image on to a USB disk via sdm. In the `sdm` burn command, I encounter following as the last output.

```
? Device /dev/sda is already mounted
```

This causes the sdm to fail and exit without updating cmdline.txt among other things. 

It seems that the newly created partitioned are getting mounted automatically mounted by `devmon`  service, while burn command is in progress.

## Workaround
Turn off devmon@devmon service. This service mounts the USB disk right after partprobe is run. 

```
sudo systemctl disable --now devmon@devmon
```

## Potential Solution

sdm should detect if devmon or similar automounting services are present. And it should unmount the partition after partprobe is done.

gitbls, 2025-03-19:
Sorry you're having an issue with sdm. Looks to me like devmon is part of the `udevil` package, which is not installed by default on any RasPiOS IMGs.

As I'm sure you can understand, sdm can't possibly examine a nearly infinite list of optional packages to see if they are installed, so checking for "similar automounting services" is a non-starter.

That said, I'm not completely clear on the scenario in which you are seeing this.

If it's part of a burn command, please provide the complete output from the sdm burn command console session so I can review it to see if there's anything sdm can do that might improve the situation.

I need to see that output bc guessing what actually happened on your system is not a productive approach to resolving the issue.

Thx

gitbls, 2025-03-26:
Closing due to lack of activity. OP please feel free to re-open this issue for further investigation, and provide additional details per my prior response.

closed 2025-03-26 by gitbls