gitbls/sdm #315: app plugin usage: apt returned an error; review /etc/sdm/apt.log

satyabrata-debug98, 2025-05-15:
> IMG '2025-05-13-raspios-bookworm-arm64.img' has 8341864 1K-blocks (8.5GB, 8.0GiB) free at end of 'apt update'
> Run Plugins Phase '1'
> Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase 1 with arguments: 
  'name=mybrowsers|apps=firefox,chromium-browser'
* Plugin apps: Start Phase 1
> IMG '2025-05-13-raspios-bookworm-arm64.img' has 8341864 1K-blocks (8.5GB, 8.0GiB) free at start of [mybrowsers] Application Installs
* Start [mybrowsers] Application Installs
> [mybrowsers] Application Installs: firefox chromium-browser firefox chromium-browser
? apt returned an error; review /etc/sdm/apt.log
? Plugin 'apps' exited with failure status '1'
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted

its failing when I try to install any kind of app.'
I have try all aproaches;
1) (in command line using --plugins) & 

2) (using the ezsdm.sh file)

i used this in the ezsdm file, as it was explained the steps:

# Install apps
# https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#apps
apps:name=mybrowsers|apps=firefox,chromium-browser
apps:name=mytools|apps=keychain,lsof,iperf3,dnsutils
apps:name=aptlibs0|apps=python3-pip,tmux,iperf3,dnsutils,libopenblas-dev
apps:name=aptlibs1|apps=libatlas3-base,libcamera-dev,python3-opencv,portaudio19-dev
apps:name=aptlibs2|apps=util-linux,procps,hostapd,iproute2,iw,haveged,dnsmasq,iptables

How to check these errors? the path '/etc/sdm/apt.log' cannot be open, since there is nothing. I tried all the commands with sudo command also, but still it does not work.

I tried mannual sdm-apt install as well.
This is the output: 
~$ sudo sdm-apt 1 install firefox
* Start manual apt operation
> apt command: install firefox     
* Manual apt operation completed

How to check if the library/app was properly installed?

Please share your suggestions.

gitbls, 2025-05-15:
If sdm indicates that there is an error, as it did for you, you need to look IN the IMG that you're customizing:
```
sudo sdm --explore 2025-05-13-raspios-bookworm-arm64.img
```
Once you are inside the IMG you can look at /etc/sdm/apt.log and /etc/sdm/history to sort out what failed.

You haven't provided sufficient details for me to say exactly what happened (I would need to see the above-mentioned logs for that), but one possibility is that you'll see that there is insufficient space inside the IMG for what you are trying to install.

If that's the case, to remedy: start with a fresh, unmodified IMG and extend it when you customize using `--extend --xmb 2048`, which will extend the IMG by 2GB. 

The script `sdm-apt` isn't really useful any more, but the documentation for it indicates "sdm-apt is an optional script that you can use to issue apt commands when in Phase 1 or via `sdm --explore`". You did not use it in Phase 1, nor via `sdm --explore`, so it didn't do what you wanted/expected.

If you are unable to sort out the issue, please post the complete contents of /etc/sdm/apt.log AND /etc/sdm/history that you'll find IN the IMG as I mentioned above. You can copy them out of the IMG to your host system using `scp`, for example.

gitbls, 2025-05-20:
Debugging your host network configuration is well beyond the scope of sdm support.

In other words, it's up to YOU to configure your host system (VM in this case) with a proper network configuration, but I am willing to assist you a bit in identifying your vbox host/Ubuntu host network configuration issue.

You failed to specify what OS you are running on the vbox host, what version of vbox, and what version of Ubuntu you are running as a vbox guest. Please provide this information. 

In addition, please use the vbox GUI, examine your vbox settings for your Ubuntu VM and provide the network setting for _Attached to:_
 
Please run these commands on your Ubuntu VM and provide the output. There is no information that would be output that needs to be obfuscated, so do _NOT_ obfuscate _any_ of the output except for any domain names in /etc/resolv.conf, which you may change to _domain.com_
```
uname -a
ifconfig
ip addr 
ip route
ping -c3 $(ip route | awk '/default/ {print $3}')
ping -c3 8.8.8.8
ping -c3 google.com
cat /etc/resolv.conf
```
Also download and run the [mini-nspawn script](https://github.com/gitbls/sdm/discussions/178) using a freshly downloaded RasPiOS IMG and provide the output from the same commands above.

**NOTE:** I tried running `mini-nspawn` as I suggest below on Ubuntu 24.04.2 and it _fails_ to start. If it fails for you as well, then my suggested solution for you is to install Debian Bookworm in another vbox VM for use with sdm. This appears to be an Ubuntu problem in 24.04 (if that is the version you are running) that is independent of sdm itself, hence not an sdm issue.

For example:
```
sudo ./mini-nspawn 2025-05-13-raspios-bookworm-arm64.img
```
Once prompted, rerun the above diagnostic commands from within the script and provide those results as well. For example:
```
bash$ sudo ./mini-nspawn 2025-05-13-raspios-bookworm-arm64-lite.img 
* Human-readable partition list
Model:  (file)
Disk /ssdy/work/2025-05-13-raspios-bookworm-arm64-lite.img: 11681793s
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start     End        Size       Type     File system  Flags
 1      16384s    1064959s   1048576s   primary  fat32        lba
 2      1064960s  11681640s  10616681s  primary  ext4

* Machine-readable partition list
BYT;
/ssdy/work/2025-05-13-raspios-bookworm-arm64-lite.img:11681793s:file:512:512:msdos::;
1:16384s:1064959s:1048576s:fat32::lba;
2:1064960s:11681640s:10616681s:ext4::;
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot/firmware.
Spawning container sdm on /mnt/sdm.
Press ^] three times within 1s to kill container.
...
sdm~# uname -a
Linux sdm 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
sdm~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.92.142  netmask 255.255.255.0  broadcast 192.168.92.255
        ether 2c:cf:67:c6:3d:7c  txqueuelen 1000  (Ethernet)
        RX packets 1031994  bytes 81824299 (78.0 MiB)
        RX errors 0  dropped 29360  overruns 0  frame 0
        TX packets 13090  bytes 1552383 (1.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 112  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2  bytes 140 (140.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 140 (140.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

sdm~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:c6:3d:7c brd ff:ff:ff:ff:ff:ff
    inet 192.168.92.142/24 brd 192.168.92.255 scope global dynamic noprefixroute eth0
       valid_lft 73688sec preferred_lft 73688sec

# etc etc for all the above-listed commands

sdm~# exit
logout
Container sdm exited successfully.
bash$
```

satyabrata-debug98, 2025-05-21:
I resolved the previous issue with --chroot with the customize command.

 But now I have issue with ssh connection:
"" ssh: connect to host 192.168.0.10 port 22: Connection refused ""

I have used the correct commands:

# ssh config and authenticaton
sshd:enablesvc=yes|port=22|listen-address=192.168.0.10|password-authentication=no

# ssh key configuration
sshkey:sshuser=userx|keytype=rsa|import-key=/home/userx/.ssh/id_rsa

I can ping this IP address 192.168.0.10, but cannot ssh. 
i am confused between sshkey and sshhostkey plugin, which one is actually used to establish ssh connection between Rpi and my PC (PC and Rpi are connected with ethernet cable in my case).

please clarify.