gitbls/sdm #316: --convert-root btrfs = Error: Invalid number. Error creating boot partition

lockheed, 2025-05-23:
I am trying to customize and burn Debian 12 image from here.

Customisation part seems to go fine, but when I try to burn the image to the sd card, I get this:

```
root@vm:/home/ubuntu# sdm --burn /dev/sdb  --convert-root btrfs --expand-root 20231109_raspi_4_bookworm.img 
* Start Burn
> Burn '20231109_raspi_4_bookworm.img' (2.6GB, 2.4GiB) to '/dev/sdb'...
* Remove partitions on /dev/sdb
* Create MBR partitions on /dev/sdb
Error: Invalid number.
? Error creating boot partition on /dev/sdb
```

Removing "--convert-root btrfs" fixes the issue, but the whole point of customising this image is to have the system on btrfs.

gitbls, 2025-05-23:
Sorry you're having an issue. This is undoubtedly an issue with your host OS, which appears to be some unknown (to me) issue with whatever version of Ubuntu you're running. Unfortunately, you didn't provide sufficient information for me to comment further.

One thing you could consider trying is adding `--gpt` to your burn command, to see if burning the disk with a GPT partition format corrects the issue.

If that doesn't resolve your issue, please provide the output from the following:
```
root@vm:/home/ubuntu# cat /etc/os-release
sudo sdm --explore 20231109_raspi_4_bookworm.img
sdm~# head -n 20 /etc/sdm/history
sdm~# exit
```
Finally, as noted [here](https://github.com/gitbls/sdm/blob/master/Docs/Description.md), *sdm is for RasPiOS IMGs*.

While it's certainly possible that sdm will operate on your Debian 12 IMG, there is NO testing of sdm done on anything other than RasPiOS IMGs. I'm curious why you're using Debian rather than RasPiOS Lite, which is effectively Debian 12, and you can easily have your sdm customization *fix* anything that you find problematic with that IMG.

lockheed, 2025-05-23:
```
# sdm --burn /dev/sdb  --gpt --convert-root btrfs --expand-root 20231109_raspi_4_bookworm.img 
* Start Burn
> Burn '20231109_raspi_4_bookworm.img' (2.6GB, 2.4GiB) to '/dev/sdb'...
* Remove partitions on /dev/sdb
> Create GPT partitions on /dev/sdb
Could not create partition 1 from 5120000 to 0
Error encountered; not saving changes.
? Error creating boot partition on /dev/sdb

```

It is a standard Ubuntu 22.04 Server VM. 
```
# cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

```
# sudo sdm --explore 20231109_raspi_4_bookworm.img 
* Mount IMG '20231109_raspi_4_bookworm.img'
mount: /dev/loop6 mounted on /mnt/sdm.
mount: /dev/loop7 mounted on /mnt/sdm/boot/firmware.
% sdm will use qemu 'aarch64'
% sdm will use systemd-nspawn on this 64-bit x86-64 host
  Retry the command with --chroot if this fails
* Enter IMG '20231109_raspi_4_bookworm.img'
```

```
# head -n 20 /etc/sdm/history
2025-05-23 10:53:50 * Start Configuration
2025-05-23 10:53:50 > Command Line: /usr/local/bin/sdm --customize 20231109_raspi_4_bookworm.img --poptions 
2025-05-23 10:53:50   noupdate,noupgrade,noautoremove
2025-05-23 10:53:50 * Host Information
2025-05-23 10:53:50    Hostname:  vm
2025-05-23 10:53:50    Memory:    7932828 kB
2025-05-23 10:53:50    uname:     Linux vm 5.15.0-140-generic #150-Ubuntu SMP Sat Apr 12 06:00:09 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
2025-05-23 10:53:50    os-release Name: Ubuntu 22.04.5 LTS
2025-05-23 10:53:50    Version: 22.04.5 LTS (Jammy Jellyfish) (jammy)
2025-05-23 10:53:50    Like OS: debian
2025-05-23 10:53:50    sdm Version: V13.10
2025-05-23 10:53:50 * IMG Information
2025-05-23 10:53:50    Name: 20231109_raspi_4_bookworm.img
2025-05-23 10:53:50    Date: 
2025-05-23 10:53:50    RasPiOS Version: 12
2025-05-23 10:53:50    RasPiOS Architecture: 64-bit aarch64
2025-05-23 10:53:50    os-release Version: 12 (bookworm)
2025-05-23 10:53:50 % sdm will use qemu 'aarch64'
2025-05-23 10:53:50 % sdm will use systemd-nspawn on this 64-bit x86-64 host
2025-05-23 10:53:50   Retry the command with --chroot if this fails
```

I need to rune Home Assistant supervised on it so it requires Debian.

gitbls, 2025-05-23:
On what virtualization platform (vbox, hyper-v, xen, etc) is this VM running?

lockheed, 2025-05-23:
There's one aspect that suggests the usb is not to blame: the sdm burn process is successful if I don't use btrfs root coversion option.

gitbls, 2025-05-23:
> I see... I get the same problem with ext4.
> 
> Can you try if you get the same issue with this image on your machine? https://raspi.debian.net/tested/20231109_raspi_4_bookworm.img.xz

Which exact machine of mine do you mean by "on your machine"?

lockheed, 2025-05-23:
They way you usually do your images.

lockheed, 2025-05-24:
Thanks. What base os have you used to run sdm?

gitbls, 2025-05-24:
> Thanks. What base os have you used to run sdm?

This was on my customized RasPiOS Lite on a Pi5.The great thing about the Pi is that it's super-simple to switch to a different OS, made even simpler if you configure you storage to easily use NFS. For instance, I have a "work disk" that I can easily access from all my Pi systems (bc I configured them that way using sdm). Swapping an OS for testing is pretty simple.

lockheed, 2025-05-25:
Ok, I tried running it on Armbian on RPi4, and I get the same error:

```
$ sudo sdm --burn /dev/sdb  --convert-root btrfs --expand-root 20231109_raspi_4_bookworm.img
* Start Burn
> Burn '20231109_raspi_4_bookworm.img' (2.6GB, 2.4GiB) to '/dev/sdb'...
* Remove partitions on /dev/sdb
* Create MBR partitions on /dev/sdb
Error: Invalid number.
? Error creating boot partition on /dev/sdb
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Armbian-unofficial 25.08.0-trunk bookworm
Release:	12
Codename:	bookworm

```

gitbls, 2025-05-26:
Let me make this easier for you. sdm burns with btrfs correctly when run on a RasPiOS system running on a Pi.  If you want to use a different OS, different hardware platform, or burn different IMGs, that's fine, but I can only offer you so much help.

I suggest that you boot RasPiOS on your Pi4 and try the failing burn.

Since this is known to work on that hardware configuration with RasPiOS, if it fails then it's your disk (either the as-yet unspecified USB SD Card reader or whatever as-yet unspecified directly connected disk you're trying to burn to.

If it works, then it's something with the OS (Armbian) you're trying to use.

If you can identify the problem to be something in sdm in working incorrectly in your environment, I'm happy to fix it.  I've done that in the past for specific issues on Mint and Ubuntu, for instance. But it's going to be up to you to identify the problem, if it is indeed an sdm issue.

sdm is all written in bash, and you can have a look at `/usr/local/sdm/sdm-cmdsubs` function `copybypart` to see what commands it is executing. Something in this function is chorking on your system. You could zero in on the failing command by editing sdm-cmdsubs and putting in some debug prints. 

Here's the output from a burn to a MicroSD card connected to an Anker MicroSD adapter, run on a Pi5:
```
bash$ date
Mon May 26 11:22:20 AM PDT 2025
bash$ sudo sdm --burn /dev/sdc  --convert-root btrfs --expand-root 2025-05-13-raspios-bookworm-arm64-lite.img
* Start Burn
> Burn '2025-05-13-raspios-bookworm-arm64-lite.img' (6.0GB, 5.6GiB) to '/dev/sdc'...
* Remove partitions on /dev/sdc
* Create MBR partitions on /dev/sdc
> Set new disk ID 'e8dfac78' on '/dev/sdc'
Disk identifier changed from 0x1f077e36 to 0xe8dfac78.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID 'e8dfac78' set successfully with sfdisk
> Create bootfs file system on /dev/sdc
mkfs.fat 4.2 (2021-01-31)
> Create 'btrfs' rootfs file system on /dev/sdc2
* Mount IMG '2025-05-13-raspios-bookworm-arm64-lite.img'
mount: /dev/loop2 mounted on /mnt/sdmX.
mount: /dev/loop3 mounted on /mnt/sdmX/boot/firmware.
* Mount device '/dev/sdc'
mount: /dev/sdc2 mounted on /mnt/sdm.
mount: /dev/sdc1 mounted on /mnt/sdm/boot/firmware.
> Copy bootfs to /dev/sdc
> Copy rootfs to /dev/sdc
> rootfs copy complete
* Mount device '/dev/sdc'
mount: /dev/sdc2 mounted on /mnt/sdm.
mount: /dev/sdc1 mounted on /mnt/sdm/boot/firmware.
% hostname not specified with --host; will not be written
> Expand Root: Expand root partition 'sdc2' (btrfs) on device '/dev/sdc' from (5.4GB, 5.1GiB) to (69.3GB, 64.5GiB)
* Mount /dev/sdc to resize the root file system
* Mount device '/dev/sdc'
mount: /dev/sdc2 mounted on /mnt/sdm.
mount: /dev/sdc1 mounted on /mnt/sdm/boot/firmware.
* Resize the btrfs root file system
% (Ignore 'on-line resizing required' message)
Resize device id 1 (/dev/sdc2) from 5.06GiB to max
> Set MBR rootfs partition ID 'e8dfac78' in cmdline.txt (was d9c86127)
> Set MBR partition Disk IDs in fstab
> Disable unneeded RasPiOS firstboot service and /etc/init.d/resize2fs_once
> Set file system type 'btrfs' in cmdline.txt and /etc/fstab
> Disable regenerate_ssh_host_keys service per sshhostkey plugin
> First System Boot Custom Boot Scripts disabled
> First System Boot automatic restart enabled with a 10 second wait
> Install btrfs-progs onto burned disk
The following additional packages will be installed:
  liblzo2-2
Suggested packages:
  duperemove
The following NEW packages will be installed:
  btrfs-progs liblzo2-2
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 769 kB of archives.
After this operation, 4781 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 liblzo2-2 arm64 2.10-2 [51.8 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 btrfs-progs arm64 6.2-1+deb12u1 [717 kB]
Fetched 769 kB in 0s (18.0 MB/s)
Selecting previously unselected package liblzo2-2:arm64.
100%(Reading database ... 83102 files and directories currently installed.)
Preparing to unpack .../liblzo2-2_2.10-2_arm64.deb ...
Unpacking liblzo2-2:arm64 (2.10-2) ...
Selecting previously unselected package btrfs-progs.
Preparing to unpack .../btrfs-progs_6.2-1+deb12u1_arm64.deb ...
Unpacking btrfs-progs (6.2-1+deb12u1) ...
Setting up liblzo2-2:arm64 (2.10-2) ...
Setting up btrfs-progs (6.2-1+deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142+rpt3+deb12u3) ...
update-initramfs: Generating /boot/initrd.img-6.12.25+rpt-rpi-v8
W: Couldn't identify type of root file system for fsck hook
'/boot/initrd.img-6.12.25+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.12.25+rpt-rpi-2712
W: Couldn't identify type of root file system for fsck hook
'/boot/initrd.img-6.12.25+rpt-rpi-2712' -> '/boot/firmware/initramfs_2712'
Processing triggers for libc-bin (2.36-9+rpt2+deb12u10) ...

* Burn Completed
> Burn elapsed time: 00:02:12
* Storage '/dev/sdc' is ready
```