gitbls/sdm #320: btwifiset syntax

jg3, 2025-07-09:
AFAIK this might be just a bug in the doc/example here: https://deepwiki.com/gitbls/sdm/5.1-network-configuration-plugins#btwifiset-plugin-overview .

In the documentation it shows the usage as `sdm --customize example.img --plugin btwifiset country=US` but using that syntax consistently gets me an error: `? Plugin btwifiset: Required key 'country' missing from argument list ''`

With a little experimentation **I found this to work**:  `--plugin btwifiset:"country=US"`

My full command line:  `sudo sdm --customize --plugin user:"adduser=pi|password=<password>" --plugin btwifiset:"country=US" --plugin L10n:host --plugin disables:piwiz --expand-root --regen-ssh-host-keys --restart 2025-05-13-raspios-bookworm-arm64-lite.img` seems to have worked, for as far as I got.

( I got all the way up to "failed to write (No space left on device)" so, on to the next hurdle!)

jg3, 2025-07-09:
Thanks, I will contact that site's owner.  I did not realize it wasn't yours.

Also thanks, I eventually did figure out the --extend option.

Thank you for creating this, it's a neat system.

At the risk of being impertinent, I just got my first system up with btwifiset and while BTBerryWiFi does see the Pi, the Pi can't find any of the available SSIDs (I did try a second Pi just to be sure).   Is there something obvious I'm missing with that?

closed 2025-07-09 by jg3

gitbls, 2025-07-09:
By the way, thanks for enlightening me to deepwiki.com. There are some tremendously cool features there, such as the documentation including how things work, architectural pictures, and in some cases pulling text from both the sdm documentation AND the corresponding code. 😃

The only major problem I saw in my super fast skim-through was that every plugin example is wrong. This is most likely due to the required use of strange characters such as `"` and `|`. The good news is that not all examples are included.

If you happen to have found a way to contact deepwiki  I'd appreciate the pointer.

I did notice that there is a way to request a repo refresh. I'll take care of that after the next sdm release so that it's as up-to-date as possible. If there's a way to report buglets during or after that process I'll follow-up with them.

Thanks again for the deepwiki pointer. My wife did dev docs at MSFT for quite a while and I know she'll enjoy seeing this.

jg3, 2025-07-09:
Yeah I accidentally found it yesterday, I have never seen it before either.
I don’t love that I couldn’t figure out if there’s anyone in charge to fix
the issue, but I do love the organization and the graphics are cool just
knowing that they were automated by the AI.

If I do find a contact I will let you know.