gitbls/sdm #322: Bug: `plugin raspiconfig:blanking=0` does not disable screen blanking

ytret, 2025-07-22:
Hello,

I want to disable screen blanking in X11. However, the appropriate file in `/etc/X11/xorg.conf.d/` that disables screen blanking is not created after the first boot. It is created, though, if I manually use `raspi-config`.

**Expected behavior:**
Screen blanking is disabled in the config file `/etc/X11/xorg.conf.d/10-blanking.conf`:
```
Section "Extensions"
    Option      "DPMS" "Disable"
EndSection

Section "ServerLayout"
    Identifier "ServerLayout0"
    Option "StandbyTime" "0"
    Option "SuspendTime" "0"
    Option "OffTime"     "0"
    Option "BlankTime"   "0"
EndSection
```
I got this file after running `sudo raspi-config` interactively and disabling screen blanking.

**Actual behavior:**
Sreen blanking is not disabled, no file in `/etc/X11/xorg.conf.d/` that disables screen blanking exists.

**Steps to reproduce:**
1. Download `2025-05-13-raspios-bookworm-arm64.img` from [the Raspberry Pi OS site](https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2025-05-13/).
2. Customize the image (maybe add a user password to be able to log in):
```
sudo sdm --customize \
    --batch \
    --autologin \
    --apt-options "noupdate,noupgrade" \
    --plugin disables:piwiz \
    --plugin system:"service-mask=sleep.target,suspend.target,hibernate.target,hybrid-sleep.target"     \
    --plugin graphics:"graphics=x11"                                                                    \
    --plugin raspiconfig:blanking=0 \
    2025-05-13-raspios-bookworm-arm64.img
```
3. Burn the image:
```
sudo sdm --burn /dev/sdcard 2025-05-13-raspios-bookworm-arm64.img
```
4. Boot, maybe log in.
5. Remove the SD card, mount it and check `/etc/X11/xorg.conf.d/`, only `99-v3d.conf` exists.

[bug-burn.log](https://github.com/user-attachments/files/21370558/bug-burn.log)
[bug-customize.log](https://github.com/user-attachments/files/21370557/bug-customize.log)

ytret, 2025-07-22:
As far as I understand, the plugin executes `raspi-config do_blanking 0 nonint`, however that command doesn't create `/etc/X11/xorg.conf.d/10-blanking.conf` even if run manually on the Raspberry Pi.

gitbls, 2025-07-22:
I think you're providing the wrong argument for blanking. `blanking=1`  DELETES the file /etc/X11/xorg.conf.d/10-blanking.conf, while `blanking=0` CREATES it. I verified this by:
```
p022w~# raspi-config do_blanking 0 nonint
p022w~# l /etc/X11/xorg.conf.d/
total 8
drwxr-xr-x  2 root root 4096 Aug 18  2021 .
drwxr-xr-x 11 root root 4096 Jul 16 11:03 ..
p022w~# raspi-config do_blanking 1 nonint
p022w~# l /etc/X11/xorg.conf.d/
total 12
drwxr-xr-x  2 root root 4096 Jul 22 08:40 .
drwxr-xr-x 11 root root 4096 Jul 16 11:03 ..
-rw-r--r--  1 root root  247 Jul 22 08:40 10-blanking.conf
p022w~# cat /etc/X11/xorg.conf.d/10-blanking.conf 
Section "Extensions"
    Option      "DPMS" "Disable"
EndSection

Section "ServerLayout"
    Identifier "ServerLayout0"
    Option "StandbyTime" "0"
    Option "SuspendTime" "0"
    Option "OffTime"     "0"
    Option "BlankTime"   "0"
EndSection
p022w~# raspi-config do_blanking 0 nonint
p022w~# cat /etc/X11/xorg.conf.d/10-blanking.conf 
cat: /etc/X11/xorg.conf.d/10-blanking.conf: No such file or directory
```
You should be able to get what you want by using`-plugin raspiconfig:blanking=1`  to disable screen blanking

ytret, 2025-07-23:
You are right, `do_blanking 1` disables blanking. Apparently it's this way because `whiptail`, which `raspi-config` uses to show the yes-or-no dialog "Would you like to enable screen blanking?", exits with code `1` if the user chooses `No`.

Thanks for your time!

closed 2025-07-23 by ytret