gitbls/sdm #328: Best way to edit (config) files on the target system

gndControl, 2025-08-20:
Hello

One of the things I need to do, to setup a Raspberry Pi is to edit a .conf file of a systemd service. This ofcourse also applies more broadly to editing essentially any file on the fs. 

My question is: What is the best/intended way to do this? 
As in my case all the default options are commented out, I can just add a new line with the config I need. My first thought was to just echo and pipe the line I need into the file, but found no directly applicable plugin to do so. The ways I think it should be possible are as follows:
- Use the copyfile plugin to copy an allready modified file to replace the existing one
- Use the system plugin with systemd-config to replace with a preconfigured file (This might work for this specific use case, but not for other generic files)
- Use the runatboot or runscript plugin to run a custom script taht would edit the file
- Use a custom plugin (this seems excessive for such a simple operation)

Did aI miss anything else?

Thanks for the help

gitbls, 2025-08-20:
You've identified most of the ways that come to mind,

There are, of course, pros and cons to each approach, but personally, I'd think about those pros and cons in terms of:
* How does it fit with the way I use sdm?
* Is it more trouble than it's worth to create and maintain?
* What else do I want to do with sdm?
* Will I understand what I did when I look at this again in 2 years?

With that perspective, my personal favored approach is to use a custom plugin. Why?
* There will be other things, as well, that you can throw in there. Other config files, additional services, other very specific system config items, etc.
* You can do it _exactly_ the way you want it: create/update the config file, optionally enable the service.  If I'm updating a file I typically rename the existing copy and then drop the new file in. Bread crumbs for WTF happened!
* It won't be much code, and since other approaches would add another sdm command plugin, there's no net plugin increase
* You can put the plugin anywhere you want (doesn't need to be /usr/local/sdm/local-plugins...specify a path). Just keep it with all your other IMG source assets.  Personally, I like `./stuff/my-plugin` and then I simply backup the entire directory. Unfortunately, there appears to be a bug with this. Will be fixed in next release.

Without any further insight into what you want to do, my overall guidance is:
* Phase 0: Copy files from the host system to `$assetdir` that you want to implement into the system
* Phase 1: Copy the files from $assetdir to their final location and (optionally?) enable the service
* You can, of course, safely copy the file directly to it's ultimate destination during Phase 0, but in the general case there are issues such as: users aren't yet created and likely issues if the bash that started sdm has an other-than-default `umask`

gitbls, 2025-08-29:
Great that the system plugin does what you need, and thanks for pointing out that the copyfile plugin should have a `replace` or `overwrite` option to properly handle the case of replacing an existing file. Coming soon...