gitbls/sdm #331: Keyboard layout applies on first boot, but gets reset after reboot

gndControl, 2025-09-04:
When a keymap is set using the l10n plugin the keymap is correctly applied on first boot. However, after reset the keymap is back to US (or whatever the default is). The strange thing is that "localectl status" and the /etc/default/keyboard file report the correct keymap is configured. The same things happens when the keymap is set using raspi-config. Note I have disabled piwiz using the disabled plugin.

At quick glance it seems the keyboard service does not get turned back on after first setup.

I'm using the latest bookworm image and sdm v14.3

gitbls, 2025-09-04:
What keymap are you setting it to with the `L10n` plugin? Desktop or Lite?

gitbls, 2025-09-04:
Thanks, and Great find!

During customization sdm disables the keyboard-setup service, and informs you "Disable keyboard-setup service; will be re-enabled during sdm FirstBoot". 

Unfortunately, the re-enable part didn't happen 😲

You can correct this in a booting/booted system by `sudo systemctl enable keyboard-setup`. The next release of sdm will re-enable it correctly during the sdm FirstBoot.

closed 2025-09-16 by gitbls